RfidResearchGroup/proxmark3 #209: Legic_clone.lua // bad argument

thebasti314, 2019-06-04:
I tried to use the Legic_clone script. The script crashed in line 231. 

pm3 --> script run Legic_clone -i mydump.bin -d -s
[+] Executing: Legic_clone.lua, args '-i mydump.bin -d -s'

read 1 bytes from mydump.bin
/proxmark3/client/scripts/Legic_clone.lua:231: bad argument #1 to 'sub' (string expected, got nil)

Any ideas?

**Desktop (please complete the following information):**
Ubuntu 18.10
pm3 --> hw version

 [ Proxmark3 RFID instrument ] 
          

 [ CLIENT ]          
  client: RRG/Iceman          

 [ PROXMARK RDV4 ]          
  external flash:                  absent           
  smartcard reader:                absent           

 [ PROXMARK RDV4 Extras ]          
  FPC USART for BT add-on support: absent           
          
 [ ARM ]
 bootrom: RRG/Iceman/master/2f2796be 2019-05-08 14:28:12
      os: RRG/Iceman/master/e9eb7643 2019-05-17 06:40:06

 [ FPGA ]
 LF image built for 2s30vq100 on 2019/ 4/18 at  9:35:32
 HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23          

 [ Hardware ]           
  --= uC: AT91SAM7S256 Rev D          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 256K bytes, Used: 240837 bytes (92%) Free: 21307 bytes ( 8%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory        

 - hw status
pm3 --> hw status
#db# Memory           
#db#   BIGBUF_SIZE.............40000          
#db#   Available memory........35904          
#db# Tracing           
#db#   tracing ................0          
#db#   traceLen ...............25638          
#db# Currently loaded FPGA image           
#db#   mode.................... HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23          
#db# LF Sampling config           
#db#   [q] divisor.............95 (125 kHz)          
#db#   [b] bps.................8          
#db#   [d] decimation..........1          
#db#   [a] averaging...........Yes          
#db#   [t] trigger threshold...0          
#db# LF T55XX config           
#db#   [a] startgap............29*8 (232)          
#db#   [b] writegap............17*8 (136)          
#db#   [c] write_0.............15*8 (120)          
#db#   [d] write_1.............47*8 (376)          
#db#   [e] readgap.............15*8 (120)          
#db# Transfer Speed           
#db#   Sending packets to client...          
#db#   Time elapsed............500ms          
#db#   Bytes transferred.......327680          
#db#   Transfer Speed PM3 -> Client = 655360  bytes/s          
#db# Various           
#db#   MF_DBGLEVEL.............1          
#db#   ToSendMax...............-1          
#db#   ToSendBit...............0          
#db#   ToSend BUFFERSIZE.......2308          
#db# Installed StandAlone Mode           
#db#    LF HID26 standalone - aka SamyRun (Samy Kamkar)          
pm3 --> 

 - data tune
pm3 --> data tune
          
[=] Measuring antenna characteristics, please wait...
          
....
          
[+] LF antenna: 23,93 V - 125.00 kHz          
[+] LF antenna: 21,67 V - 134.00 kHz          
[+] LF optimal: 24,50 V - 126,32 kHz          
[+] LF antenna is OK  
          
[+] HF antenna: 15,63 V - 13.56 MHz          
[+] HF antenna is OK           
          
[+] Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.

RfidResearchGroup, 2019-06-04:
Looks like your binary file is very small..

```
read 1 bytes from mydump.bin
```

thebasti314, 2019-06-04:
I checked it: the file is a binary and is 1024 bytes big. 
$ ls -lisa
3278259   4 -rw-r--r--  1 root root   1024 Jun  4 05:53 mydump.bin

iceman1001, 2019-06-04:
I would start looking at the script where it loads the file.  It seem to some issue there

iceman1001, 2019-06-05:
the script most likely wants a eml version of the file.  It doesn't seem to load binary files. 
Try convert your bin -> eml

iceman1001, 2020-04-04:
@uhei   Maybe something more you can have a look at.

uhei, 2020-04-05:
> @uhei Maybe something more you can have a look at.

Already working on it (s. http://www.proxmark.org/forum/viewtopic.php?id=7710).

iceman1001, 2020-04-05:
nice.  I think I can close this issue since  legic_clone.lua isn't in the repo any longer.
legic.lua is the current script that is used.

closed 2020-04-05 by iceman1001