RfidResearchGroup/proxmark3 #215: About ISO 14443B' tags

jacopo-j, 2019-06-24:
Hello. I'm interested in reading ISO 14443B' (Innovatron pre-ISO 14443B) Calypso cards, which at the current state of things don't seem to be supported – or even detected – by the Proxmark client.

I am willing to implement at least some basic features for getting information and/or reading those tags myself, however I'm struggling to find any sort of documentation about how to communicate with the tags. Communication sniffing in 14b mode seems to produce unreliable results, and I still weren't able to have the card respond to raw 14b commands.

libnfc can at least select ISO 14443B' cards and is able to print the ATS and UID, but I couldn't find any information about how to select the card myself and send raw APDU commands.

Calypso cards (Navigo cards in particular) were previously discussed on Proxmark forums [here](http://www.proxmark.org/forum/viewtopic.php?id=2533) and [here](http://www.proxmark.org/forum/viewtopic.php?id=2572). A [calypso.lua](https://github.com/RfidResearchGroup/proxmark3/blob/master/client/scripts/calypso.lua) script exists in this repository but only works with standard 14443B cards.

Does anybody have further information about this topic?

joanbono, 2019-06-24:
Interested on that also.
I found a while ago Navigoat, an android application which was taking information from the Navigo card (last stations and so on), maybe it;s possible to use that information to decode some of the obtained data somehow (can be found [here](https://github.com/sp1d3rb0y/Navigoat/tree/master/app/src/main/res/xml) )

I should have one or two navigos at home, but I'm not living in Paris :/

iceman1001, 2019-06-24:
Indeed interesting to get calypso support.
The calypso lua script was thought of getting the cardpeek apdu's as a base to operate it.
As you noticed the 14B wasn't good and 14B' non-existent.    The need to implement signal modulation/demodulatio ,  encoding / decoding of 14B' protocol,   and client support..

I would say start with getting 14B better.  There is a fix for fpga signaling in offical repo that is on my todo list to implement in this repo.  That will increase the 14a/14b/15 communications.  And could most likely be uses for FeliCa and 14B'  protocols. 

Old issues I did over at Iceman fork.
https://github.com/iceman1001/proxmark3/issues/41
https://github.com/iceman1001/proxmark3/issues/39

doegox, 2019-06-30:
FTR modulations are the same as for typeB, typeB' has just a different "anticollision" phase. (and latest Navigo Easy are pure typeB, not typeB' anymore)

doegox, 2019-06-30:
You can find an example how to talk to typeB' here:
https://github.com/nfc-tools/libnfc/blob/master/examples/pn53x-tamashell-scripts/ReadNavigo.sh
For newer Navigo typeB, it looks more like Mobibi:
https://github.com/nfc-tools/libnfc/blob/master/examples/pn53x-tamashell-scripts/ReadMobib.sh

iceman1001, 2020-09-30:
I am not sure how to read the anticollision phase from those tamashell scripts.

doegox, 2020-09-30:
Very first command equivalent is
`hf 14b raw -c -p 010b3f80`
but it seems the Proxmark3 doesn't see the answer.

Here are (decimated) captures from another proxmark with `hf sniff`
First: sniffing libnfc  tamashell `42:010b3f80`
[image]
Then sniffing pm3 `hf 14b raw -c -p 010b3f80`
[image]
Note also the two extra signals at end of pm3 command ?!

doegox, 2020-10-01:
So after bugfix and addition of 14b' framing support, here we are:
[image]

iceman1001, 2020-10-01:
Remarkable!

iceman1001, 2020-10-02:
What is the select / anti collision for 14b'   ?

closed 2020-10-09 by iceman1001

valtoo16, 2021-10-04:
> So after bugfix and addition of 14b' framing support, here we are: [image]

hello, sorry for the reopening...
But I am facing this problem and with the latest relase and an rdv4 proxmark the command is not running. Did you publish your changes or did you just do it "locally"
If so, your modification will interest me :)

iceman1001, 2021-10-04:
yes, it was all commited.

valtoo16, 2021-10-04:
> yes, it was all commited.

it's strange because I do not have the return of the order on a calypso card... ( and with hf sniff, i have a modulation)

doegox, 2021-10-04:
Are you sure your Calypso is 14b' ? All modern Calypso are 14b nowadays

valtoo16, 2021-10-05:
> Are you sure your Calypso is 14b' ? All modern Calypso are 14b nowadays

I don' know, if the card repond to 14b select (05 00 00), it can't be 14b 'at the same time?
(I don't know I can't find any documentation on this subject)

https://image.prntscr.com/image/nVxeqSHVQ_ybOm6TlgK4jg.png

doegox, 2021-10-05:
No it can't be both. If it replies to 14b select, it *is* 14b.
Read the specs for more info.