RfidResearchGroup/proxmark3 #217: PM3Easy, all leds are on

the-d-kid, 2019-06-30:
Hi guys
I have a problem, this is very similar to what has already been discussed here, but I read all day forums and nothing helped me.
During the flash of my PM, something went wrong.
Now when he connects to his PC, all the LEDs are on. It doesn't matter if the button is pressed or not, no matter what OS is being used (I try it on win10 and ubuntu). I tried a lot of advice from the forum (hold button, kill modemmanager), but either I did something wrong, or they didn't help me. I have another PM3, and it works fine (no problem with flashing, using it). Just in case: I use the same cable and the same port USB. When connected Win10 does not define PM3. She writes "Unknown USB Device (Device Descriptor Request Failed)". Are there any chances to resurrect it without the use of third-party devices? (Segger J-LINK or Bus Pirate)
P.S. When trying to flash it on any OS "Waiting Proxmark" goes on forever
P.S.S Sorry for my English, I use google translate

iceman1001, 2019-06-30:
did you compile the project for PM3 EASY support?

the-d-kid, 2019-06-30:
i followed [these](https://github.com/Proxmark/proxmark3/wiki/Ubuntu-Linux) instructions

joanbono, 2019-07-01:
Hi @the-d-kid .

Create a file named `Makefile.platform` with the following:

```make
# If you want to use it, copy this file as Makefile.platform and adjust it to your needs
PLATFORM=PM3EASY
```

And compile again

(You can find instructions [here](https://github.com/RfidResearchGroup/proxmark3/blob/7e31e7f930c62e5cb6d1388d6d33424ae58d4fbb/common/Makefile.hal#L63) )

iceman1001, 2019-07-01:
yeah, that document could need an updated text.   Please do that :)

the-d-kid, 2019-07-01:
@joanbono Do I understand correctly, I need to create a file Makefile.platform in proxmark3/common/ and and write one line "PLATFORM=PM3EASY"?
This is work on official firmware or i need to compile iceman?

joanbono, 2019-07-01:
Hi @the-d-kid . The steps will be:

```sh
git clone https://github.com/RfidResearchGroup/proxmark3
cd proxmark3

# Now create the file
echo "PLATFORM=PM3EASY" > Makefile.platform

# Now it's time to compile
make clean && make all

# Reflash the pm3
./client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf armsrc/obj/fullimage.elf

# Run the client and use the pm3
cd client
./proxmark3 /dev/ttyACM0
```

the-d-kid, 2019-07-01:
@joanbono, i try it, but still "Waiting Proxmark" goes on forever, while on the other PM3Easy everything is fine

joanbono, 2019-07-02:
@the-d-kid so you are stuck in the flashing process now?

the-d-kid, 2019-07-02:
@joanbono Yes, that's the problem. Without flashing, PM3 does not work, it behaves like a brick, I described it above

joanbono, 2019-07-02:
If it's bricked, you will need to reflash it again with a JTAG or a Bus Pirate. It happened to me a while ago with PM3 Easy. Check [this](https://joanbono.github.io/PoC/Flashing_Proxmark3.html), @the-d-kid .

the-d-kid, 2019-07-02:
@joanbono Thank you. I wanted to make sure that I really need a JTAG of a Bus Pirate to solve this, since I don’t have it. Looks like I really have to look for him.

closed 2019-07-08 by the-d-kid