RfidResearchGroup/proxmark3 #220: Hitag sniff hangs if not first command

ViRb3, 2019-07-03:
Hitag sniff works only if it's the first command you run. If it's preceded by any other command (e.g. `hf search`) it hangs here forever:

https://github.com/RfidResearchGroup/proxmark3/blob/master/armsrc/hitag2.c#L742

ViRb3, 2019-07-03:
It looks to me that TC0 is not used or initialized in the sniff command yet we are waiting for it to be zeroed?

ViRb3, 2019-07-03:
No clue if a proper fix, but the below seems to work:
https://github.com/RfidResearchGroup/proxmark3/blob/a3962c16c3ef14c9c5d82755b2b36dcaea87b2f1/armsrc/hitag2.c#L742
Change to:
```c
while (AT91C_BASE_TC1->TC_CV > 0) {}; // wait until TC0 returned to zero
```

iceman1001, 2019-07-04:
Clock 0 - AT91C_BASE_TC0 
vs
Clock 1 - AT91C_BASE_TC1

Hm,  is there any more references to clock 0 in the hitag code?  If no,  then a change to clock1 is ok.

iceman1001, 2019-07-04:
A quick look at it,  looks like it only use clock1. 
Make a PR for your suggested fix :)

ViRb3, 2019-07-04:
Wouldn't it be better if we use clock0 instead?

iceman1001, 2019-07-04:
Don't think it does any difference.  clock1  in the sniff code will do just fine.

PR #221 (2019-07-04): Fix clock deadlock in hitag sniff
Fix #220