RfidResearchGroup/proxmark3 #223: [idea] Use LZ4 instead of zlib

slurdge, 2019-07-05:
Hi,

I tried some tests to replace zlib with lz4 in proxmark3.
Benefits of lz4:

- Eas(ier) to integrate, only 2 .c files for the client part (fpga_compress) and 1 .c file for embedded
- Faster for embedded
- Smaller for embedded

Drawbacks of lz4:

- Compression ratio is a bit worse

However, in my initial test, it ends up in a win because while the fpga_all.z is ~5Kb bigger, it's offset by not including zlib. I went from 81% space taken on a PM3EZ to 65%.

Current status:

- [X] Remove zlib and replace it with lz4 (in Makefiles,etc.)
- [X] Compress fullimage.data.bin.z with lz4
- [X] Compress fpga buffers with lz4
- [X] Rewrite start.c to be able to use lz4
- [ ] Rewrite fpga_loader to be able to use lz4
- [ ] Use block API with a ring buffer of appropriate size

Of course this is a very early look into it. I discovered that due to RAM constraint, there is a need to have a hard window (of maybe ~16KB ? BigBuff is around 40k so I guess it's ok). So it would maybe decrease a bit compression.

If we are trying to stream from proxmark3->client, this would be an absolute win IMHO as we would have much better performance.

Please tell me if you're interested by these experiments.

doegox, 2019-07-05:
completely :)

slurdge, 2019-07-05:
I played with the ringbuffer API for fpga bit files.

 Buffer Size (Kb)  |  Compressed size  
------------ | -------------
  16 |  48944
  32 |  46173
  48 |  44483
  56 | 43939
  64 | 43758
 128 | 43747
Zlib | 38207
 
Note that 128Kb == Normal API (which is normal, dataset < 128Kb).
From here, I would suggest increasing BigBuf from 40000 to 48 or 56*1024 bytes. It would preserve streaming capacities (so in the future, we can have more bitfiles) while having a close ratio to the maximum proposed by LZ4.
While Zlib compress better, it has more than 5k more code (at least in my compilations)

iceman1001, 2019-07-05:
Very interested!
My interest is in compressing the transfers between device & client.   Like  LF READ ...  Those 40kb download takes forever on a FPC / Bluetooth connecting.   Here the speed gains will be very good to have.

Super good suggestion @slurdge

iceman1001, 2019-07-05:
And remember that on RDV4 we have 200kb extra flash memory to play with.   From 256-512kb is another 256kb.
So we maybe don't need to compress the files....   But we can keep them uncompress for switching fpga images faster?:...

slurdge, 2019-07-07:
I had a look at the block/ringbuffer/streaming API. I succeeded in compressing and uncompressing the files, at least for the start.c.
However fpga files seems to error and make the FPGA unusable. Is there any way to get data out of the proxmark (like a debug print) so I can verify what's wrong with the decompression ?

doegox, 2019-07-07:
Look at Dbprintf functions

iceman1001, 2019-07-07:
you could download it aswell.  There is a readmem also.

slurdge, 2019-07-08:
Ok thanks I could debug. I pushed all my work into https://github.com/slurdge/proxmark3/tree/lz4
Basic results is everything works. However lz4 insists on inlining the main decompressing function, so it takes quite a bit of space.
Combined with the fact that we can only have a window of ~40000 bytes (and thus a block of this size), we end up losing approximately ~10Kb of space.

So, LZ4 lib is around the same size (of less if we hand optimize the case for the proxmark3) but compression ratio is quite worse. It seems the interlacing window does change the compression ratio but not by a huge factor.

That being said, it simplifies quite a bit the files needed for compilation. Also, it should be easier to compress data on the proxmark3 itself. To finish, I had the feeling that swapping fpga bitcode was a bit faster, but I can't produce number to substantiate that claim.

If either @doegox or @iceman1001 wants to play with it, I would be very happy to provide support and help.

iceman1001, 2019-07-08:
How about you try compressing the download function on the go with current zlib implemenation?
ie,  lf search downloads 40kb raw signal  ,  this download can be generic compress and uncompress before delivering the data back to the calling function.   comms.c / dl_it()  functions..   @doegox  can explain more :)

slurdge, 2019-07-09:
Should be possible but we do not include zlib compression on armsrc, it's only decompression for now. So it would increase the space needed for the firmware.
There's also the matter of needed RAM for compression... If we have 40kb raw, we have max 16kb for compression. But actually I think there is less free, so we would need to compress with a windows of around 8Kb

doegox, 2019-07-09:
About including zlib compression, see the (buggy at the moment) experiments in https://github.com/RfidResearchGroup/proxmark3/commits/bt_transfers:

* ffff5574e31c004074e299827cddfa333110db41
* 8f86d27112b10db0e0723ee5eeb20951985061c5

iceman1001, 2019-07-09:
Since we have so little memory,  we were discussing the idea of just compressing each package of 512bytes.  
and unpack on client side.

The links doegox provided tells a somewhat story.  Then the idea was to compress on the fly all data.
and send what was compress as it went along.  That idea isn't any good.

Good thing is that zlib should be able to work with a small memory footprint.
Some inital tests from doegox on  lf data files of 40kb,  shows resulting compressed filesizes of 6-12kb.
So the gain for Bluetooth comms (fpc) will have great potential. However this would speed up comms over usb aswell.  A great function to have in the project.

Since the fullimage is > larger than 256kb,  you will need to jtag to upload it to device.  
Current flasher will overwrite bootrom if you use it.  :(
but it will allow you to compress on device side :)

slurdge, 2019-07-09:
I don't have yet 512KB proxmark. However I managed to further push the lz4 branch down in size. With thumb opts, I'm at 83% space taken.
I agree that compressing all coms will be difficult with very small gains (with regular commands). So either discriminate on size (i.e. > 1K data is compressed). Both zlib&lz4 allow tuning the window for data so they should be able to compress small data packets.

iceman1001, 2019-07-09:
well,  you easily remove iso15 and any other HF stuff in the armsrc makefile and make a smaller fullimage to flash on your device.    Since its the lf read  -> lf signal download transfer that is the biggest target for compression

slurdge, 2019-07-24:
I did some more tests.

_  | Zlib | LZ4
--|----| ----
Size decompressor+compressed bitstream) | 441 blocks  | 449 blocks
HF upload | 1131 Ticks | 920 Ticks
LF upload | 1130 Ticks | 878 Ticks

So, in human units, we lost 2KiB of Flash, but gain ~20% speed in uploads.
Shall we create a branch on the main repository so people can try it in their workflow ?

slurdge, 2019-07-24:
I pushed LZ4 branch on my repository rebased on latest master.

iceman1001, 2019-07-24:
So ... what does HF / LF upload mean?   Uploading fpga images from arm -> fpga?  
since I am very keen on downloading from device to client.  BT transfers are slow and lf operations usually downloads 12000 - 40000 bytes from bigbuffer.  These transfers is the ones that we could benefit very much in speeds :)

slurdge, 2019-07-24:
Yes FPGA upload. Transfer to/from client are next :-)

iceman1001, 2019-07-24:
ok,  cool..  then go for it. 20% speed up in swapping images is awesome!

slurdge, 2019-07-25:
hmm, from my first tests, we send more than often 512bytes of data even though we do not use it. One example if hf search, it sends a 512 bytes buffer at some point. Compressed, its around 20 bytes. I do it transparently on the comm layer, I just prepend a MAGIC+SIZE (8 bytes) and detect it on the other end in comms.c

iceman1001, 2019-07-26:
I think you missed the new NG format for transport messages between device and client :)

slurdge, 2019-07-29:
I didn't, but the OLD format is used for most of the functions, so that's where I tried to reduce the size... I discussed a bit with doegox on how to improve the situation

iceman1001, 2019-07-29:
The idea is that all functions shall be converted to NG,   the old format shall be become a page in the history book

iceman1001, 2020-04-04:
@slurdge   ping!

slurdge, 2020-04-05:
pong! will try to take a look

iceman1001, 2020-06-02:
@slurdge  .....ping!

slurdge, 2020-06-02:
I'm looking at it right now, but I have hard time wrapping my head around how the dependencies are now managed. For lz4, I need to build two versions of the same lib with different switches.
For now I put it in common/lz4 but I guess I have to create a Makefile here that outputs a lib.a ? @doegox I may bother you with this one...

slurdge, 2020-06-06:
So, I made LZ4 working again with the latest version.
https://github.com/slurdge/proxmark3/tree/lz4_new
I included the compression of FPGA stream & data section.
I have the same results as before, we lose around 4Kb of flash for 20% of speed up in the FPGA loading.
Now, if we want to compress the communications, we will gain those 4Kb back because compression of LZ4 is much lighter than Zlib.

We really need:

1. Test this branch and see if anything is broken. In my tests everything works, but you never know;
2. Take a decision if we want to include it or not.

In the current state we are close to the 256Kb limit, but its possible to lower it again so in my opinion, we should do the switch and get on with a simpler architecture.

For the comms: I can either compress all data packets (by chunks of 512 bytes I believe), or we could decide to specifically compress the data acquisition buffers on the fly. Both approaches have their up and down, and we also need to take a decision here. I don't want to recreate the whole patch again in a few months.

1. Compression of all data packets: easier to put in place (only with NG packets), lower compression, agnostic to all data;
2. Compression of data: harder to put in place, we would need to forward information as to compress or not all the way down the call chain and possibly across the comm boundary, better compression, concerns only acquisition data

1. has lower latency, 2. has higher throughput (since we could transmit more than BIGBUF at a time) but same latency

iceman1001, 2020-06-06:
Excellent work bro!
I tried you lz4_new,  
And it doesn't compile.

[image]

We popped the 256 limit for RDV4 compilation already,  so I don't see no problem with it.
I am curious on how much less space the `lf read` trace will take when we compress it.
The number 1 seems interesting for me.  However with NG packages we already don't send so much.  Not sure the overhead here.  But for full 512 packages, its good.   both when uploading to device and downloading from device.

Not sure what you meant with number3,  wouldn't we transmit more than bigbuff also in both alternatives?

slurdge, 2020-06-06:
Hmm strange what are you compiling with ? I tried to put the -Werror=array-bounds and it doesn't trigger on my branch... You can try to silence it and see if it would work.

I can change the code but the basic is to recast this address...

lf read compression: depends if data is random or structured. I think we could expect around 20% compression.

Number 3 is a typo, it's "1".  We wouldn't transmit more than bigbuff in case of 1, but faster. 2. transmit more but slower. I think 1 would be easier to be frank.

doegox, 2020-06-07:
I tried to simplify these 32b operations:
start.c
```diff
-    int avail_in = *(&__data_start__ + 3);
-        avail_in = (avail_in << 8) + *(&__data_start__+2);
-        avail_in = (avail_in << 8) + *(&__data_start__+1);
-        avail_in = (avail_in << 8) + *(&__data_start__+0);  // compressed size. Correct.
+    int avail_in = *(int*)(&__data_start__); // compressed size. Correct.
```
fpgaloader.c
```diff
-        int cmp_bytes = *(compressed_fpga_stream->next_in + 3);
-            cmp_bytes = (cmp_bytes << 8) + *(compressed_fpga_stream->next_in+2);
-            cmp_bytes = (cmp_bytes << 8) + *(compressed_fpga_stream->next_in+1);
-            cmp_bytes = (cmp_bytes << 8) + *(compressed_fpga_stream->next_in+0);
+        int cmp_bytes = *(int*)(compressed_fpga_stream->next_in);
```
but this is strange. The change in start.c seems fine but if I do the change in ftpgaloader.c, the pm3 doesn't boon anymore.
Still, when I print both versions in appmain.c 
```
#include "lz4.h"       // uncompress
typedef struct lz4_stream_s {
    LZ4_streamDecode_t* lz4StreamDecode;
    char * next_in;
    int avail_in;
} lz4_stream;
typedef lz4_stream* lz4_streamp;
...
lz4_stream compressed_fpga_stream1;
lz4_streamp compressed_fpga_stream = &compressed_fpga_stream1;
compressed_fpga_stream->next_in = (char*)&compressed_fpga_stream;
Dbprintf("DDD %02X %02X %02X %02X", *(compressed_fpga_stream->next_in + 0), *(compressed_fpga_stream->next_in + 1), *(compressed_fpga_stream->next_in + 2), *(compressed_fpga_stream->next_in + 3));
        int cmp_bytes = *(compressed_fpga_stream->next_in + 3);
            cmp_bytes = (cmp_bytes << 8) + *(compressed_fpga_stream->next_in+2);
            cmp_bytes = (cmp_bytes << 8) + *(compressed_fpga_stream->next_in+1);
            cmp_bytes = (cmp_bytes << 8) + *(compressed_fpga_stream->next_in+0);
Dbprintf("DDD %08x", cmp_bytes);
        int cmp_bytes2 = *(int*)(compressed_fpga_stream->next_in);
Dbprintf("DDD %08x", cmp_bytes2);
```
both codes return the same value so endianess is correct.
I'm lost...

iceman1001, 2020-06-07:
I tried @doegox int explicit conversion,  now decompress_safe complains. 

[image]

iceman1001, 2020-06-07:
... if I remove the + 4 for,  it passes to the next error...

[image]

slurdge, 2020-06-07:
@iceman1001 your error comes from the fact that it considers char * to be char[1]. I don't know why your compiler thinks that, if you give me the version of gcc or clang I can investigate.
@doegox I was lost too on this error and hence split in 4 instructions.

doegox, 2020-06-07:
this one works for me, should be good for you too @iceman1001 
```diff
diff --git a/armsrc/fpgaloader.c b/armsrc/fpgaloader.c
index 0b7b24cb..bb50375f 100644
--- a/armsrc/fpgaloader.c
+++ b/armsrc/fpgaloader.c
@@ -190,10 +190,8 @@ bool FpgaSetupSscDma(uint8_t *buf, int len) {
 static int get_from_fpga_combined_stream(lz4_streamp compressed_fpga_stream, uint8_t *output_buffer) {
     if (fpga_image_ptr == output_buffer + FPGA_RING_BUFFER_BYTES) { // need more data
         fpga_image_ptr = output_buffer;
-        int cmp_bytes = *(compressed_fpga_stream->next_in + 3);
-            cmp_bytes = (cmp_bytes << 8) + *(compressed_fpga_stream->next_in+2);
-            cmp_bytes = (cmp_bytes << 8) + *(compressed_fpga_stream->next_in+1);
-            cmp_bytes = (cmp_bytes << 8) + *(compressed_fpga_stream->next_in+0);
+        int cmp_bytes;
+        memcpy(&cmp_bytes, compressed_fpga_stream->next_in, sizeof(int));
         compressed_fpga_stream->next_in += 4;
         compressed_fpga_stream->avail_in -= cmp_bytes + 4;
         int res = LZ4_decompress_safe_continue(compressed_fpga_stream->lz4StreamDecode, 
diff --git a/armsrc/start.c b/armsrc/start.c
index 8c769f56..25bfb2ad 100644
--- a/armsrc/start.c
+++ b/armsrc/start.c
@@ -16,6 +16,7 @@
 #include "appmain.h"
 #include "lz4.h"
 #include "BigBuf.h"
+#include "string.h"
 
 static uint8_t *next_free_memory;
 extern struct common_area common_area;
@@ -23,11 +24,8 @@ extern char __data_src_start__, __data_start__, __data_end__, __bss_start__, __b
 
 static void uncompress_data_section(void) {
     next_free_memory = BigBuf_get_addr();
-
-    int avail_in = *(&__data_start__ + 3);
-        avail_in = (avail_in << 8) + *(&__data_start__+2);
-        avail_in = (avail_in << 8) + *(&__data_start__+1);
-        avail_in = (avail_in << 8) + *(&__data_start__+0);  // compressed size. Correct.
+    int avail_in;
+    memcpy(&avail_in, &__data_start__, sizeof(int));
     int avail_out = &__data_end__ - &__data_start__;  // uncompressed size. Correct.
     // uncompress data segment to RAM
     int res = LZ4_decompress_safe(&__data_src_start__ + 4, &__data_start__, avail_in, avail_out);
````

doegox, 2020-06-07:
For the memmove:
```diff
diff --git a/armsrc/Makefile b/armsrc/Makefile
index 14747c7e..5544c1ac 100644
--- a/armsrc/Makefile
+++ b/armsrc/Makefile
@@ -78,7 +78,7 @@ FPGA_BITSTREAMS = fpga_lf.bit fpga_hf.bit
 #the lz4 source files required for decompressing the fpga config at run time
 SRC_LZ4 = lz4.c
 #additional defines required to compile lz4
-LZ4_CFLAGS = -DNEED_MEMMOVE -DLZ4_MEMORY_USAGE=8
+LZ4_CFLAGS = -DLZ4_MEMORY_USAGE=8
 APP_CFLAGS += $(LZ4_CFLAGS)
 # lz4 includes:
 APP_CFLAGS += -I../common/lz4
diff --git a/armsrc/string.c b/armsrc/string.c
index 8dbe2680..300e911d 100644
--- a/armsrc/string.c
+++ b/armsrc/string.c
@@ -20,6 +20,23 @@ void *memcpy(void *dest, const void *src, int len) {
     return dest;
 }
 
+void *memmove (void *dest, const void *src, size_t len)
+{
+  char *d = dest;
+  const char *s = src;
+  if (d < s)
+    while (len--)
+      *d++ = *s++;
+  else
+    {
+      char *lasts = (char*)s + (len-1);
+      char *lastd = d + (len-1);
+      while (len--)
+        *lastd-- = *lasts--;
+    }
+  return dest;
+}
+
 void *memset(void *dest, int c, int len) {
     uint8_t *d = dest;
     while ((len--) > 0) {
diff --git a/armsrc/string.h b/armsrc/string.h
index f8388d0a..53f8e8f7 100644
--- a/armsrc/string.h
+++ b/armsrc/string.h
@@ -16,6 +16,7 @@
 
 int strlen(const char *str);
 void *memcpy(void *dest, const void *src, int len);
+void *memmove (void *dest, const void *src, size_t len);
 void *memset(void *dest, int c, int len);
 int memcmp(const void *av, const void *bv, int len);
 void memxor(uint8_t *dest, uint8_t *src, size_t len);
@@ -182,26 +182,6 @@
 #include <string.h>   /* memset, memcpy */
 #define MEM_INIT(p,v,s)   memset((p),(v),(s))
 
-#ifdef NEED_MEMMOVE
-void *
-memmove (void *dest, const void *src, size_t len)
-{
-  char *d = dest;
-  const char *s = src;
-  if (d < s)
-    while (len--)
-      *d++ = *s++;
-  else
-    {
-      char *lasts = (char*)s + (len-1);
-      char *lastd = d + (len-1);
-      while (len--)
-        *lastd-- = *lasts--;
-    }
-  return dest;
-}
-#endif
-
 /*-************************************
 *  Common Constants
 **************************************/
```

doegox, 2020-06-07:
BTW @slurdge could you open a PR for your branch with title starting with "WIP" ? So we can push this kind of stuff directly in your branch. Thanks!

doegox, 2020-06-07:
And for the last warnings, I'll see how to rework the makefiles.

slurdge, 2020-06-07:
done.

doegox, 2020-06-07:
Thanks I pushed a few compilation fixes.
Indeed the `*(int*) ` introduced alignment problems. Clang alerted us about it on OSX for fpga_compress for the same code, so I chose to memcpy to int both on ARM & host codes.

iceman1001, 2020-06-07:
WSL
GCC:   `gcc (Ubuntu 9.3.0-10ubuntu2) 9.3.0`

Pulling latest fixes from you guys,   I still get an issue.
[image]

iceman1001, 2020-06-07:
As far as I can see,  those are all used as pointers,  but we don't define them as such.  We define them as  external char.     I really would need to get some write up how this whole thing works,   for the sake of trying to document the whole proxmark3 project.   The common_os,  the ldscript, these pointers,  etc.   and of course how this new compression works with the images.  The previous zlib wasn't super documented either but this time, please.   Having it would help understanding the overall flow,  or if a bug,  or when a refactoring is done,  These are the parts where I have struggled with long time.

```
__data_src_start__
__data_start__
__data_end__
```

doegox, 2020-06-07:
@iceman1001 the code you're compiling is not the code of the branch.
You should have https://github.com/RfidResearchGroup/proxmark3/pull/759/commits/d48fcbf5eb3f8f2d388b15cc9aca91a0c5571beb

iceman1001, 2020-06-07:
Pretty sure I pulled..   So I did again....  of course changes.. :)
however still compilation errors.

[image]

[image]

closed 2020-06-07 by iceman1001