RfidResearchGroup/proxmark3 #224: proxmark3.sh crashes

iceman1001, 2019-07-06:
**Describe the bug**
the changes introduced to proxmark3.sh script to autodetect com port on windows, crashes on proxspace.  This one is for you @slurdge 

**To Reproduce**
Steps to reproduce the behavior:
1. run proxspace
2. compile
3. run ' ./proxmark3.sh'
4. See error

**Expected behavior**
client to autodetect com port and run :)

**Screenshots**
```
pm3 ~$ ./proxmark3.sh
./proxmark3.sh: line 67: syntax error in conditional expression: unexpected token `('
./proxmark3.sh: line 67: syntax error near `MINGW(3'
./proxmark3.sh: line 67: `elif [[ "$HOSTOS" =~ MINGW(32|64)_NT* ]]; then'
pm3 ~$
```

doegox, 2019-07-06:
Bummer, we tested it under proxspace with @slurdge 
Just by curiosity @iceman1001 you used the latest proxspace ? As you've a few of them...
I'd suggest to change it into simply `elif [[ "$HOSTOS" =~ MINGW ]]; then`
Can you test @iceman1001 ?

slurdge, 2019-07-06:
Yes it works perfectly on my proxspace setup... It seems that the bash included doesn't support the `(` in the expression.

iceman1001, 2019-07-06:
no..  
my old ps didn't work.  regex in bash doesn't seem to work.
tested also on ps3.1 ,   where is does work but have the following new issue..
[image]

Where if you have more than one com port device hooked up it only looks at row 2.

slurdge, 2019-07-06:
Ah! We feared that may be the case in some setups. It will involve more complicated path to find the first "correct" line.

slurdge, 2019-07-06:
I have a simple fix:
```
wmic path Win32_SerialPort where "PNPDeviceID like '%VID_9AC4&PID_4B8F%'" get DeviceID, PNPDeviceID
```
Which is actually much simpler... Since we do not need the second filtering with this method.

doegox, 2019-07-06:
Yes please :)

iceman1001, 2019-07-06:
I still wonder about that limit of vid/pid.   There is more vid/pid in the wild..

slurdge, 2019-07-06:
Well the proxmark3.sh is for convenience, so it's alright to cater to the main case IMO. That's why the script also selects the first available proxmark3 even if there's more than one.

iceman1001, 2019-07-06:
Fair enough limitation.

commit 5fbbbc30 (2019-07-06): Fix #224
commit 672158e0 (2019-07-06): Merge pull request #225 from slurdge/master

Fix #224
closed 2019-07-06 by iceman1001