RfidResearchGroup/proxmark3 #230: [idea] hw disconnect and dynamic standalone modes

thesle3p, 2019-07-08:
Now that the RDV4 has a  battery module would it be possible to implement a hw disconnect command, for example if I wanted to  hide the rdv4  in my sleeve and read block 7 of every iclass tag it comes in range of I would make sure the battery was charged and run hw disconnect hf iclass readblk b 07 k <redacted key>, the command to be run in standalone mode being the argument hw disconnect takes. That way we don't have to fuss with a USB cable to bluetooth connection to use the proxmark3 in the field.

iceman1001, 2019-07-08:
interesting idea. 
but since the pm3 client stil have much logic of commands there is no generic way of implementing a functionality like you suggest.

thesle3p, 2019-07-08:
Hmmm, I guess there is not enough memory on the proxmark3 to push just enough code to run the specified command and store captured data. Oh well.

closed 2019-07-08 by thesle3p