RfidResearchGroup/proxmark3 #231: Trace load is broken

ViRb3, 2019-07-08:
## Reproduce:
`lf hitag read 26` - reads UID and saves to trace
`lf hitag list` - shows trace, all good
`trace list` - again, all good
`trace save test` - saves good
`hw reset` - clears buffer
`trace load test.bin` - says proper amount of bytes loaded
`trace list` - says 0 bytes in trace
`lf hitag list` - same

Am I missing something?

iceman1001, 2019-07-09:
Strange,  try load/save/list with some other traces like 14a ?

```
trace list hitag 
```

ViRb3, 2019-07-09:
How can I capture a different trace?

iceman1001, 2019-07-09:
just read any card,   look at the trace list  if you got some data..
```
hf 14a info  
trace list 14a 
trace save my14a.trace
data buffclear
trace load my14a.trace
trace list 14a
```

Your use of **hw reset**  will reset your device.  Not what you want to use.

ViRb3, 2019-07-09:
```
hf 14a info
trace list 14a // 103 bytes loaded
trace save asd // 103 bytes saved
data buffclear
trace list // empty
trace load asd.bin // 103 bytes loaded
trace list // 77 bytes loaded, nothing listed
```

iceman1001, 2019-07-09:
I belive you will need to instruct the list command that you already have data, or else it will try to download the trace from device again.

trace list 14a 1    

The helptext should have given hints about this
trace list h

ViRb3, 2019-07-09:
My bad, many thanks!

closed 2019-07-09 by ViRb3