RfidResearchGroup/proxmark3 #237: some problem in android.

xianglin1998, 2019-07-11:
1、ukbhit() function,  in util.c - > line 29 
the ukbhit() function always return -1, because tcgetattr() can't running on the non-console program.
fix:
`int flags;
    if (flags = fcntl(STDIN_FILENO, F_GETFL, 0) < 0) {
        LOGD("error: fcntl");
        return -1;
    }
    flags |= O_NONBLOCK;
    if (fcntl(STDIN_FILENO, F_SETFL, flags) < 0) {
        LOGD("error: fcntl");
        return -1;
    }
    //try read a char!
    int c = getchar();
    //if not -1，have keyboard input!
    bool ret = c != EOF;
    //rollback a char to stdin
    if (ret) ungetc(c, stdin);
    //block!
    flags &= ~O_NONBLOCK;
    if (fcntl(STDIN_FILENO, F_SETFL, flags) < 0) {
        perror("fcntl");
        return -1;
    }`
2、get_my_executable_directory() function , in proxmark3.c -> line 335
yes, the can installed app is run on art or davilk, is not console program, so it can't get runtime path with current xxx.apk env, so get_my_executable_directory always return NULL, very very sorry. i am only can return a const path, such as "/sdcard/"
other problem: some .c file, there are using relative path may be need change to absolute path.

iceman1001, 2019-07-11:
Great findings,
do you have any solutions for it on android?
Maybe hardcode a path for Android using #defines ...

xianglin1998, 2019-07-11:
I transplanted pm3(official & iceman yours) client to android, but i am direct change source code,  not anywhere using defines in ,  sorry.

xianglin1998, 2019-07-11:
Over time, I will try to release an open source Android rdv4 client, including the communication layer, and a simple UI. I will try to use # define for porting.

doegox, 2019-07-11:
Your solution for `ukbhit` is elegant but I've the following problem:
By default `getchar` will only get input after ENTER is pressed.
To modify this behavior, I would need to remove the ICANON flag by using `tcgetattr`/`tcsetattr` and we are back to your initial problem.

Maybe it's easier if you add a ukbhit code specific to Android ?
```
#ifdef __ANDROID__
int ukbhit(void) {
...
```
and if you can't detect the equivalent of a key press, just always return 0.

doegox, 2019-07-11:
ICANON setting depends indeed on the platform, it might be that it's disabled on Android.
Maybe we can use your function and add a tentative `tcgetattr`. If it doesn't return -1, remove ICANON with `tcsetattr` else, just run your function as it is. I'll do some tests later.

xianglin1998, 2019-07-11:
ok, thanks.

doegox, 2019-07-11:
see 844d732297d12f52fb43a3ac0696420605aa9af2

xianglin1998, 2019-07-11:
ok, very good, i am try porting client 2 android, if i encounter other problem,  feedback again. :)

doegox, 2019-07-11:
About 2、get_my_executable_directory() function , in proxmark3.c -> line 335
I guess you will have to add some patches specific to Android, I don't see how to make it generic.
Please open new issues if you find more things to adjust in the main repository.
Thank you.

closed 2019-07-11 by doegox

iceman1001, 2019-07-12:
@xianglin1998   you have a link to your android code?

xianglin1998, 2019-07-12:
not have, is a close sourece in current version, atfer short time,  i can separate from that,  thar project have all rdv4 android lib, use cmake build, and base ui, and some tools ui, such as dump editor and key editor and other ui tools, but i need some time.

xianglin1998, 2019-07-12:
if you have interested, i can after from my sourece project separate, shared to you.

iceman1001, 2019-07-12:
@xianglin1998   that would be very intersting to see.  I have seen a picture from your android app.

iceman1001, 2019-07-12:
if you make it open source,  other ppl can start helping out.