RfidResearchGroup/proxmark3 #240: Last bit omitted in rawdemod

ViRb3, 2019-07-11:
Following @doegox's commands for manually reading a hitag2 UID:
```
lf cmdread d 50 z 166 o 116 c 000111
data ltrim 200
data norm
data rawdemod am
data printdemodbuffer o 5 x
```
Using the hitag2 module:
```
lf hitag read 26
// or
lf hitag info
// or
if hitag read 21 ... (makes dump)
```

Will result in the same UID, however, the last bit is `0` using the manual read, while it is a `1` using the hitag2 module. I tested this on two different cards, however, I cannot verify which one is the true UID.

doegox, 2019-07-11:
Good catch, you're right.
I tried on a car key to get another ID and by chance its last bit is a 1.
My manual steps return an ID=xxxx4D6A but another reader (ACG-LF) returns xxxx4D6B.
Here we see the Manchester decoder stopped before the last weaker "1"(HL) bit:
[image]

iceman1001, 2019-07-12:
there are most certain same issues with NR demod ..

iceman1001, 2019-10-08:
@ViRb3   Some news?

doegox, 2019-10-08:
😙

iceman1001, 2020-09-10:
@doegox   what is your output from car key and hitag today?

doegox, 2020-09-16:
Manual decoding gets still last bit wrong, but `lf hitag reader` works fine
```
[usb] pm3 --> lf cmdread d 50 z 166 o 116 c 000111
[+] Sending command...
[#] Done, saved 42216 out of 0 seen samples at 8 bits/sample

[+] downloading response signal data
[usb] pm3 --> data ltrim 200
[usb] pm3 --> data norm
[usb] pm3 --> data rawdemod am
11111011100110111101001001101011
0101.0
[usb] pm3 --> data printdemodbuffer o 5 x
[+] DemodBuffer: 737A4D6A[3]

[usb] pm3 --> lf hitag read 26
[+]  UID: 737a4d6b
```

iceman1001, 2026-09-10:
hitag support is ok now.

closed 2026-09-10 by iceman1001