RfidResearchGroup/proxmark3 #241: Reader and Proxmark distance, experiencing instability

hornyd, 2019-07-11:
I am mainly focusing on simulating Mifare cards.
In order to have the reader to detect the card, it requires a specific distance between the reader (VIX HID) and the Proxmark.

- Load the binary in the Proxmark
- Simulate the card

Approach the proxmark to the HID reader, the antenna is very sensitive. moving the proxmark a couple of millimeter can affect the good functioning of the simulation.

Anyone has a trick or a change could improve the sensitivity of the signal, tune the hw?
Now it is very complicated to exactly identify where the proxmark has to be placed in the field.

iceman1001, 2019-07-12:
Which Proxmark3 device do you have?   Which HF antennas are you using?
On a RDV4, the HF simulation is quite good on both dual,  single and large HF antenna.

hornyd, 2019-07-15:
I have the proxmark3, RDV4.

[+] LF antenna: 68.83 V - 125.00 kHz
[+] LF antenna: 34.84 V - 134.00 kHz
[+] LF optimal: 69.25 V - 123.71 kHz
[+] LF antenna is OK

[+] HF antenna: 48.58 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.

[usb] pm3 --> hw status
#db# Memory
#db#   BIGBUF_SIZE.............40000
#db#   Available memory........40000
#db# Tracing
#db#   tracing ................1
#db#   traceLen ...............0
#db# Currently loaded FPGA image
#db#   mode.................... HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23
#db# Flash memory
#db#   Baudrate................24 MHz
#db#   Init....................OK
#db#   Memory size.............2 mbits / 256 kb
#db#   Unique ID...............0xD567A882A75C9425
#db# Smart card module (ISO 7816)
#db#   version.................v3.11
#db# LF Sampling config
#db#   [q] divisor.............95 ( 125 kHz )
#db#   [b] bps.................8
#db#   [d] decimation..........1
#db#   [a] averaging...........Yes
#db#   [t] trigger threshold...0
#db# LF T55XX config
#db#   [a] startgap............29*8 (232)
#db#   [b] writegap............17*8 (136)
#db#   [c] write_0.............15*8 (120)
#db#   [d] write_1.............47*8 (376)
#db#   [e] readgap.............15*8 (120)
#db# Transfer Speed
#db#   Sending packets to client...
#db#   Time elapsed............-1ms
#db#   Bytes transferred.......512
#db#   Transfer Speed PM3 -> Client = 0 bytes/s
#db# Various
#db#   DBGLEVEL................1
#db#   ToSendMax...............-1
#db#   ToSendBit...............0
#db#   ToSend BUFFERSIZE.......2308
#db# Installed StandAlone Mode
#db#   LF HID26 standalone - aka SamyRun (Samy Kamkar)
#db# Flash memory dictionary loaded

==========================================

I tried to use aluminium but it is still very sensitive compare to a card.

iceman1001, 2019-07-15:
Hrm,  well,  in offical repo they have some enhancements to the modulation functions.
If you test that, and see if its equally picky,  then we know its a software related issue.

hornyd, 2019-07-15:
Do you have some example how to change it and to use it?

iceman1001, 2019-07-15:
like downloading / compiling / flashing and test?

hornyd, 2019-07-15:
No sorry, I thought there was some ways to change the modulation manually or have some tuning. I tried already the official repo and I got the same. I need to put the antenna about 9cm distance and move around a couple of millimeters to get the HID VIX reader to read successfully.

iceman1001, 2019-07-15:
Bugger,  @0xFFFF   Maybe you have an idea.

iceman1001, 2019-07-15:
Which reader is HID VIX ??  Got a link?

hornyd, 2019-07-15:
This is the device I am using:
https://vixtechnology.com/media/1264/vix-corporate-brochure_27-september_reduced-size.pdf
I cannot find the technical specs from this reader cp 6100. It is NXP ISO14443 compliant (not emv)

iceman1001, 2019-07-15:
Vix... hm  they are the providers of the local transportation ticket system... 
HID VIX,  that would be mix of two company names. Either you have a VIX CP 6100 or you have a HID
CP indicate a card programmer,  which means a branding only.  Have you tried making your own antenna?

hornyd, 2019-07-16:
[image]
I found a way to manage the noise by putting aluminium and it seems to reduce instability by 90%. It works everytime now. So maybe it would be an improvement to be done on the Proxmark antenna

gtpy, 2019-07-16:
How does it look from the other side? Identical? Since I also have problems simulating HF, I would also like to try it with aluminium.

hornyd, 2019-07-16:
Yes, same, i made a U with Aluminium and put the pm3 inside. You can adjust the U to reduce up to 100% if you completely cover the pm3

gtpy, 2019-07-16:
Great. I’ll try it on my readers next evenings.

iceman1001, 2019-07-16:
you add the aluminumn to reduce noice???  Sounds more like you are reducing the communication between reader and pm3

iceman1001, 2019-09-13:
@doegox  just pushed some nice fixes for LF fsk simulation. If your reader was using ProxCard II, or LF..  You should definitely pull latest code and test again.

doegox, 2019-09-13:
Hmm well the thread is about HF, not LF...

iceman1001, 2019-09-13:
sorry,  mixed things up.

iceman1001, 2019-10-08:
Direct the signal a metal shield behind antenna. Someone wrote they got 1m reading range doing so. 
However your setup indicates the pm3 to be very close. 
if you are running Mifare simulation,   afterwards, can you post your trace list output?  And maybe share the trace file?

```
hf mf list
trace save hid_vx_mf_simulation.bin
```

Otherwise I don't think we can do much here,

hornyd, 2020-01-19:
Hi I am so sorry it took so long. I got into something else and lost track.

Now I am back on this project and I unfortunately deleted the old project from my machine. So I rebuild from scratch ProxSpace and the latest build of this git see below:

[ ARM ]
  bootrom: RRG/Iceman/master/release (git)
       os: RRG/Iceman/master/release (git)
  compiled with GCC 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

 [ FPGA ]
  LF image built for 2s30vq100 on 2020-01-12 at 15:31: 2
  HF image built for 2s30vq100 on 2020-01-12 at 15:31:16

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 279037 bytes (53%) Free: 245251 bytes (47%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

Now, the problem is that the Vix reader and the NFC ACR122 are not even able to read the Mifare Classic like before. It is stuck. I tried distance, aluminium, different readers, updated the firmware of the proxmark again but I am getting block to read from the block 0.

[usb] pm3 --> hf mf eload 4 hf-mf-560F59E1-data.eml
[+] loaded 4096 bytes from text file hf-mf-560F59E1-data.eml
[=] Copying to emulator memory
................................................................................................................................................................................................................................................................

[+] Loaded 256 blocks from file: hf-mf-560F59E1-data.eml

This image  hf-mf-560F59E1-data.eml was working with the build I was using in this issue. and not now anymore

[usb] pm3 --> hf mf sim t 4 u 560F59E1
[=] Mifare 4K  | 4 byte UID  56 0F 59 E1
[=] Options [ numreads: 0, flags: 1026 (0x402) ]
#db# Enforcing Mifare 4K ATQA/SAK
#db# 4B UID: 560f59e1
#db# ATQA  : 00 02
#db# SAK   : 18
[usb] pm3 -->

I saved the traces as attached.

hornyd, 2020-01-19:
Trace attached

iceman1001, 2020-01-19:
Looks like the sim doesn't quite manage to logg everything.
What that depends on doesn't make sense.   Your pm3 never answers the authentication command

Looks like  ok reader anti-col/select.
```
  432097096 |  432098152 | Rdr |26                                                                       |     | REQA
  432099324 |  432101692 | Tag |02  00                                                                   |     |
  432110502 |  432112966 | Rdr |93  20                                                                   |     | ANTICOLL
  432114138 |  432119962 | Tag |56  0f  59  e1  e1                                                       |     |
  432141476 |  432151940 | Rdr |93  70  56  0f  59  e1  e1  a4  3c                                       |  ok | SELECT_UID
  432153176 |  432156760 | Tag |18  37  cd
```

Bad trace but looks like reader is trying to autenticate block0 but missing stuff.
Either reader remembers uid for previous tries and go direct for 93 70 select
or its bad simulation.
```
  720380798 |  720383166 | Tag |02  00                                                                   |     |
  720396616 |  720407080 | Rdr |93  70  56  0f  59  e1  e1  a4  3c                                       |  ok | SELECT_UID
  720408316 |  720411900 | Tag |18  37  cd                                                               |     |
  720602404 |  720607108 | Rdr |60  00  f5  7b                                                           |  ok | AUTH-A(0)
```

hornyd, 2020-01-19:
Is there a way to get the 12 Dec version of the project? So I could compare the differences? I am not able to find it back. I used exactly the same hardware and as you saw on the image of Vix a bit up it was working awsome.

iceman1001, 2020-01-20:
you would just checkout the source code from that date and compile as normal.

hornyd, 2020-01-27:
I managed to rebuild the project properly but I did not succeed to restore the old build. Somehow, I did not managed to make it working.

However, with the latest project, checking the commands, indeed it cannot read the card header somehow

I load the image which worked before as below:

[usb] pm3 --> hf mf eload 4 hf-mf-560F59E1-data.eml
[+] loaded 4096 bytes from text file hf-mf-560F59E1-data.eml
[=] Copying to emulator memory
................................................................................................................................................................................................................................................................

[+] Loaded 256 blocks from file: hf-mf-560F59E1-data.eml
[usb] pm3 --> hf mf sim e t 4 v
[=] Mifare 4K  |  UID  N/A
[=] Options [ numreads: 0, flags: 1040 (0x410) ]
#db# Enforcing Mifare 4K ATQA/SAK
#db# 4B UID: 560f59e1
#db# ATQA  : 00 02
#db# SAK   : 18
[usb] pm3 -->

when I try to read from ACR reader or the VIX it just rejects the proxmark.

In order to identify the issue and how to make it working again, how can I debug or see some logs (not the APDU but more what proxmark project is doing), is there anything I could do to better point where is wrong?

hornyd, 2020-02-01:
@iceman1001  anything could help to identify the root cause? I will try to restore the December version and see if I can get it working then

iceman1001, 2020-02-04:
sorry, I am busy with Hitag stuff.

hornyd, 2020-02-04:
@iceman1001  OK no problem thx, in the meantime I will try to see whats wrong. We try to use your project part of our CI/CD for validator code test so we have to get it run for many test cases if possible.
I will come back to you if I find anything

iceman1001, 2020-02-12:
try again,  when looking into some hitag stuff,  someone showed me a problem with the current reading of adc.. which caused problems in the field detection of HF.

iceman1001, 2020-06-02:
Closing because of inactivity

closed 2020-06-02 by iceman1001