RfidResearchGroup/proxmark3 #255: hf_bog currently broken

doegox, 2019-07-16:
```
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: obj/fullimage.stage1.elf section `.bss' will not fit in region `ram'
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: section .commonarea VMA [000000000020ffe0,000000000020ffee] overlaps section .bss VMA [0000000000202718,0000000000210323]
/usr/lib/gcc/arm-none-eabi/7.3.1/../../../arm-none-eabi/bin/ld: region `ram' overflowed by 836 bytes
collect2: error: ld returned 1 exit status
```
I didn't investigate/bisect yet

doegox, 2019-07-16:
Bug appears between 0ea7fa7784f99f95a4629ac4619e70adeacb43fc and 7aabab2462144d7da3d92d60cb85a764f4ea07bf which is the merge of 996ed197fbfb9af8af6b8437adbe177e8d5a14db (remove_aes_c).
@slurdge could you look at it?
It seems mbedtls eats up more RAM than previous code...

doegox, 2019-07-16:
to test quickly:
```
make clean;make -j8 STANDALONE=HF_BOG os
```

slurdge, 2019-07-16:
Ah, I see. I compile with most of the package as thumb and not wit HF_BOG.
Will take a look.

cjbrigato, 2019-07-17:
#258 Please give back our 9k ram.

closed 2019-07-17 by iceman1001