RfidResearchGroup/proxmark3 #258: new aes implementation or anything that changed makes ram overflow really soon

cjbrigato, 2019-07-17:
**Describe the bug**
```
/usr/local/Caskroom/gcc-arm-embedded/6-2017-q2-update/gcc-arm-none-eabi-6-2017-q2-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: obj/fullimage.stage1.elf section `.bss' will not fit in region `ram'
/usr/local/Caskroom/gcc-arm-embedded/6-2017-q2-update/gcc-arm-none-eabi-6-2017-q2-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: section .commonarea VMA [000000000020ffe0,000000000020ffee] overlaps section .bss VMA [0000000000202130,0000000000210069]
/usr/local/Caskroom/gcc-arm-embedded/6-2017-q2-update/gcc-arm-none-eabi-6-2017-q2-update/bin/../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/bin/ld: region `ram' overflowed by 138 bytes
```

**To Reproduce**
Steps to reproduce the behavior:
* add any feature to armsrc
* enjoy the compile time
* don't enjoy the link time

**Expected behavior**
Before : 
```
arm-none-eabi-size obj/aes.o
   text	   data	    bss	    dec	    hex	filename
  15080	      0	      0	  15080	   3ae8	obj/aes.o
```
After : 
```
arm-none-eabi-size obj/aes.o
   text	   data	    bss	    dec	    hex	filename
  10355	      0	   8748	  19103	   4a9f	obj/aes.o
```

**Additional context**
Don't really know what's going on there but this 9k ram aes thing ain't have future alongside with our 40K bigbuff.

cjbrigato, 2019-07-17:
```
rm-none-eabi-nm -S -r --size-sort obj/aes.o | grep " b "
00000000 00000400 b RT3
00000000 00000400 b RT2
00000000 00000400 b RT1
00000000 00000400 b RT0
00000000 00000400 b FT3
00000000 00000400 b FT2
00000000 00000400 b FT1
00000000 00000400 b FT0
00000000 00000100 b RSb
00000000 00000100 b FSb
00000000 00000028 b RCON
00000000 00000004 b aes_init_done
```

cjbrigato, 2019-07-17:
Can we *AT LEAST* define some of the classics  ?
`MBEDTLS_AES_FEWER_TABLES` etc ?

cjbrigato, 2019-07-17:
with both MBEDTLS_AES_FEWER_TABLES and MBEDTLS_AES_ROM_TABLES
```
arm-none-eabi-size obj/aes.o
   text	   data	    bss	    dec	    hex	filename
  11983	      0	      0	  11983	   2ecf	obj/aes.o
```

iceman1001, 2019-07-17:
As always,  thanks bro!

closed 2019-07-17 by iceman1001

slurdge, 2019-07-17:
Shouldn't we able to have those 9k taken from BigBuf ? We would be able to reclaim the 2k taken by the tables in ROM. I'll take a look. Thanks @cjbrigato, nice catch (I was focused only on Flash usage)