RfidResearchGroup/proxmark3 #260: proxmark client exits status of 0 when serial port is unavailable.

videoman, 2019-07-17:
If you attempt to run a second proxmark client on the same serial port that is in use, you get 
"[!] ERROR: serial port /dev/ttyACM0  is claimed by another process"

But the exit status is "0" meaning that there was no error- so any other program that needs the exit status to identify an error will fail.

**To Reproduce**
Steps to reproduce the behavior:
1. run proxmark client and connect to hardware
2. run proxymark -c in another shell.
3. second program should have an exit status of >0, due to error, but doesn't
4. See error

**Expected behavior**
We should get a non-zero exit status to be POSIX compliant.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Raspberry Pi - Linux hostname01 4.19.57-v7+ #1244 SMP Thu Jul 4 18:45:25 BST 2019 armv7l GNU/Linux

iceman1001, 2019-07-17:
Good catch,   I never run the client against same serial port but with proxmark3.sh it will. 

Dunno if the second client should go into OFFLINE mode instead of exiting...

videoman, 2019-07-17:
I'm ok with it exiting, and giving a non-zero exit status. Gives the program trying to use it some useful feedback. May this should only be when using the "-c" option that it exits non-zero. Then I know something has gone wrong- EG: I left the client connected someplace else.

doegox, 2019-07-17:
@iceman1001 we could generalize it for other port issues, like if there is no Pm3 connected, but you gave a port to the client, the client should exit. (if someone gave a port info, he probably doesn't intend to work offline!)

Only when a disconnection occurs in the middle of a session, we don't exit but fall back to offline mode.
Shall we move in that direction ?

merlokk, 2019-07-17:
if you gave a port and gave option to wait `-w` it also waits and then exit if not found

doegox, 2019-07-17:
yes today it goes offline after timeout and with the change it would exit(1), nothing to handle specifically for the "-w"

iceman1001, 2019-07-17:
I like the direction @doegox  suggests.  It will solve @videoman original problem and it will solve the aspects Philippe mention.

doegox, 2019-07-17:
259c730bb06f0c45ae8050ea281507c5b390094e should be enough to satisfy everybody, please test :)

iceman1001, 2019-07-17:
I think that should cover OP's issue and it follows the nice retun values we want to use now.

doegox, 2019-07-17:
ok let's close it. @videoman please open again if needed.

closed 2019-07-17 by doegox

iceman1001, 2019-07-17:
Great fix @doegox !