RfidResearchGroup/proxmark3 #265: HF failing after LF

doegox, 2019-07-17:
`./proxmark3.sh -c "lf read s d 100;hf tune;hf tune"`

```
pm3 --> lf read
[...]

pm3 --> hf tune          
[+] Measuring HF antenna, click button or press Enter to exit          
[!] Timeout while waiting for Proxmark HF initialization, aborting          

pm3 --> hf tune          
[+] Measuring HF antenna, click button or press Enter to exit          
(ok)
```

First call to `hf tune` after a `lf read` fails, second call works fine.

iceman1001, 2019-07-28:
Confirmed.

iceman1001, 2019-07-28:
Increasing timeouts solved it on my setup.

closed 2019-07-28 by iceman1001

doegox, 2019-08-07:
The problem is wider, any command requiring a swap of FPGA images will get an increased delay every time we go LF<>HF.

E.g.
```
[usb] pm3 --> lf read
[...]
[usb] pm3 --> hf search
[=] Checking for known tags...
[!] timeout while waiting for reply.          
```

I implemented a mechanism of waiting time extension request where the Pm3 indicates to the client that it will take exceptionally more time than usual.
Usage: 
https://github.com/RfidResearchGroup/proxmark3/blob/bc28eb04e437e65c94f86e94c4da7b0ef58cf92e/armsrc/fpgaloader.c#L398-L399

@cjbrigato maybe you could add a send_wtx() into spiffs_gc with a proper timing? how much can take the garbage collector in worst cases ?

iceman1001, 2019-08-12:
I still don't understand why we get an _increasing_ timeout instead of a fixed timeout. 

Your workaround solves some issues,

doegox, 2019-08-12:
we can discuss it if it's not clear.

commit 00350db2 (2019-07-28): Chg: `hf tune` - increase timeout vaules. fix #265