RfidResearchGroup/proxmark3 #268: hf mf hardnested test not works in offline mode

merlokk, 2019-07-18:
**Describe the bug**
`hf mf hardnested t 1 000000000000` not works in offline mode.

```
Test [hf mf hardnested] job: bash -lc "cd ~/client;./proxmark3 -c 'hf mf hardnested t 1 000000000000'"
[+] execute command from commandline: hf mf hardnested t 1 000000000000 
          
[=] Running in OFFLINE mode. Check "proxmark3.exe -h" if it's not what you want.
          
pm3 --> hf mf hardnested t 1 000000000000          
[!] This command is not available in this mode          
Sending bytes to proxmark failed - offline
```

**To Reproduce**
start appveyour ci

merlokk, 2019-07-18:
```
[offline] pm3 --> hf mf hardnested t 1 000000000000
[!] This command is not available in this mode
```

doegox, 2019-07-18:
indeed, thanks, we'll fix

doegox, 2019-07-18:
temp fix for now, till we find a better way to isolate online & offline components of such mixed commands.
But the build failed even earlier with some weird Bash cygheap?? https://ci.appveyor.com/project/RfidResearchGroup/proxmark3/builds/26087684
The build just before went well till the hardnested test.

iceman1001, 2019-08-28:
time to close?

closed 2019-08-28 by doegox