RfidResearchGroup/proxmark3 #271: "lf hid clone" for 37-bit cards not working after writing 26-bit cards.

videoman, 2019-07-18:
**Describe the bug**
When I attempt to write a 37-bit ID to a t5577 card, it only writes a 26-bit ID to the card.

From the  "lf hid clone 1029a0f4d2" 

**To Reproduce**
Run the following commands from inside the proxmark3 client software connected online with a t5577 card sitting on top of the 125k coil:
First 
"lf hid clone 2004FEF4D3" will write a 26-bit card.
"lf hid read" will read it.
"lf t55xx wipe"
"lf search" to confirm no card data
"lf hid clone 1029a0f4d2" should write a 37-bit card id.
"lf hid read" shows a card with only 26-bits was written.

**Expected behavior**
The command should write a 37-bit ID to the card. 

After running "lf hid sim 1029a0f4d2", I was then able to write that 37-bit card ID to the t5577 card.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Raspbian - Linux hostname-01 4.19.57-v7+ #1244 SMP Thu Jul 4 18:45:25 BST 2019 armv7l GNU/Linux
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ]

 [ CLIENT ]
  client: RRG/Iceman
  compiled with GCC 8.3.0

 [ PROXMARK RDV4 ]
  external flash:                  present
  smartcard reader:                present

 [ PROXMARK RDV4 Extras ]
  FPC USART for BT add-on support: absent

 [ ARM ]
  bootrom: RRG/Iceman/master/e51b4f0f 2019-07-11 20:25:19
       os: RRG/Iceman/master/8cb35893 2019-07-16 21:19:27
  compiled with GCC 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 [ FPGA ]
  LF image built for 2s30vq100 on 2019/ 4/18 at  9:35:32
  HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23

 [ Hardware ]
  --= uC: AT91SAM7S512 Rev B
  --= Embedded Processor: ARM7TDMI
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 239472 bytes (46%) Free: 284816 bytes (54%)
  --= Second Nonvolatile Program Memory Size: None
  --= Internal SRAM Size: 64K bytes
  --= Architecture Identifier: AT91SAM7Sxx Series
  --= Nonvolatile Program Memory Type: Embedded Flash Memory

 - hw status
[usb] pm3 --> hw status
#db# Memory
#db#   BIGBUF_SIZE.............40000
#db#   Available memory........40000
#db# Tracing
#db#   tracing ................0
#db#   traceLen ...............0
#db# Currently loaded FPGA image
#db#   mode.................... LF image built for 2s30vq100 on 2019/ 4/18 at  9:35:32
#db# Flash memory
#db#   Baudrate................24 MHz
#db#   Init....................OK
#db#   Memory size.............2 mbits / 256 kb
#db#   Unique ID...............0xD567A882A77FB525
#db# Smart card module (ISO 7816)
#db#   version.................v2.06
#db# LF Sampling config
#db#   [q] divisor.............95 ( 125 kHz )
#db#   [b] bps.................8
#db#   [d] decimation..........1
#db#   [a] averaging...........Yes
#db#   [t] trigger threshold...0
#db# LF T55XX config
#db#   [a] startgap............29*8 (232)
#db#   [b] writegap............17*8 (136)
#db#   [c] write_0.............15*8 (120)
#db#   [d] write_1.............47*8 (376)
#db#   [e] readgap.............15*8 (120)
#db# Transfer Speed
#db#   Sending packets to client...
#db#   Time elapsed............500ms
#db#   Bytes transferred.......108032
#db#   Transfer Speed PM3 -> Client = 216064 bytes/s
#db# Various
#db#   DBGLEVEL................1
#db#   ToSendMax...............-1
#db#   ToSendBit...............0
#db#   ToSend BUFFERSIZE.......2308
#db# Installed StandAlone Mode
#db#   LF HID26 standalone - aka SamyRun (Samy Kamkar)
#db# Flash memory dictionary loaded

 - data tune
[usb] pm3 --> data tune

[=] Measuring antenna characteristics, please wait...

...

[+] LF antenna: 42.06 V - 125.00 kHz
[+] LF antenna: 49.71 V - 134.00 kHz
[+] LF optimal: 52.68 V - 130.43 kHz
[+] LF antenna is OK

[+] HF antenna: 45.32 V - 13.56 MHz
[+] HF antenna is OK

[+] Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.

**Additional context**
It seems that after a reset of the modes/proxmark it was then able to write the 37-bit ID to the t5577 card. Should I just be doing "hw reset" before each write? Seems like its not switching between bit lengths, or someplace in the code its got the bit length set, and needs a reset to change bit lengths for writing.

mwalker33, 2019-09-14:
Is this still an issue.
I run your test on the latest and could not repeat the bug!
```
[usb] pm3 --> lf hid clone 2004FEF4D3
[=] Preparing to clone HID tag with ID 2004fef4d3

[usb] pm3 --> lf hid read
[+] HID Prox TAG ID: 2004fef4d3 (31337) - Format Len: 26bit - OEM: 000 - FC: 127 - Card: 31337

[usb] pm3 --> lf t55 wipe

[=] Beginning Wipe of a T55xx tag (assuming the tag is not password protected)
.... wipe output

[usb] pm3 --> lf sea
[=] NOTE: some demods output possible binary
[=] if it finds something that looks like a tag
[=] False Positives ARE possible
[=]
[=] Checking for known tags...

[-] No known 125/134 kHz tags found!

[usb] pm3 --> lf hid clone 1029a0f4d2
[=] Preparing to clone HID tag with ID 1029a0f4d2

[usb] pm3 --> lf hid read
[+] HID Prox TAG ID: 1029a0f4d2 (31337) - Format Len: 37bit - OEM: 000 - FC: 666 - Card: 31337
```

iceman1001, 2019-09-16:
Besides from pulling latest and flash-all on your device,  
I can't replicate your problem.    Closing this one.

[image]

closed 2019-09-16 by iceman1001