RfidResearchGroup/proxmark3 #273: "lf hid wiegand" for 37-bit card off by one.

videoman, 2019-07-18:
**Describe the bug**
When using the "lf hid wiegand" command the 37-bit output is off.

**To Reproduce**
Run:
"lf hid wiegand 0 666 31337" 
The 37-bit weigand shows: "1029A0F4D3"

**Expected behavior**
The 37-bit weigand hex should show: "1029a0f4d2"

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: Raspbian - Linux hostname 1 4.19.57-v7+ #1244 SMP Thu Jul 4 18:45:25 BST 2019 armv7l GNU/Linux
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 - hw status
 - data tune

**Additional context**
The other card number for the 37-bit ID came from the mainline proxmark3 code using “lf hid encode H10304 f 666 c 31337”

videoman, 2019-07-18:
[usb] pm3 --> lf hid read
[+] HID Prox TAG ID: 1029a0f4d2 (31337) - Format Len: 37bit - OEM: 000 - FC: 666 - Card: 31337
[usb] pm3 -->

ViRb3, 2019-09-24:
Related to #240?

iceman1001, 2019-09-24:
try the new wiegand command (straight from offical repo)
``` 
wiegand list
wiegand encode
wiegand decode
```

I think this issue is sorted.  @videoman  just needs to verify it.

videoman, 2019-09-24:
The 37bit encoding works now.

closed 2019-09-24 by videoman