RfidResearchGroup/proxmark3 #280: village of the damned proxmarks

doegox, 2019-07-19:
I see a path to a wall of bricked Proxmarks...

Soon, people will try to flash a large OS image (e.g. once  @cjbrigato SPIFFS is merged) with latest flash code **but** old bootloader.
* It happens if you flash only the OS image.
* It happens even if you flash both bootloader and OS image in one go! Because it's still the old bootloader which will
  * report happily the 512k support (unless it's a very old bootloader, like 1 month old...)
  * flash the new bootloader
  * flash the large FW and wrap addresses after 256k -> in the bootrom area

Right now, the only was to prevent such bricking is to:
* Flash the new bootloader
* Reboot the PM3
* Flash the the new OS image

PoC:
```
# bootloader from yesterday
git checkout 4e9e7d6da004140c7c4a2ed2c3f560ac9a4d6763
make clean
make -j8
./flash-all.sh 
git checkout master
# inflate the firmware > 256k, I won't tell how
make clean
make -j8
./flash-all.sh 
# => bricked Proxmark!
```

[image]

What are our options?
* add some info on new bootloader like "yes I really support 512k" and the flasher accepting to flash large fw only if bootloader reports that new info?

cc @slurdge @cjbrigato @iceman1001

iceman1001, 2019-07-19:
You opened pandoras box,   brickjuly is in making.   It was the days before DEFCON...

This one:
Maybe only flash 512, if bootloader reports support for it.     That would be the least to expect.

doegox, 2019-07-19:
Right now (old) bootloader already reports 512k with CHIP_INFO so we need to add some                    DEVICE_INFO_FLAG_SUPPORTS_LARGE_IMAGES;

slurdge, 2019-07-19:
We should add versionning to the bootloader ASAP as the same problem can happend in the future.
As a quick fix, we can (for a while) instruct the board to reset _again_ after bootloader upload. That way, we always ensure the latest bootloader is present before flashing image.

The problem also happens if you flash only large image and someone flashed the older new bootloader before.

However: the timeframe to have the support for 512k with CHIP_INFO and bricking is quite small.

slurdge, 2019-07-19:
If we add more&more flags it will complicate the workflow (more bugs) and will run out of bits. I would suggest a new message that reports the version of the bootloader. From then, we can add all the logic inside flasher (with if > V1 then XX_IS_AVAILABLE etc.)

doegox, 2019-07-19:
ok I support the idea of a bootloader version, updated manually in bootloader and flasher when such big changes are implemented

slurdge, 2019-07-19:
Yes :-) For now if you want to remove all possibilities of bricks, you can force push an history rewrite that forces the available memory to 256. Nobody will be able to flash more than 256 unless they patch their flasher.c file.

doegox, 2019-07-19:
no, no, don't mess up the master branch :) till the FW is below 256k, we've a bit of time to propose a fixed flasher

doegox, 2019-07-19:
https://github.com/RfidResearchGroup/proxmark3/tree/bl_versionning

doegox, 2019-07-20:
Now on master :)

closed 2019-07-20 by doegox