RfidResearchGroup/proxmark3 #285: [post merge ] Spiffs continuation/documentation/impact

cjbrigato, 2019-07-23:
This issue follow the merge of SPIFFS implementation from #257 and discuss post merge actions and eventual improvement on this topic. 

What is needed right now : 
1. DOCUMENTATION
2. Patch to hf_bog to rely on spiffs and not on overwriting it the old way
3.?

doegox, 2019-07-23:
- [ ] use PM3_SUCCESS and alike when it makes sense

cjbrigato, 2019-07-23:
#286 for HF_BOG

cjbrigato, 2019-07-23:
@doegox please mind editing my own op post for easy access to the list and status :)

doegox, 2019-07-23:
indeed I didn't realize I had this super power :)

cjbrigato, 2019-07-23:
Is this sarcasm or anything I did not get ? I'm not sure haha.
I saw "Collaborator" rights on the repo in your case I assumed you had such power, just like for editing PR when authorized.

doegox, 2019-07-24:
honestly, I forgot about being able to edit others stuffs. It sounds so intrusive to me, I wonder why it's allowed. Imagine a forum like that...

cjbrigato, 2019-07-24:
When this is asked as an actual usefull tool, things are different. such power are made for the one who forget them like you :D

doegox, 2019-07-24:
i'll clean away this discussion, there'll be enough important stuff to discuss in this issue ;)

doegox, 2019-07-24:
FTR now Makefile properly detects incompatible PLATFORM & STANDALONE (e.g. HF_COLIN requiring Flash on a non RDV4)

iceman1001, 2019-07-24:
nice list,   good one.

cjbrigato, 2019-07-31:
spiffs now also permit json reading and parsing on device-side. See https://github.com/RfidResearchGroup/proxmark3/pull/299

iceman1001, 2025-02-20:
Dictionaries has been converted to files. 
the last sector has been freed up for SPIFFS. 
Dynamic identification of SPIFF memory which is larger than the original 256kb on RDV4 has been implemented.

closed 2025-06-08 by iceman1001