RfidResearchGroup/proxmark3 #290: [idea] Add unbriking guid using Openocd and raspberry pi zero

eivbsmed, 2019-07-25:
**Is your feature request related to a problem? Please describe.**

I bout a proxymark 3 RDV 4.01 and bricked it following the the getting documentation running flash-all.sh The online store i got it from has so far been not been willing to give a good answer on support. This is the first time having to do some thing like this and would love to make it whit out messing up the device even more.

**Describe the solution you'd like**
Add some instructions to the folder proxmark3/tools/jtag_openocd/ on how it should be used and wired using the interface descriptions in the default install for raspberry pi.

I have started on a guide written for me to trace my steps that can be built on.

---

Using etcher to write 2017-07-10-raspbina-buster-light.img to a memory card

Before adding it to PI Zero W or any other PI add to boot on memory card:

add `ssh` file to boot
add `wpa_supplicant.conf`

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
network={
    ssid="YOUR_SSID"
    psk="YOUR_WIFI_PASSWORD"
    key_mgmt=WPA-PSK
}
```

Boot the pi
Some how find the ip and `ssh pi@<ip>`
You should know the password

On pi: 
---

sudo raspi-config
- expand file system
- set memory splitt to 16 there will be no video

`sudo apt update`
`sudo apt upgrade`
`sudo apt install tmux`

`tmux` continue here so all is not lost if nettwork is interupted. Compiling takes some time

Next steps are from https://learn.adafruit.com/programming-microcontrollers-using-openocd-on-raspberry-pi/compiling-openocd

`sudo apt-get install git autoconf libtool make pkg-config libusb-1.0-0 libusb-1.0-0-dev`
`mkdir repos`
`cd repos`
`git clone http://openocd.zylin.com/openocd`
`cd openocd`
`./bootstrap`
`./configure --enable-sysfsgpio --enable-bcm2835gpio`
`make`
`sudo make install`

Check that it was installed
`openocd --version`
```
Open On-Chip Debugger 0.10.0+dev-00921-g263deb38 (2019-07-22-16:13)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
```

`cat  /usr/local/share/openocd/scripts/interface/sysfsgpio-raspberrypi.cfg`
```
#
# Config for using RaspberryPi's expansion header
#
# This is best used with a fast enough buffer but also
# is suitable for direct connection if the target voltage
# matches RPi's 3.3V
#
# Do not forget the GND connection, pin 6 of the expansion header.
#

interface sysfsgpio

# Each of the JTAG lines need a gpio number set: tck tms tdi tdo
# Header pin numbers: 23 22 19 21
sysfsgpio_jtag_nums 11 25 10 9

# Each of the SWD lines need a gpio number set: swclk swdio
# Header pin numbers: 23 22
sysfsgpio_swd_nums 11 25

# If you define trst or srst, use appropriate reset_config
# Header pin numbers: TRST - 26, SRST - 18

# sysfsgpio_trst_num 7
# reset_config trst_only

# sysfsgpio_srst_num 24
# reset_config srst_only srst_push_pull

# or if you have both connected,
# reset_config trst_and_srst srst_push_pull
```

The next steps Im not sure about so im hoping some one will fill them in

doegox, 2019-07-25:
Thanks for your feedback! Some remarks / questions:

* Do you know how the pm3 got bricked ? Because of ModemManager?
* Yes it's in our plans to write some unified jtag doc in the repo, even if some very good info exist already elsewhere
* When I did jtag on RPi, `apt-get install openocd` was good enough. And I dropped https://github.com/RfidResearchGroup/proxmark3/blob/master/tools/jtag_openocd/interface-raspberrypi.cfg in the repo for that purpose

eivbsmed, 2019-07-25:
Could have been modem manager. I did not pay attention to it because i did not thing it was some ting i got. Looking at it now it could have been. This is the output of the fatal flash.

```
⋊> ~/R/R/proxmark3 on master ◦ ./flash-all.sh
Waiting for Proxmark to appear...
[+] Waiting for Proxmark3 to appear on /dev/pm3-0          
.Found          
[+] Entering bootloader...          
[+] (Press and release the button only to abort )          
[+] Waiting for Proxmark3 to appear on /dev/pm3-0          
. Found          
[=] End of flash: 0x00140000          
Available memory on this board: UNKNOWN
         
[!!] Note: Your bootloader does not understand the new CHIP_INFO command          
[!!] It is recommended that you update your bootloader
         
[+] Loading ELF file bootrom/obj/bootrom.elf          
[+] Loading usable ELF segments:          
[+]    0 : V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94          
[+]    1 : V 0x00200000 P 0x00100200 (0x00000de0->0x00000de0) [R X] @0x298          
         
[+] Loading ELF file armsrc/obj/fullimage.elf          
[+] Loading usable ELF segments:          
[+]    0 : V 0x00102000 P 0x00102000 (0x00038218->0x00038218) [R X] @0x94          
[+]    1 : V 0x00200000 P 0x0013a218 (0x00001334->0x00001334) [RW ] @0x382ac          
[=] Note: Extending previous segment from 0x38218 to 0x3954c bytes          
         
         
[+] Flashing...
         
[+] Writing segments for file: bootrom/obj/bootrom.elf          
[+]  0x00100000..0x001001ff [0x200 / 1 blocks]          
. OK          
[+]  0x00100200..0x00100fdf [0xde0 / 7 blocks]          
.....[!] Received packet OLD frame with payload too short? 118/534          
[!] Received packet OLD frame with payload too short? 22/534          
[=] You can cancel this operation by pressing the pm3 button
```
At that point all the light on the proxymark where on and the cup on the pc went to 100% and stayed there for 1 hour until i had to kill it. Plugging it in now sets all the light to on. 
Do you know if wiring the JTAG can hurt the device? If you know on some good guides on this using raspberry pi is out there let med know. I'm a little confused whit all the variation of the proxymark devices and how they compare.

doegox, 2019-07-25:
From your logs I'd say the bootrom was properly flashed but not the OS.
Did you try pressing the button while plugging the device in, to force bootloader mode ? (and 2 LEDs turn on), then flash only the OS image.
But first of all, make sure ModemManager is not there anymore, see https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Linux-Installation-Instructions.md#check-modemmanager

eivbsmed, 2019-07-25:
Don't looks like it working. All blue power lights turn and A-D once it get power even if the button is pressed or not...

doegox, 2019-07-25:
Hmm ok.

> Do you know if wiring the JTAG can hurt the device? If you know on some good guides on this using raspberry pi is out there let med know.

You're pretty close, with openocd already installed.
For wiring, see https://github.com/RfidResearchGroup/proxmark3/blob/master/tools/jtag_openocd/interface-raspberrypi.cfg#L4-L13 and **don't connect the Pm3 on USB**.
Then as I said, prepare a `openocd_configuration` file based on `openocd_configuration.sample` and use `openocd_flash_recovery.sh`, it should be enough.

iceman1001, 2019-09-01:
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/jtag_notes.md

closed 2019-10-11 by iceman1001