RfidResearchGroup/proxmark3 #293: proxmark3 rdv4 red led lightup

johnny9900, 2019-07-27:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 - hw status
 - data tune

**Additional context**
Add any other context about the problem here.

johnny9900, 2019-07-27:
i bot a new proxmark3 rdv4 from lab401. mounted the standalone kit. installed the rfidresearchgroup software on osx 10.12. yet i still cannot connect to the pm3 regardless using the /dev/cu.usbxxx or /dev/cu.pm3-bt ... it keep saying 

[=] Using UART port /dev/cu.usbmodemicema1           
unknown command:: 0x61334d50[!!] ERROR: cannot communicate with the Proxmark

[=] Using UART port /dev/cu.PM3_RDV40-DevB           
[!!] ERROR: cannot communicate with the Proxmark

please help me.. thanks..

iceman1001, 2019-07-27:
hard to tell what you did wrong. 

Did you follow the instructions in the BT manual?

you find it here.
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/bt_manual_v10.md

johnny9900, 2019-07-27:
may i know what is the RED led meant for? it is permanently ON.
i cannot attempt proxmark3 connection as those 2 error kept throwing me out to mac prompt..

iceman1001, 2019-07-27:
Looks like your device is not running the latest firmware and client. 
Did you compile and flash like the instructions says?

johnny9900, 2019-07-27:
udo proxmark3-flasher /dev/tty.usbmodemicema1 -b /usr/local/share/firmware/bootrom.elf /usr/local/share/firmware/fullimage.elf
[+] Waiting for Proxmark3 to appear on /dev/tty.usbmodemicema1           
.Found           
[+] Entering bootloader...           
[+] (Press and release the button only to abort )          
[+] Waiting for Proxmark3 to appear on /dev/tty.usbmodemicema1           
. Found           
[!!] Note: Your bootloader does not understand the new CMD_BL_VERSION command           
[!!] It is recommended that you first update your bootloader alone,           
[!!] reboot the Proxmark3 then only update the main firmware 
          
Available memory on this board: UNKNOWN 
          
[!!] Note: Your bootloader does not understand the new CHIP_INFO command           
[!!] It is recommended that you first update your bootloader alone,           
[!!] reboot the Proxmark3 then only update the main firmware 
          
[=] Permitted flash range: 0x00100000-0x00140000          
[!!] Note: Your bootloader does not understand the new START_FLASH command           
[!!] It is recommended that you first update your bootloader alone,           
[!!] reboot the Proxmark3 then only update the main firmware 
          
[+] Loading ELF file /usr/local/share/firmware/bootrom.elf           
[+] Loading usable ELF segments:          
[+]    0 : V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94          
[+]    1 : V 0x00200000 P 0x00100200 (0x00000e30->0x00000e30) [R X] @0x298          
          
[+] Loading ELF file /usr/local/share/firmware/fullimage.elf           
[+] Loading usable ELF segments:          
[+]    0 : V 0x00102000 P 0x00102000 (0x0003e6b0->0x0003e6b0) [R X] @0x94          
[!!] Error: PHDR is not contained in Flash          
[!!] Aborted on error.

johnny9900, 2019-07-27:
what can be wrong?

johnny9900, 2019-07-27:
i flash with bootloader only sucessfully but still this error comes out:

./proxmark3 /dev/cu.usbmodemicema1 

          
██████╗ ███╗   ███╗ ████╗      ...iceman fork          
██╔══██╗████╗ ████║   ══█║       ...dedicated to RDV40           
██████╔╝██╔████╔██║ ████╔╝           
██╔═══╝ ██║╚██╔╝██║   ══█║     iceman@icesql.net          
██║     ██║ ╚═╝ ██║ ████╔╝    https://github.com/rfidresearchgroup/proxmark3/          
╚═╝     ╚═╝     ╚═╝ ╚═══╝  pre-release v4.0          

Support iceman on patreon,   https://www.patreon.com/iceman1001/          

          
[=] Using UART port /dev/cu.usbmodemicema1           
unknown command:: 0x61334d50

johnny9900, 2019-07-27:
thanks. it works now after turning off and on pm3 again. multiple times.

iceman1001, 2019-07-27:
Good!

closed 2019-07-27 by iceman1001