RfidResearchGroup/proxmark3 #294: bluetooth PM3 unable to connect

johnny9900, 2019-07-27:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
 - hw status
 - data tune

**Additional context**
Add any other context about the problem here.

johnny9900, 2019-07-27:
despite reflashing the pm3, the standalone device is still not able to connect via bluetooth.

./proxmark3 /dev/cu.PM3_RDV40-DevB 

          
██████╗ ███╗   ███╗ ████╗      ...iceman fork          
██╔══██╗████╗ ████║   ══█║       ...dedicated to RDV40           
██████╔╝██╔████╔██║ ████╔╝           
██╔═══╝ ██║╚██╔╝██║   ══█║     iceman@icesql.net          
██║     ██║ ╚═╝ ██║ ████╔╝    https://github.com/rfidresearchgroup/proxmark3/          
╚═╝     ╚═╝     ╚═╝ ╚═══╝  pre-release v4.0          

Support iceman on patreon,   https://www.patreon.com/iceman1001/          

          
[=] Using UART port /dev/cu.PM3_RDV40-DevB           
[!!] ERROR: cannot communicate with the Proxmark

iceman1001, 2019-07-27:
What is the contents of your Makefile.platfrom file?

iceman1001, 2019-08-14:
@johnny9900   If you filled in the issue report I wouldn't have to ask this,  but 

```
    OS: [e.g. iOS]
    inside proxmark3 client run the following commands and paste the output here.
    hw version
    hw status
    data tune

```

yezhuang, 2019-08-17:
Having the same issue. 

OS: MacOS Mojave 10.14.6

[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ] 
          

 [ CLIENT ]          
  client: RRG/Iceman          
  compiled with Clang/LLVM 4.2.1 Compatible Apple LLVM 10.0.1 (clang-1001.0.46.4)          

 [ PROXMARK RDV4 ]          
  external flash:                  present           
  smartcard reader:                present           

 [ PROXMARK RDV4 Extras ]          
  FPC USART for BT add-on support: absent           
          
 [ ARM ]
  bootrom: RRG/Iceman// 2019-08-16 20:15:35
       os: RRG/Iceman// 2019-08-16 20:15:53
  compiled with GCC 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]

 [ FPGA ]
  LF image built for 2s30vq100 on 2019/ 7/31 at 15:57:16
  HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 271172 bytes (52%) Free: 253116 bytes (48%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory          

          
[usb] pm3 --> hw status
#db# Memory           
#db#   BIGBUF_SIZE.............40000          
#db#   Available memory........40000          
#db# Tracing           
#db#   tracing ................0          
#db#   traceLen ...............779          
#db# Currently loaded FPGA image           
#db#   mode.................... HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23          
#db# Flash memory           
#db#   Baudrate................24 MHz           
#db#   Init....................OK           
#db#   Memory size.............2 mbits / 256 kb           
#db#   Unique ID...............0xD567A882A7438B25          
#db# Smart card module (ISO 7816)           
#db#   version.................v3.10           
#db# LF Sampling config           
#db#   [q] divisor.............95 ( 125 kHz )          
#db#   [b] bps.................8          
#db#   [d] decimation..........1          
#db#   [a] averaging...........Yes          
#db#   [t] trigger threshold...0          
#db# LF T55XX config           
#db#            [r]               [a]   [b]   [c]   [d]   [e]   [f]   [g]          
#db#            mode            |start|write|write|write| read|write|write          
#db#                            | gap | gap |  0  |  1  | gap |  2  |  3          
#db# ---------------------------+-----+-----+-----+-----+-----+-----+------          
#db# fixed bit length (default) |  29 |  17 |  15 |  47 |  15 | N/A | N/A |           
#db#     long leading reference |  31 |  20 |  18 |  50 |  15 | N/A | N/A |           
#db#               leading zero |  31 |  20 |  18 |  40 |  15 | N/A | N/A |           
#db#    1 of 4 coding reference |  29 |  17 |  15 |  31 |  15 |  47 |  63 |           
#db#           
#db# Transfer Speed           
#db#   Sending packets to client...          
#db#   Time elapsed............500ms          
#db#   Bytes transferred.......337408          
#db#   Transfer Speed PM3 -> Client = 674816 bytes/s          
#db# Various           
#db#   DBGLEVEL................1          
#db#   ToSendMax...............72          
#db#   ToSendBit...............8          
#db#   ToSend BUFFERSIZE.......2308          
#db#   Slow clock..............31550 Hz          
#db# Installed StandAlone Mode           
#db#   LF HID26 standalone - aka SamyRun (Samy Kamkar)          
#db# Flash memory dictionary loaded           
[usb] pm3 --> hw tune
          
[=] Measuring antenna characteristics, please wait...
          
[=] You can cancel this operation by pressing the pm3 button          
..
          
[+] LF antenna: 70.95 V - 125.00 kHz          
[+] LF antenna: 41.92 V - 134.00 kHz          
[+] LF optimal: 71.80 V - 126.32 kHz          
[+] LF antenna is OK  
          
[+] HF antenna: 47.59 V - 13.56 MHz          
[+] HF antenna is OK           
          
[+] Displaying LF tuning graph. Divisor 89 is 134kHz, 95 is 125kHz.
          
No GUI in this build!
[usb] pm3 -->

iceman1001, 2019-08-17:
You haven't compiled the firmware with support for BT.
```
[ PROXMARK RDV4 Extras ]
FPC USART for BT add-on support: absent
```

The homebrew repo readme would be a good start for you to read.
https://github.com/RfidResearchGroup/homebrew-proxmark3

If you have more issues on OSX,  there is some info in the BT manual.
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/bt_manual_v10.md

yezhuang, 2019-08-19:
Thanks for your reply. brew install --with-blueshark proxmark3 requires me pull the HEAD repo but the latest repo is having issue with recognizing the RDV4 port on MacOS. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

iceman1001, 2019-08-20:
Nice that its solved.

closed 2019-08-20 by iceman1001