RfidResearchGroup/proxmark3 #295: better error messages when flashing on outdated...

iceman1001, 2019-07-28:
The exit message of PHDR is not contained in the Flash is troublesome.
Even though the messages is absolutely correct, they are not user friendly.

[image]

iceman1001, 2019-07-28:
@slurdge   What do you think?

slurdge, 2019-07-28:
I somewhat agree. There's however a "Available Memory" and "Permitted flash range" beforehand. Since proxmark3 is not an entry-level device (IMO), user should be smart enough to either Google it and deduce her image is too big.
I mean, when running the proxmark3 commands, messages are even more cryptic....
Still, we can/should improve the situation by either changing this message, or, better if you ask me, put an explanation on the return error such as:
"Aborted on error. Common causes are:
 - Image is too big for your board ("PHDR is not contained"). Please check your board size and compilation options
 - Image is corrupted. Try to make clean and rebuild.

My 2c

iceman1001, 2019-07-28:
I like that.  PR?

doegox, 2019-07-28:
What about simply "image is not fitting in permitted Flash range" ?

doegox, 2019-07-28:
About guidance "when flashing on outdated bootrom", there are twice red paragraphs indicating user should first update his bootrom...

mwalker33, 2019-07-29:
Just an quick comment.
When I updated, I noticed and read the "red" text first of got focesed on that rather then my actual message (could not find file) at the bottom.
On Windows (at least on mine) there is no text color and I still saw that first, so I think its clear that the bootloader should be updated first.
So, while this still may be a little cryptic, and just an idea.

[!!] Aborted on error.  Please address any issues above and try again.

To me that says:  It went wrong, the issue has been reported above.  I need to fix something, while at the some time giving the tech (PHDR) an non tech  (Fix the boot loader) responses.

If the tech output will help identifiy an end users issue (e.g. via forum) then I think it needs to be there somewhere.

Just a thought.

slurdge, 2019-07-29:
@mwalker33 Yes that's my understanding as well. I'll try to make a slightly better PR, but I'm a bit busy right now :)

iceman1001, 2019-07-29:
maybe its just need two lines of "========================================="
one above / one below the red errors to emphasize it on non-color systems

doegox, 2019-08-21:
can we close it now?

closed 2019-08-21 by iceman1001