RfidResearchGroup/proxmark3 #296: Add raw HF signal plotting

Fl0-0, 2019-07-29:
A great feature added by @pwpiwi  on official PM3 repo:

https://github.com/Proxmark/proxmark3/pull/786

iceman1001, 2019-07-29:
What do ppl actually use that trace for?   I seen the commit,  I never heard anyone use it and for ppl who is actually interested in the signal uses a scope. 
If its for trying to see if signal vs demodulated data matches?  I fail to see that commands purpose

doegox, 2019-07-29:
You can't see anything with a scope, you'll only see the 13.56 MHz. So you want to see the analog signal behind the 13.56 modulation

iceman1001, 2019-07-29:
You mean that you don't see anything on the dedicated DEBUG pin TP7 on the fpga?  Strange,  ..
well not on a RDV4 since it doesn't have testpoint 7,   but you can still hook up your scope on the fpga pin...

doegox, 2019-07-29:
I didn't know about that pin. Anyway, the functionality might be helpful, we don't always have an oscillo at hand

iceman1001, 2019-07-29:
.. yup, now as a further strict limit of available mem on fpga,  this function can save 30 000 samples..

Which leads me back to my question,   for what do you use this function?

doegox, 2019-07-29:
As said in the official PR, to see how signal looks like when card is (too) close or far from antenna, as you can see with LF today. And having some help when codec fails. 
IMHO it's interesting to see how signals look like when attempting to sniff both read and card, when relative positions of pm3, reader and card are quite critical, and learn what are the interesting positions to get a good trace.

Fl0-0, 2019-07-30:
I have used this function to debug HF reads (weak signal, too strong, tag position, noise), it is a great contribution and helps a lot when you don't a have a scope.

RfidResearchGroup, 2019-07-30:
I am not convinced at all about this command  but that doesn’t matter. It doesn’t hurt having it until the space in hf fpga image is needed for another demod. 

Lets put it on the roadmap and see if someone in the community make it happen.

iceman1001, 2020-01-12:
Done https://github.com/RfidResearchGroup/proxmark3/commit/6e3dde9d76a15bba0b8b2f13d6e787594298860b

closed 2020-01-12 by iceman1001