RfidResearchGroup/proxmark3 #300: Unable to use HF_COLIN standalone mode or SPIFFS features

luminouw, 2019-07-31:
**Describe the bug**
I was able to use HF_COLIN standalone mode up until commit [daae890].
Every version after that fails during the "WRITING FLASH" procedure and the Proxmark timeouts.
Also new SPIFFS commands don't work as expected and make the Proxmark3 timeouts as well.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone latest master
2. make clean && make all && ./flash-all.sh
3. Connect to Proxmark3
4. As indicated by @cjbrigato I wipe pages 1 and 2 (mem wipe p 1 / mem wipe p 2)
5. hw standalone
6. Approach Vigik badge, Vigik is detected and keys found
7. Gets stuck with the flashing "WRITING FLASH" message

**Expected behavior**
Get Vigik keys and info, write them to flash and then enter sim mode to replay tag

**Desktop:**
 - OS: Ubuntu 16.04 / macOS Sierra
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ] 
          

 [ CLIENT ]          
  client: RRG/Iceman          
  compiled with Clang/LLVM 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.39.2)          

 [ PROXMARK RDV4 ]          
  external flash:                  present           
  smartcard reader:                present           

 [ PROXMARK RDV4 Extras ]          
  FPC USART for BT add-on support: present           
          
 [ ARM ]
  bootrom: RRG/Iceman/master/141ab65f 2019-07-31 21:41:03
       os: RRG/Iceman/master/141ab65f 2019-07-31 21:41:31
  compiled with GCC 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]

 [ FPGA ]
  LF image built for 2s30vq100 on 2019/ 7/31 at 15:57:16
  HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 277830 bytes (53%) Free: 246458 bytes (47%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory          
```

 - hw status
```
[usb] pm3 --> hw status
#db# Memory           
#db#   BIGBUF_SIZE.............40000          
#db#   Available memory........39944          
#db# Tracing           
#db#   tracing ................1          
#db#   traceLen ...............0          
#db# Currently loaded FPGA image           
#db#   mode.................... HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23          
#db# Flash memory           
#db#   Baudrate................24 MHz           
#db#   Init....................OK           
#db#   Memory size.............2 mbits / 256 kb           
#db#   Unique ID...............0xD567A882A7608525          
#db# Smart card module (ISO 7816)           
#db#   version.................v3.11           
#db# LF Sampling config           
#db#   [q] divisor.............95 ( 125 kHz )          
#db#   [b] bps.................8          
#db#   [d] decimation..........1          
#db#   [a] averaging...........Yes          
#db#   [t] trigger threshold...0          
#db# LF T55XX config           
#db# fixed bit length (default)           
#db#   [a] startgap............29*8 (232)          
#db#   [b] writegap............17*8 (136)          
#db#   [c] write_0.............15*8 (120)          
#db#   [d] write_1.............47*8 (376)          
#db#   [e] readgap.............0*8 (0)          
#db# long leading reference           
#db#   [a] startgap............unconfigured           
#db#   [b] writegap............unconfigured           
#db#   [c] write_0.............unconfigured           
#db#   [d] write_1.............unconfigured           
#db#   [e] readgap.............0*8 (0)          
#db# leading zero           
#db#   [a] startgap............unconfigured           
#db#   [b] writegap............unconfigured           
#db#   [c] write_0.............unconfigured           
#db#   [d] write_1.............unconfigured           
#db#   [e] readgap.............0*8 (0)          
#db# 1 of 4 coding reference           
#db#   [a] startgap............unconfigured           
#db#   [b] writegap............unconfigured           
#db#   [c] write_0.............unconfigured           
#db#   [d] write_1.............unconfigured           
#db#   [e] readgap.............unconfigured           
#db#   [f] write_2.............unconfigured           
#db#   [f] write_3.............unconfigured           
#db# Transfer Speed           
#db#   Sending packets to client...          
#db#   Time elapsed............500ms          
#db#   Bytes transferred.......666112          
#db#   Transfer Speed PM3 -> Client = 1332224 bytes/s          
#db# Various           
#db#   DBGLEVEL................1          
#db#   ToSendMax...............-1          
#db#   ToSendBit...............0          
#db#   ToSend BUFFERSIZE.......2308          
#db# Installed StandAlone Mode           
#db#   HF Mifare ultra fast sniff/sim/clone - aka VIGIKPWN (Colin Brigato)          
#db# Flash memory dictionary loaded           
#db#   Mifare..................808 keys          
#db#   T55x7...................109 keys          
#db#   iClass..................7 keys   
```

 - data tune
```
[usb] pm3 --> data tune
          
[=] Measuring antenna characteristics, please wait...
          
....
          
[+] LF antenna: 37.25 V - 125.00 kHz          
[+] LF antenna: 30.02 V - 134.00 kHz          
[+] LF optimal: 37.25 V - 125.00 kHz          
[+] LF antenna is OK  
          
[+] HF antenna: 47.59 V - 13.56 MHz          
[+] HF antenna is OK           
          
[+] Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.
          
No GUI in this build!
```

**Additional context**
The "Transfer Speed PM3 -> Client" line fails randomly with USB connection, always works with Bluetooth.

SPIFFS commands will fail and Proxmark3 will stop answering until power reset.
```
[usb] pm3 --> mem wipe p 1
[+] Flash WIPE ok          
[usb] pm3 --> mem wipe p 2
[+] Flash WIPE ok          
[usb] pm3 --> mem spiffs test
#db# ---------------------------          
#db# Testing SPIFFS operations          
#db# ---------------------------          
#db# (all test are made using lazy safetylevel)          
#db# * Mounting filesystem (lazy).......          
[usb] pm3 --> hw ping
[=] Ping sent          
[!] Ping response timeout           
```

cjbrigato, 2019-08-01:
I'm dumb. I made a mistake on twitter, indeed.
`mem wipe p 0`
`mem wipe p 1`
not 1;2

sorry for this. please confirm

iceman1001, 2019-08-01:
There is a bit of mismatch in the armsrc/Makefile.
SRC_SPIFFS  demands WITH_FLASH directive but isn't controlled by it.
I did an adaptation of it to sort that particular case.

luminouw, 2019-08-01:
Hey there, transfer speed issue has been fixed by #302.

SPIFFS issue still present on latest master for me (f6befc2 at the time of writing this).

```
[usb] pm3 --> mem wipe p 0
[+] Flash WIPE ok          
[usb] pm3 --> mem wipe p 1
[+] Flash WIPE ok          
[usb] pm3 --> mem spiffs test
#db# ---------------------------          
#db# Testing SPIFFS operations          
#db# ---------------------------          
#db# (all test are made using lazy safetylevel)          
#db# * Mounting filesystem (lazy).......          
[usb] pm3 --> hw ping
[=] Ping sent          
[!] Ping response timeout           
[usb] pm3 --> mem spiffs tree
[usb] pm3 --> 
< i still have the shell here if I try to quit CTRL+C or CTRL+D following messager appears and i need to unplug the PM3 to get back my shell>
UART:: write time-out
```

Same tests with debug level 4
```
[usb] pm3 --> hw dbg 4
#db# Debug level: 4          
[usb] pm3 --> mem wipe p 0
#db# Checkbusy in...          
#db# Flash Write enabled          
#db# Checkbusy in...          
#db# FlashStop          
[+] Flash WIPE ok          
[usb] pm3 --> mem wipe p 1
#db# Checkbusy in...          
#db# Flash Write enabled          
#db# Checkbusy in...          
#db# FlashStop          
[+] Flash WIPE ok          
[usb] pm3 --> mem spiffs test
#db# ---------------------------          
#db# Testing SPIFFS operations          
#db# ---------------------------          
#db# (all test are made using lazy safetylevel)          
#db# * Mounting filesystem (lazy).......          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
#db# Checkbusy in...          
#db# Checkbusy in...          
#db# FlashStop          
[usb] pm3 --> 
```

doegox, 2019-08-01:
> There is a bit of mismatch in the armsrc/Makefile.
> SRC_SPIFFS demands WITH_FLASH directive but isn't controlled by it.
> I did an adaptation of it to sort that particular case.

Thanks, it's more coherent, but actually the dependency of SRC_FLASH itself against WITH_FLASH wasn't required, it's more "lap belt and shoulder harness" as we say in French, in case someone didn't protect their use with the macro in the rest of the code.

cjbrigato, 2019-08-01:
@luminouw your "reproduce" steps are wrong since it would not produce an image with HF_COLIN at all.
Can you : 
- `git clone`, not pull, the actual master in a *new repo*,
- `cd proxmark3;cp Makefile.platform.sample Makefile.platform`
- modify the Makefile.platform and add just in extra `BT_ADDON` and in standalone `HF_COLIN`
- `make clean; make; make ` (yes it's wanted), no "-j" please for the last make
- `client/flasher /dev/tty*** armsrc/obj/fullimage.elf`
- `client/proxmark3 /dev/tty***`
- `hw status`
- `mem wipe p 0`
- `mem wipe p 1`
- `mem wipe p 2`
- `mem spiffs test`

? thx in advance

doegox, 2019-08-01:
**if** this fails too, we should test some basic read/write of your flash (mem load / mem dump)

luminouw, 2019-08-01:
Yeah my bad, I actually pulled before with an existing Makefile.platform.
Here is the output of all previous commands with a clean repo clone.

- clone / copy&edit Makemake.platform / make
```
luminouw@ubuntu ~/dev $ git clone https://github.com/RfidResearchGroup/proxmark3.git
Clonage dans 'proxmark3'...                                                 
remote: Enumerating objects: 81, done.                            
remote: Counting objects: 100% (81/81), done.                                                       
remote: Compressing objects: 100% (70/70), done.  
remote: Total 39971 (delta 23), reused 26 (delta 11), pack-reused 39890
Réception d'objets: 100% (39971/39971), 30.22 MiB | 5.89 MiB/s, fait.
Résolution des deltas: 100% (31116/31116), fait.            
Vérification de la connectivité... fait.          
luminouw@ubuntu ~/dev $ cd proxmark3                                                                                                                                                                                                        
luminouw@ubuntu ~/dev/proxmark3 (master) $ cp Makefile.platform.sample Makefile.platform                                                                                                                                                    
luminouw@ubuntu ~/dev/proxmark3 (master) $ vi Makefile.platform
luminouw@ubuntu ~/dev/proxmark3 (master) $ make clean; make; make
                                           
===================================================================       
Platform name:     Proxmark3 rdv4                                     
PLATFORM:          PM3RDV4                                            
Platform extras:   BTADDON                                               
Included options:  SMARTCARD FLASH FPC_USART_HOST LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA HFSNIFF FPC_USART
Standalone mode:   HF_COLIN                
===================================================================
[!] Platform definitions changed, cleaning bootrom/armsrc/recovery first...
[*] MAKE armsrc/Standalone/Makefile.hal        
make[1]: rien à faire pour « Standalone/Makefile.hal ».
[*] MAKE client/clean                                                                                                          
[*] MAKE bootrom/clean                     
[*] MAKE armsrc/clean                                                                               
[*] MAKE recovery/clean                                         
[*] MAKE mfkey/clean                                   
[*] MAKE nonce2key/clean                         
                                                            
===================================================================
Platform name:     Proxmark3 rdv4       
PLATFORM:          PM3RDV4         
Platform extras:   BTADDON               
Included options:  SMARTCARD FLASH FPC_USART_HOST LF HITAG ISO15693 LEGICRF ISO14443b ISO14443a ICLASS FELICA HFSNIFF FPC_USART
Standalone mode:   HF_COLIN              
===================================================================       
[*] MAKE armsrc/Standalone/Makefile.hal
make[1]: rien à faire pour « Standalone/Makefile.hal ».
[*] MAKE client/all                        
[*] MAKE liblua for linux                                                             
[-] CC lapi.c                              
[-] CC lcode.c                             
[-] CC lctype.c                       
[-] CC ldebug.c                                
[-] CC ldo.c                                     
[-] CC ldump.c                             
[-] CC lfunc.c                                                                                  
[-] CC lgc.c
[-] CC llex.c
[-] CC lmem.c
[-] CC lobject.c
[-] CC lopcodes.c
[-] CC lparser.c
[-] CC lstate.c
[-] CC lstring.c
[-] CC ltable.c
[-] CC ltm.c
[-] CC lundump.c
[-] CC lvm.c
[-] CC lzio.c
[-] CC lauxlib.c
[-] CC lbaselib.c
[-] CC lbitlib.c
[... 595 lines cut ...]
#db#   DBGLEVEL................1
#db#   ToSendMax...............-1
#db#   ToSendBit...............0
#db#   ToSend BUFFERSIZE.......2308
#db# Installed StandAlone Mode
#db#   HF Mifare ultra fast sniff/sim/clone - aka VIGIKPWN (Colin Brigato)
#db# Flash memory dictionary loaded
#db#   Mifare..................808 keys
#db#   T55x7...................109 keys
#db#   iClass..................7 keys
```

- final mem spiffs test
```
[usb] pm3 --> mem spiffs test
#db# ---------------------------
#db# Testing SPIFFS operations
#db# ---------------------------
#db# (all test are made using lazy safetylevel)
#db# * Mounting filesystem (lazy).......
[usb] pm3 -->
```

---
On top of that, if I add a few prints in spiffs.c, seems like it's entering the rdv40_spiffs_mount() function but doesn't go past the SPIFFS_MOUNT call (line 136). Haven't tried to debug this one yet to see where it actually stops.

doegox, 2019-08-01:
@luminouw just to be sure do you mind testing your flash ?
See https://gist.github.com/doegox/c35925b14c7b8dc4baf42d61d9c71f1b
You can reduce the nr of loops, 20 is more than enough.
Thanks.

doegox, 2019-08-01:
And also some info on your system, as the declared USB speed in hw status is twice larger than usual.
Laptop model? OS? kernel version?

doegox, 2019-08-01:
also to try to isolate the issue, could you test (if possible):
* another RDV4 on your same machine
* your RDV4 on another machine (not same hw model)

luminouw, 2019-08-01:
@doegox test-flash script returned success (20 loops)
After wiping pages 0,1,2 and dumping flash to file, file is indeed full of FFFFFFFF

As for setup : 
- Desktop machine (acer something, not at home atm)
- OS Ubuntu 16.04
- Kernel info `Linux ubuntu 4.4.0-157-generic #185-Ubuntu SMP Tue Jul 23 09:17:01 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`

I tried on macOS Sierra yesterday as well (MBP 2010), not with latest master from this morning though.

Might be able to test with another RDV4 tomorrow or this weekend, I'll let you know : )

Is there any way to "hard reset" RDV4 from scratch, even if that means JTAG and stuff (got all i need at home) ? Maybe a fresh base would help ?

cjbrigato, 2019-08-01:
Hum.
Backup your whole flash, then try a "full wipe".
*be sure to backup full flash so you don't loose signature !*

when you're ready (*after a backup*, again):
`mem wipe i`
`mem spiffs test`
I smell something along the lines even if we take only the first 128kb, I smell a Garbage collector run or something along the lines going crazy for some reason.

cjbrigato, 2019-08-01:
@doegox WOW!
https://github.com/RfidResearchGroup/proxmark3/blob/f6befc235fe65fa92dc603406d448f3eac07960a/armsrc/flashmem.c#L556
Did not see before but you use `ReadDataCont` multiple times in a row without even a checkbusy. 
This is not how you use it, this function is here to split the init wor for loops or Register read/Writing enable operations to avoid a full init/stop regarding the spi... Still you NEED to `Flash_Checkbusy` in between ! since his usb speed is Twice as normal, this will just don't work, and that should never have.
This is a strong candidate here for the problem, making a flash_init being done while busy or two init in a row etc. 
This may also explain your first BUSY_TIMEOUT tries where in fact it just need some more precaution regarding ReadDataCont. 
Check WriteData for example, it uses WriteDataCont in a loop with checks in between.
This must be corrected before annoying @luminouw with any other checks.

luminouw, 2019-08-01:
@cjbrigato Thanks for the update : ] I'll wait until further fixes for the ultimate backup / wipe solution.

doegox, 2019-08-01:
@cjbrigato so Flash_ReadDataCont should have
```
    if (!len || Flash_CheckBusy(BUSY_TIMEOUT)) return 0;
```
as for  Flash_ReadData ?
Or is there any valid reason to mandate the busy check out of ReadDataCont ?

doegox, 2019-08-01:
btw while at it, Flash_WriteData does not check return value of Flash_CheckBusy and maybe Flash_WriteDataCont would also require some Flash_CheckBusy

cjbrigato, 2019-08-01:
> @cjbrigato so Flash_ReadDataCont should have
no. DataCont is to be minimal, there are case where the Check is better outside.
It's not the functions that have to be changed : it's their usage. I know I should have at least made a little doc. I'm used to writing such drivers so I did not thought about it. It's classic when using SPI, you want some minimal functions to speed things and optimise cases.

> btw while at it, Flash_WriteData does not check return value of Flash_CheckBusy
This is because Checkbusy contains this : 
```    
WaitUS(WINBOND_WRITE_DELAY);
```
:)

cjbrigato, 2019-08-01:
I think we would benefit from a "SAFE" level modes with tracking of flash being inited/stop already, because low level function and higher ones assuming whole operation are mixed.
Spiffs lazy/safe mechanism would make all this more coherent and safe.

doegox, 2019-08-01:
btw this one is also called without CheckBusy 
* https://github.com/RfidResearchGroup/proxmark3/blob/f6befc235fe65fa92dc603406d448f3eac07960a/armsrc/appmain.c#L1826
as well as WriteDataCont:
* https://github.com/RfidResearchGroup/proxmark3/blob/f6befc235fe65fa92dc603406d448f3eac07960a/armsrc/appmain.c#L1773-L1782

doegox, 2019-08-01:
> no. DataCont is to be minimal, there are case where the Check is better outside.
> It's not the functions that have to be changed : it's their usage.

What about a flag in their API?
At least that would force people to think about the need for a BusyCheck or not, and avoid faulty usages.

cjbrigato, 2019-08-01:
I'll just refactor flashmem driving the same way I did with SPIFFS and safety levels. If one don't want to optimize he can rely on highest safety level, at cost of performance regarding eventual useless Reinitialisation. Still he will be able to use the same forms loop/Controls for the same result. If one want to optimize and take in charge the Init,Stop, then he can rely on the lazy level which will provide helpers regarding the changed/initial/normal states.
Then if one want to take everything at charge, the unsafe level will need to add the CheckBusy, WriteEnables,Eventually Reading registers, etc.
With all the same base code, probably compatible with some `#ifdef MY_OPTI_LEVEL` chunks.

In the meantime, can you just quickly patch the print_info so @luminouw can continue his adventure into destroying all our certitude with his Twice Faster USB ? :D

iceman1001, 2019-08-01:
His usb speed is ver odd.

doegox, 2019-08-02:
@luminou please test latest commit

luminouw, 2019-08-02:
@doegox Results below, doesn't look like it changed quoi que ce soit : )

```
 [ ARM ]
  bootrom: RRG/Iceman/master/96ed9076 2019-08-03 00:17:22
       os: RRG/Iceman/master/96ed9076 2019-08-03 00:18:03
  compiled with GCC 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
```

and transfer speed

```
#db# Transfer Speed           
#db#   Sending packets to client...          
#db#   Time elapsed............500ms          
#db#   Bytes transferred.......681984          
#db#   Transfer Speed PM3 -> Client = 1363968 bytes/s    
```
What should be the correct range speed for it not look suspicious ?

Didn't get the chance to try with another RDV4, next window will be next week.

iceman1001, 2019-08-03:
On Linux you will see 600 000-ish bytes/s

On MingW 
```
#db# Transfer Speed
#db#   Sending packets to client...
#db#   Time elapsed............501ms
#db#   Bytes transferred.......55808
#db#   Transfer Speed PM3 -> Client = 111393 bytes/s
```

cjbrigato, 2019-08-03:
@luminouw it needs a schemas.json file. it's now fully dynamic. But then it needs spiffs more than ever.

luminouw, 2019-08-05:
Gave it a try with latest master and full flash wipe (`mem wipe i`) after backuping signature.
Still high transfer speeds reported and `mem spiffs test` stop at `#db# * Mounting filesystem (lazy).......`

Successfully restored signature on my device, and re-tested after that, same problem. Will try with a different RDV4 later this week and report here.

cjbrigato, 2019-08-05:
This is both strange and sad. I've managed to have 5 others people to test it and prove working, and for each one which get it working makes me worry a little more for either your device or a race_conditional state that will be hell to seek and destroy.

doegox, 2019-08-05:
> Will try with a different RDV4 later this week and report here.

Good idea.
Did you try your RDV4 on different machines/OSes/VMs/... ?
Which setups exactly? (hw, host OS, VM OS if under VM,...)

luminouw, 2019-08-05:
@doegox Yeah same errors / messages from both setups (also tried compiling / flashing on both platforms)
Will give it a go in my windows 10 home machine too just in case.

Is there any way to fully reset the RDV4 as in "out of the box" state ? Ready to get my hands dirty with JTAG if needed :]

iceman1001, 2019-08-05:
By now I am starting to consider that idea that @luminouw  has a faulty device

doegox, 2019-08-05:
> Is there any way to fully reset the RDV4 as in "out of the box" state ? Ready to get my hands dirty with JTAG if needed :]

Yes, which hardware do you have for doing JTAG?

See https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/jtag_notes.md and https://github.com/RfidResearchGroup/proxmark3/tree/master/tools/jtag_openocd

doegox, 2019-08-05:
ok some progress...
I asked @luminouw to try with CONN_SPEED_TEST_MIN_TIME=5000 and `hw status` took... 10s.
So it's not USB which is twice faster, it's internal PM3 clock which is abnormally twice slower.
This can explain easier why spiffs fails...
@iceman1001 @cjbrigato any idea how we are in this situation?
48MHz clock should be ok else the USB would be screwed as well.

iceman1001, 2019-08-05:
The arm chip?   bad soldering on a pin?

cjbrigato, 2019-08-05:
@doegox this was my guess when you were here ! I made some joke about his time being twice as our. It was my guess since when he showed he was on real hardware not VM and that it was independant.

Let's see if this is physical or not.
@iceman1001 btw spiffs relye on AT91C_BASE_TC's , where the speed test rely on low precision timer from RTTC so this may be unrelated or it is. 
just a check @luminouw you are sure you have absolutely no other relicates of includes regarding some AT91 or some registers ? no modification at all in at91sam7s512.h , nor in ticks.c either ?

cjbrigato, 2019-08-05:
BTW @doegox  we can test the Clocks With OpenOCD while he is jtagging.

an exemple is what is used during actual unbrickking in the chip-atsam917s

```
sam7x.cpu configure -event reset-init {
	soft_reset_halt
	mww 0xfffffd00 0xa5000004	# RSTC_CR: Reset peripherals
	mww 0xfffffd44 0x00008000	# WDT_MR: disable watchdog
	mww 0xfffffd08 0xa5000001	# RSTC_MR enable user reset
	mww 0xfffffc20 0x00005001	# CKGR_MOR : enable the main oscillator
	sleep 10
	mww 0xfffffc2c 0x000b1c02	# CKGR_PLLR: 16MHz * 12/2 = 96MHz
	sleep 10
	mww 0xfffffc30 0x00000007	# PMC_MCKR : MCK = PLL / 2 = 48 MHz
	sleep 10
	mww 0xffffff60 0x00480100	# MC_FMR: flash mode (FWS=1,FMCN=72)
	sleep 100
}
```

cjbrigato, 2019-08-05:
also @doegox maybe i'm drawn into to much device and messing everything up, but since USB is functionning, what clock would actually be misbehavior ?
I though this all started with the Slow clock @32Khz then builded up by configuring Main oscillator then PLL etc etc. Isn't there a all should fail or nothing should fail case ?

iceman1001, 2019-08-05:
yup, mess up the clocks and all kinds of stuff goes wrong.

doegox, 2019-08-05:
I'd say presumably his MCK is ok (else no USB) but his slow clock is twice slower, why???

iceman1001, 2019-08-05:
His MCU is bad?   or bad soldering?

cjbrigato, 2019-08-05:
Strange.

iceman1001, 2019-08-05:
Wrong power?

doegox, 2019-08-05:
We dumped clock registers.
@luminouw has different values for
* PMC_MCFR 00011041  => mainf=0x1041 => ((256000 + (mainf / 2)) / mainf) = 3E
* RTTC_RTMR 0000003E          

While I have on mine:
* PMC_MCFR 00011EEB  => mainf=0x1EEB => ((256000 + (mainf / 2)) / mainf) = 20
* RTTC_RTMR 00000020          

Other values are identical:
* PMC_MOR 00000801          
* PMC_PLLR 100B3F02          
* PMC_MCKR 00000007

doegox, 2019-08-06:
If I understand correctly the goal of PMC_MCFR, it looks like either his external Xtal is slower, or his slow clock faster than normal.
I can't find how slow clock speed is setup. It's said to be from internal RC but I didn  find details.

iceman1001, 2019-08-06:
clocks..  I have a special relation with the ARM clocks on pm3. I wanted to forget.

---looking at it ,  functions that should have delta time
armsrc/ticks.c   line 
https://github.com/RfidResearchGroup/proxmark3/blob/master/armsrc/ticks.c#L29
This loop should have deltatime,  I don't think its used so much nowdays (spindelay)

and there is also this kind of loop inside 
common/usb_cdc.c 
https://github.com/RfidResearchGroup/proxmark3/blob/master/common/usb_cdc.c#L465

Question is how much delay the extra check ( if-statement)  will add to the clock precision ?

----
    // we are using a 16 MHz crystal as the basis for everything
    // slow clock runs at 32Khz typical regardless of crystal

Ask proxgrind about the crystal? 

----
Silly question,  but still?
There isn't a tools/jtag_openocd/ cfg that configurs the clock ? Like it does it temporary but not permanent?

doegox, 2019-08-06:
Yeah there is a kind of debug clock, but unrelated to main clk or slow clk as far as I understand.
Regarding slow clock, "RC Oscillator ranges between 22 kHz and 42 kHz" and there can be quite large deviations, e.g. Application Note doc6148 talks about possible 30% and an error compensation algorithm.

doegox, 2019-08-06:
Ok, so StartTickCount is computing the compensation, so even if the RC is not at 32kHz exactly, slow clock is supposed to run at 1kHz, based on the external 16MHz xtal. The only reason I see why his slow clock is twice slower (as 5s of slow clock took in reality 10s) is that the external xtal is also twice slower. 
But if it was the case (like faulty xtal), how comes the USB was working?
Anyway at this point it's pretty clear it's a hardware problem.

doegox, 2019-08-06:
> ---looking at it , functions that should have delta time
> armsrc/ticks.c line
> https://github.com/RfidResearchGroup/proxmark3/blob/master/armsrc/ticks.c#L29
> This loop should have deltatime, I don't think its used so much nowdays (spindelay)

This one is fine I think because ` AT91C_BASE_PWMC->PWMC_ENA = PWM_CHANNEL(0);` is resetting the counter as well.
But the comparison would look safer with `if (now >= (uint16_t)(start + ticks))` (e.g. in case we start using interrupts)

luminouw, 2019-08-06:
Ok update here,

ran some tests with @doegox help yesterday, with a RPi0 and JTAG, after compiling openocd and checking the wires, I was able to dump the flash, enter interactive mode and flash the recovery to the RDV4.

When unJTAGging the device and plugging it back via USB, I was able to run some tests (see the clocks speed dump). Turned off everything, plugged back (JTAG) it to reflash one last time with latest repo commit recovery image. Unplugged, micro USB in ... and nothing. Not a single light would turn on, tried with the blueshark battery, same thing (although blueshark + usb charges the battery).

If i plug it back jtag style, I can access the flash. If I dump my flash a few seconds appart, there are some bitflips between files, although @doegox intuition is it's just a reading issue and the flash actually = last recovery image I flashed.

Today when looking closer with a small usb microscope thingy, noticed a couple of resistors that look pretty bad (read burnt/fried), next to the 16Mhz crystal, not sure the condition pre-existed before and JTAGging somehow messed things up or if there was a short during my last manipulation (although the wires didn't move and i didn't change the setup from previous flashes).

tl;dr Got a RDV4 that only lights up when JTAGging now :'(

iceman1001, 2019-08-12:
Sounds like faulty device.  Sadly.
I will close this issue.  and if @luminouw still has same problem with replacement unit,  then he can start a new issue

closed 2019-08-12 by iceman1001