RfidResearchGroup/proxmark3 #301: hw status transfer speed randomly fails in USB mode

luminouw, 2019-07-31:
**Describe the bug**
When running hw status in USB mode only, the Transfer Speed tests either return 0 bytes transferred or too many according to @iceman1001 

When using Bluetooth connectivity, commands succeeds every time.

**To Reproduce**
Steps to reproduce the behavior:
1. Flash latest version
2. Run hw status
3. Error scenario 1 (most frequent)

```
#db# Transfer Speed           
#db#   Sending packets to client...          
#db#   Time elapsed............-1ms          
#db#   Bytes transferred.......512          
#db#   Transfer Speed PM3 -> Client = 0 bytes/s               
```
3. Error scenario 2 (every few attemps)
```
#db# Transfer Speed           
#db#   Sending packets to client...          
#db#   Time elapsed............500ms          
#db#   Bytes transferred.......645120          
#db#   Transfer Speed PM3 -> Client = 1290240 bytes/s    
```

**Expected behavior**
Consistent value on transfer speed in USB mode.

**Desktop (please complete the following information):**
 - OS: Ubuntu 16.04 / macOS Sierra
 - inside proxmark3 client run the following commands and paste the output here.
 - hw version
```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ] 
          

 [ CLIENT ]          
  client: RRG/Iceman          
  compiled with Clang/LLVM 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.39.2)          

 [ PROXMARK RDV4 ]          
  external flash:                  present           
  smartcard reader:                present           

 [ PROXMARK RDV4 Extras ]          
  FPC USART for BT add-on support: present           
          
 [ ARM ]
  bootrom: RRG/Iceman/master/141ab65f 2019-07-31 21:41:03
       os: RRG/Iceman/master/141ab65f 2019-07-31 21:41:31
  compiled with GCC 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]

 [ FPGA ]
  LF image built for 2s30vq100 on 2019/ 7/31 at 15:57:16
  HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 277830 bytes (53%) Free: 246458 bytes (47%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory    
```
 - hw status
```
[usb] pm3 --> hw status
#db# Memory           
#db#   BIGBUF_SIZE.............40000          
#db#   Available memory........39944          
#db# Tracing           
#db#   tracing ................1          
#db#   traceLen ...............0          
#db# Currently loaded FPGA image           
#db#   mode.................... HF image built for 2s30vq100 on 2018/ 9/ 3 at 21:40:23          
#db# Flash memory           
#db#   Baudrate................24 MHz           
#db#   Init....................OK           
#db#   Memory size.............2 mbits / 256 kb           
#db#   Unique ID...............0xD567A882A7608525          
#db# Smart card module (ISO 7816)           
#db#   version.................v3.11           
#db# LF Sampling config           
#db#   [q] divisor.............95 ( 125 kHz )          
#db#   [b] bps.................8          
#db#   [d] decimation..........1          
#db#   [a] averaging...........Yes          
#db#   [t] trigger threshold...0          
#db# LF T55XX config           
#db# fixed bit length (default)           
#db#   [a] startgap............29*8 (232)          
#db#   [b] writegap............17*8 (136)          
#db#   [c] write_0.............15*8 (120)          
#db#   [d] write_1.............47*8 (376)          
#db#   [e] readgap.............0*8 (0)          
#db# long leading reference           
#db#   [a] startgap............unconfigured           
#db#   [b] writegap............unconfigured           
#db#   [c] write_0.............unconfigured           
#db#   [d] write_1.............unconfigured           
#db#   [e] readgap.............0*8 (0)          
#db# leading zero           
#db#   [a] startgap............unconfigured           
#db#   [b] writegap............unconfigured           
#db#   [c] write_0.............unconfigured           
#db#   [d] write_1.............unconfigured           
#db#   [e] readgap.............0*8 (0)          
#db# 1 of 4 coding reference           
#db#   [a] startgap............unconfigured           
#db#   [b] writegap............unconfigured           
#db#   [c] write_0.............unconfigured           
#db#   [d] write_1.............unconfigured           
#db#   [e] readgap.............unconfigured           
#db#   [f] write_2.............unconfigured           
#db#   [f] write_3.............unconfigured           
#db# Transfer Speed           
#db#   Sending packets to client...          
#db#   Time elapsed............500ms          
#db#   Bytes transferred.......645632          
#db#   Transfer Speed PM3 -> Client = 1291264 bytes/s          
#db# Various           
#db#   DBGLEVEL................1          
#db#   ToSendMax...............-1          
#db#   ToSendBit...............0          
#db#   ToSend BUFFERSIZE.......2308          
#db# Installed StandAlone Mode           
#db#   HF Mifare ultra fast sniff/sim/clone - aka VIGIKPWN (Colin Brigato)          
#db# Flash memory dictionary loaded           
#db#   Mifare..................808 keys          
#db#   T55x7...................109 keys          
#db#   iClass..................7 keys   
```
 - data tune
```
[usb] pm3 --> data tune
          
[=] Measuring antenna characteristics, please wait...
          
....
          
[+] LF antenna: 37.25 V - 125.00 kHz          
[+] LF antenna: 30.02 V - 134.00 kHz          
[+] LF optimal: 37.25 V - 125.00 kHz          
[+] LF antenna is OK  
          
[+] HF antenna: 47.59 V - 13.56 MHz          
[+] HF antenna is OK           
          
[+] Displaying LF tuning graph. Divisor 89 is 134khz, 95 is 125khz.
```

cjbrigato, 2019-07-31:
You're speed is somehow really faster than I'm used too.
Anyway the issues lies in the fact that the `test_data` used for transfer speed measuring is made of :
```
uint8_t *test_data = BigBuf_get_addr();
```
but since you got such a speed; you actually get *past* the BigBuf boundaries, or even the whole ATSAM7S512 SRAM boundaries, hence the stop at `645120bytes`.
Quite interesting that it does not make the device in a random unusable behavior, this could have been the case.
So if this is to be fixed anyhow, @iceman1001 / @doegox (fyi) , just make the test_data circular or more safe. Since the biggest possible size of a sent packet is predictable, we can make it fixed, so instead of starting at `BigBuf_get_addr();` then throwing caution through the wind we could just make `[PM3_CMD_DATA_SIZE]` sized

cjbrigato, 2019-07-31:
Well with some more check and tries and reading again, it is probable my former post is 100% bullsh*t.

cjbrigato, 2019-07-31:
Ok got it.

So this is a debug output regarding the Start/stop/delta times when everything works : 

```
#db# DBG: start_time : 81622, stop_time : 81623 , delta_time : 1 , bytes t : 512
#db# DBG: start_time : 81622, stop_time : 81624 , delta_time : 2 , bytes t : 1024
#db# DBG: start_time : 81622, stop_time : 81626 , delta_time : 4 , bytes t : 1536
#db# DBG: start_time : 81622, stop_time : 81627 , delta_time : 5 , bytes t : 2048
#db# DBG: start_time : 81622, stop_time : 81629 , delta_time : 7 , bytes t : 2560
#db# DBG: start_time : 81622, stop_time : 81630 , delta_time : 8 , bytes t : 3072
...
#db# DBG: start_time : 81622, stop_time : 82116 , delta_time : 494 , bytes t : 167936
#db# DBG: start_time : 81622, stop_time : 82118 , delta_time : 496 , bytes t : 168448
#db# DBG: start_time : 81622, stop_time : 82119 , delta_time : 497 , bytes t : 168960
#db# DBG: start_time : 81622, stop_time : 82121 , delta_time : 499 , bytes t : 169472
#db# DBG: start_time : 81622, stop_time : 82122 , delta_time : 500 , bytes t : 169984
#db#   Time elapsed............500ms
#db#   Bytes transferred.......169984
#db#   Transfer Speed PM3 -> Client = 339968 bytes/s
```

And there is when it doesn't : 
```
#db# Transfer Speed
#db#   Sending packets to client...
#db# DBG: start_time : 5456, stop_time : 5456 , delta_time : -1 , bytes t : 512
#db#   Time elapsed............-1ms
#db#   Bytes transferred.......512
#db#   Transfer Speed PM3 -> Client = 0 bytes/s
```
or again : 
```
#db# DBG: start_time : 82736, stop_time : 82736 , delta_time : -1 , bytes t : 512
```

Since `delta_time` rely on GetTickCountDelta : 
```
uint32_t RAMFUNC GetTickCountDelta(uint32_t start_ticks) {
    uint32_t stop_ticks = AT91C_BASE_RTTC->RTTC_RTVR;
    if (stop_ticks > start_ticks)
        return stop_ticks - start_ticks;
    return (UINT32_MAX - start_ticks) + stop_ticks;
}
```
In our case, when `stop_ticks == start_ticks` , indeed we return `UINT32_MAX`, and so after the 2's comple, its indeed a -1, or again way greater (so definitely way more delta than  the 500ms we stop at).

I'll send a micro-PR regarding this.

cjbrigato, 2019-07-31:
PRed.

iceman1001, 2019-08-01:
Excellent find!

PR #302 (2019-08-01): Off by one in GetTickCountDelta (FIXes #301)
FIXes #301 and prevent a `-1`aka`UINT_MAX` delta.
commit b4e004ab (2019-08-01): Off by one in GetTickCountDelta (FIXes #301)

FIXes #301 and prevent a `-1`aka`UINT_MAX` delta.
commit da815d74 (2019-08-01): Merge pull request #302 from cjbrigato/patch-1

Off by one in GetTickCountDelta (FIXes #301)
closed 2019-08-01 by iceman1001