RfidResearchGroup/proxmark3 #309: Add a way to disable full RATS support for Simulatio,n

cjbrigato, 2019-08-04:
**Is your feature request related to a problem? Please describe.**
commit 63f9f11066cca02078dc9348f615c9c1ee94ebe0 has permitted a full rats support in the good way for simulation.
Still, such feature in between timing-critical operation has made the simulation less reliable against time-frustrated readers.
As an exemple, when facing some android readers on TagInfo, a Mifare1k can be seen as a KievMetro Cards and then used unaccordingly, with auth made with the MAD classic key seen as correct when they are not, thus thinking the tag is always answering "10 B2 00 00 00 .." etc 
blocks. This being one very specific case, you can have all sort of behavior here.

**Describe the solution you'd like**
Just an internal flag to the `mifare1ksim` function to deal flexibly with such case where timings are critical. Here we talk about a flag to disable/hard-make something which won't loose cycle in conditionals, but this could be done for some other feature I guess, so we could provide custom modes from the most flexible but less reliable , to the less flexible but most reliable.

**Describe alternatives you've considered**
Knocking at my neighbour door at 4am in the morning. Did not lead to any comparable results.

**Additional context**
I link 4 images of the *same* simulation session describing this case, with the GOOD and the WRONG case. (overall 30%success against an unrooted NoteS9)

Good :
http://apoui.net/android/scr_GOOD_SIM1.jpg
http://apoui.net/android/scr_GOOD_SIM2.jpg
Wrong :
http://apoui.net/android/scr_WRONG_SIM1.jpg
http://apoui.net/android/scr_WRONG_SIM2.jpg

cjbrigato, 2019-08-04:
CC @McEloff (thanks for the great work btw !)

iceman1001, 2019-08-05:
question is if the addition of an extra if-statement (ISO..CMD..RATS)  is the cause,   because rats is only enabled in 2K simulation if I understand the code correct in the Mifaresiminit.

Does your android readers try to read RATS? or is it a problem with pm3 simulation?

cjbrigato, 2019-08-05:
@iceman1001 yep I've seen just right now that RATS should not be a problem.
What happens is that somehow and I don't know how, the MifareSim1k from HF_COLIN standalone don't manage to trigger a correct Simulation regarding flags.

Or inverse.

When I do `hf mf sim` without any other things after having a proper Mifare 1K tag (like from my standalone) in the shell it works flawlessy...*BUT*
it also says this : 
`Options [ numreads: 0, flags: 16 (0x10) ]`
Which is nonsense since the result of such call being (in order)
```
    uint16_t flags = 0;
    flags |= FLAG_MF_1K;
    flags |= FLAG_UID_IN_EMUL;
```
un less I miss something from client...
So i made this simple : 
```
#include "unistd.h"
#include "stdio.h"
#include "inttypes.h"

#define FLAG_UID_IN_EMUL        0x10
#define FLAG_MF_1K              0x100

int main()
{
    uint16_t flags = 0;
    flags |= FLAG_MF_1K;
    flags |= FLAG_UID_IN_EMUL;
    printf("flags: %d (0x%02x)\n", flags,flags);
    return 0;
}
```
which prints this : 
```
flags: 272 (0x110)
```

Just as My HF_COLIN Standalone done.

Then I tried to force 16/0x10 as Mifare1ksim flags in my standalone, regardless of everything else : it works.

So I don't understand.
If correct flags are sent for the uid_in_emul + mifare 1k, this don't works (trigger RATS support).
It works from client... which doesn't seems to send correct flags.

I'm lost here, but since i'm tired, @iceman1001 please save my soul by pointing your finger to the more than evident thing I'm missing here.

cjbrigato, 2019-08-05:
FOUND IT  @iceman1001 

This is because of this

When you pass flags 0x16 (not passing the 1K) : 
```
usb] pm3 --> hf mf sim
[=] Mifare   |  UID  N/A
[=] Options [ numreads: 0, flags: 16 (0x10) ]
```
what happens in this in `MifareSimInit`: 
```
    if ((flags & FLAG_UID_IN_EMUL) == FLAG_UID_IN_EMUL) {
        uint8_t block0[16];
        emlGetMemBt(block0, 0, 16);

        // If uid size defined, copy only uid from EMUL to use, backward compatibility for 'hf_colin.c', 'hf_mattyrun.c'
        if ((flags & (FLAG_4B_UID_IN_DATA | FLAG_7B_UID_IN_DATA | FLAG_10B_UID_IN_DATA)) != 0) {
            memcpy(datain, block0, 10);  // load 10bytes from EMUL to the datain pointer. to be used below.
        } else {
            // Check for 4 bytes uid: bcc corrected and single size uid bits in ATQA
            if ((block0[0] ^ block0[1] ^ block0[2] ^ block0[3]) == block0[4] && (block0[6] & 0xc0) == 0) {
                flags |= FLAG_4B_UID_IN_DATA;
                memcpy(datain, block0, 4);
                rSAK[0] = block0[5];
                memcpy(rATQA, &block0[6], sizeof(rATQA));
            }
            // Check for 7 bytes UID: double size uid bits in ATQA
            else if ((block0[8] & 0xc0) == 0x40) {
                flags |= FLAG_7B_UID_IN_DATA;
                memcpy(datain, block0, 7);
                rSAK[0] = block0[7];
                memcpy(rATQA, &block0[8], sizeof(rATQA));
            } else {
                Dbprintf("[-] ERROR: Invalid dump. UID/SAK/ATQA not found");
                return false;
            }
        }
```

So in my case, the block0 is : 
``` 
block0[16]={0x3E,0x8A,0x2D,0x7D,0xE4,0x88,0x04,0x00,0x47,0x51,0x35,0xD9,0x49,0x10,0x40,0x08};
```
the the SAK will be 0x88.

BUT when you pass flags 0x110(Adding the FLAG_MF_1K)
you also trigger this *after* the former block : 
```
    if ((flags & FLAG_MF_MINI) == FLAG_MF_MINI) {
        memcpy(rATQA, rATQA_Mini, sizeof(rATQA));
        rSAK[0] = rSAK_Mini;
        Dbprintf("Mifare Mini");
    } else if ((flags & FLAG_MF_1K) == FLAG_MF_1K) {
        memcpy(rATQA, rATQA_1k, sizeof(rATQA));
        rSAK[0] = rSAK_1k;
        Dbprintf("Mifare 1K");
    } else if ((flags & FLAG_MF_2K) == FLAG_MF_2K) {
        memcpy(rATQA, rATQA_2k, sizeof(rATQA));
        rSAK[0] = rSAK_2k;
        *rats = rRATS;
        *rats_len = sizeof(rRATS);
        Dbprintf("Mifare 2K with RATS support");
    } else if ((flags & FLAG_MF_4K) == FLAG_MF_4K) {
        memcpy(rATQA, rATQA_4k, sizeof(rATQA));
        rSAK[0] = rSAK_4k;
        Dbprintf("Mifare 4K");
    }
```
So the SAK is now set to 0x88.

This is problem regarding how this is handled after. Not having this sak correctly set is one thing  but not having the rest initialized with correct awareness of SAK will make things flaky.

So the bug is very conditional here , to trigger it : 
- have a block[0] in emulator made Mifare 1k sim with 0x88 SAK
- set the flags both FLAG_UID_IN_EMUL and  FLAG_MF_1K setting for emulation : `hf mf sim t 1`

not so rare  I think. 
So it should : 
- be printed  somewhere that a 0x08 sak will be enforced regardless of emulator memory when you explicitely tells you want Mifare 1k Despite this being the default
- update the emulator memory and simulation preparation according to such modification at every place this has an impact (CRCs etc)

or we could just behave the same regardless of explicitly telling the defaults... and always checkk the emulator memory for an eventual custom SAK.

Btw, having the defaults and Explicitely told defaults resulting in different flags (so different result here) is somehow strange... Should after all the FLAG_MF_1K passed  to the device by the client, so 0x110 it should be ?

McEloff, 2019-08-05:
That is right. If you set FLAG_MF_1K directly, simulator uses SAK and ATQA for 1K tag.
If you set only FLAG_UID_IN_EMUL, then simulator uses UID, SAK, ATQA from dump.

iceman1001, 2019-08-05:
yeah,  maybe not clear instructions.
Loading a dump could be using  uid/saq/atqa from dump,   or  still even provide with yr own.
Given vigik protection,  where it reads sak from block0 and expects a different sak in the anticollision.

doegox, 2019-08-05:
what about just changing the messages like this
```diff
 if ((flags & FLAG_MF_MINI) == FLAG_MF_MINI) {
         memcpy(rATQA, rATQA_Mini, sizeof(rATQA));
         rSAK[0] = rSAK_Mini;
-        Dbprintf("Mifare Mini");
+        Dbprintf("Enforcing Mifare Mini SAK (0x%04X)", rSAK[0]);
     }
```
etc.

cjbrigato, 2019-08-05:
As @iceman1001 explained, it is most subtile for vigik : 
> Given vigik protection, where it reads sak from block0 and expects a different sak in the anticollision.
Yes, regarding vigik, there are Genuine tags which are expected to both have a SAK 0x88 but still have a 0x08 in block 0 with according BCC/CRC etc, so a random classic Chinese tag cannot behave accordingly.

I think we need to make more coherent what the "default" behavior is, what enforces what, and if a good soul is ready to provide a coherent way (despites the FlagsGate :p) to enforce SAK independant from Block0 as feature it would be very appreciated.

iceman1001, 2019-08-05:
enforcing default sak/atqa
with a overriding possibility, that allows to set which sak/atqa to be using in anticollision regardless of block0

doegox, 2019-08-06:
see 46153548640f036e0fae6fbfaaf9c703a5bf5103 and 12a0540d91e621a4b0139e36eecf77ea826480f1.
Is it ok for you as such? If yes, you can close the ticket.

cjbrigato, 2019-08-07:
So cool. does it work ?

doegox, 2019-08-07:
I did some tests with my phone and Taginfo, yes I think so.

doegox, 2019-09-30:
@cjbrigato can we close this one ?

cjbrigato, 2019-10-02:
definitely @doegox

closed 2019-10-02 by cjbrigato