RfidResearchGroup/proxmark3 #314: Broken Proxmark3 kit after flashing

colourful-land, 2019-08-12:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. flash the bootrom
````
$ sudo client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf 
Loading ELF file 'bootrom/obj/bootrom.elf'...
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000c84->0x00000c84) [R X] @0x298

Waiting for Proxmark to appear on /dev/ttyACM0 .
 Found.
#db# unknown command:: 0xc10d410d          
#db# unknown command:: 0xc10d4154          
Entering bootloader...
(Press and release the button only to abort)
Waiting for Proxmark to appear on /dev/ttyACM0 .......
 Found.

Flashing...
Writing segments for file: bootrom/obj/bootrom.elf
 0x00100000..0x001001ff [0x200 / 1 blocks]. OK
 0x00100200..0x00100e83 [0xc84 / 7 blocks]....... OK

Resetting hardware...
All done.

Have a nice day!
````
2. Flash the full image;
````
$ sudo client/flasher /dev/ttyACM0 armsrc/obj/fullimage.elf 
Loading ELF file 'armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x0002f070->0x0002f070) [R X] @0x94
1: V 0x00200000 P 0x00131070 (0x000018e0->0x000018e0) [RW ] @0x2f104
Note: Extending previous segment from 0x2f070 to 0x30950 bytes

Waiting for Proxmark to appear on /dev/ttyACM0 .
 Found.
Entering bootloader...
(Press and release the button only to abort)
Waiting for Proxmark to appear on /dev/ttyACM0 .......
 Found.

Flashing...
Writing segments for file: armsrc/obj/fullimage.elf
 0x00102000..0x0013294f [0x30950 / 389 blocks]............................Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button          
^C
````

I used ^C to stop because the device is audiblly recycling.

3. Attempt to redo the flashing by pressing the button down and holding it while reconnect:
````
$ sudo client/flasher /dev/ttyACM0 armsrc/obj/fullimage.elf 
Loading ELF file 'armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x0002bb80->0x0002bb80) [R X] @0x94
1: V 0x00200000 P 0x0012db80 (0x00001960->0x00001960) [RW ] @0x2bc14
Note: Extending previous segment from 0x2bb80 to 0x2d4e0 bytes

Waiting for Proxmark to appear on /dev/ttyACM0 .
 Found.
Waiting for a response from the proxmark...          
You can cancel this operation by pressing the pm3 button          
^C
````
I quit with `^C` because the device has reset (audible clicking sound from a relay + `dmesg` shows the device hopped to `ttyACM1`

**Expected behavior**
Flash elf successfully.

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Ubuntu 19.04.

I could do `hw version` and `hw status` because it no longger works. But before I flash it, it works fine reading and cloning my devices (using viking)

**Additional context**
The repair procedure I took is copied from [another bug](https://github.com/RfidResearchGroup/proxmark3/issues/35#issuecomment-431681848) in which it worked. In my case it didn't.

> **In short** (if your device gets enumerated on ACM0 adjust accordingly):
> 
>     1. unplug device
> 
>     2. press button and keep it pressed (IMPORTANT)
> 
>     3. plug in device
> 
>     4. run flash command  sudo client/flasher /dev/ttyACM0 armsrc/obj/fullimage.elf
> 
>     5. wait until flash is finished.
> 
>     6. release button.
> 
>     7. un/plug device.

iceman1001, 2019-08-12:
hm...  have you modified your   Makefile.platform file to target your device? 
and ran make clean; make ?

colourful-land, 2019-08-13:
> hm... have you modified your Makefile.platform file to target your device?
> and ran make clean; make ?

yes, I did make clean before making. I later tried the latest release instead of git master branch to do step 3 (fixing the broken device by holding down the button and reflash) and it failed with the same behaviour.

No I did not modify Makefile.platform. I went straight to make.

iceman1001, 2019-08-13:
that will be a problem...  since by default this repo targets RDV4...  
You will need to read  https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/4_Advanced-compilation-parameters.md

strasharo, 2019-08-18:
Same just happened to me with a RDV4 clone. Never had issues like this before. I'm flashing with:
client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf armsrc/obj/fullimage.elf

Using git bisect I found that the last good commit for me without the issue is this one:
https://github.com/RfidResearchGroup/proxmark3/commit/1da98c7aa6f72c1424069256534bc87a8512b1d2
The next one https://github.com/RfidResearchGroup/proxmark3/commit/961f98c125e6cce26fd593987b209c21669df245 couldn't compile because of this exception:

```
spiffs.c: In function 'copy_in_spiffs':
spiffs.c:282:20: error: implicit declaration of function 'BigBuf_malloc'; did you mean '__builtin_malloc'? [-Werror=implicit-function-declaration]
     uint8_t *mem = BigBuf_malloc(size);
                    ^~~~~~~~~~~~~
                    __builtin_malloc
spiffs.c:282:20: error: initialization makes pointer from integer without a cast [-Werror=int-conversion]
spiffs.c: In function 'rdv40_spiffs_safe_print_fsinfo':
spiffs.c:541:5: error: implicit declaration of function 'DbpString'; did you mean 'Dbprintf'? [-Werror=implicit-function-declaration]
     DbpString(_BLUE_("Flash Memory FileSystem Info (SPIFFS)"));
     ^~~~~~~~~
     Dbprintf
spiffs.c:541:15: error: implicit declaration of function '_BLUE_'; did you mean '__LINE__'? [-Werror=implicit-function-declaration]
     DbpString(_BLUE_("Flash Memory FileSystem Info (SPIFFS)"));
               ^~~~~~
               __LINE__
cc1: all warnings being treated as errors
make[1]: *** [../common_arm/Makefile.common:87: obj/spiffs.o] Error 1
make: *** [Makefile:52: armsrc/all] Error 2

```

And the next one has the issue present:
https://github.com/RfidResearchGroup/proxmark3/commit/0ace6bffb821c13b07cabd95ec74f69308cace17

So the issue appeared in either 961f98c125e6cce26fd593987b209c21669df245, which couldn't compile or 0ace6bffb821c13b07cabd95ec74f69308cace17.

iceman1001, 2019-08-20:
Have your issues resolved itself using the latest source?

javastraat, 2019-08-27:
rtfm einstein, played one hour to get my PM3OTHER to work in one second ...... :) (special edition signed by the iceman)
all works fine on rdv and pm3

iceman1001, 2019-08-27:
Yo Einstein!  I missed you at CCCamp!   And I met babak (tools founder US) at DEFCON!  
And I have a lockpick now.  haha,    Thought you sold that one on EBAY ;)

javastraat, 2019-08-27:
ah nice, had to work :( nice that you went to ccc. yeap i know babak :)

iceman1001, 2019-08-27:
I have to visit Amsterdam then.  :)

iceman1001, 2019-08-28:
closed due to inactivity by OP.

closed 2019-08-28 by iceman1001

colourful-land, 2019-09-10:
Okay I rebuilt and make clean and make install with the following `Makefile.platform`:
````
PLATFORM=PM3OTHER
````
I got
````
$ pm3-flash-bootrom bootrom/obj/bootrom.elf
[=] Waiting for Proxmark3 to appear...
[=] Session log /home/weiwu/.proxmark3/log_20190910.txt
[+] About to use the following file:          
[+]     /usr/local/bin/../share/proxmark3/firmware/bootrom.elf          
[+] Waiting for Proxmark3 to appear on /dev/pm3-0           
.Found           
[=] You can cancel this operation by pressing the pm3 button          
[!] Communicating with Proxmark3 device failed           
^C
````

iceman1001, 2019-09-10:
Have you tried following the suggestions in the troubleshooting document?
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Troubleshooting.md

colourful-land, 2019-09-10:
> Have you tried following the suggestions in the troubleshooting document?
> https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Troubleshooting.md

The list suggested running `pm3-flash-bootrom` which I did. But I'll switch to another computer on which I can safely remove modem-manager to retest now that you point to that as a possible culprit.

† [pm3-flash-bootroom.txt](https://github.com/RfidResearchGroup/proxmark3/files/3594236/pm3-flash-bootroom.txt)