RfidResearchGroup/proxmark3 #316: SPIFFS Writing to flash is brokken after summer restruct.

cjbrigato, 2019-08-12:
**Describe the bug**
#db# errno -10010
 when attempting ny kkind of writing.
Reading is fine.
will check ...

cjbrigato, 2019-08-12:
omg... sorry @doegox but there are some absolute mess and nonsense regarding include management here...

This will be hell to debug here.

cjbrigato, 2019-08-12:
@doegox why on earth did you add all the include s*it even in low Spiffs original implementation files... this is properly undebugable.

cjbrigato, 2019-08-12:
wontfix

closed 2019-08-12 by cjbrigato

doegox, 2019-08-12:
it works for me...
do you still have a bug or was it sth else ?

iceman1001, 2019-08-12:
@cjbrigato  You have still a problem?

cjbrigato, 2019-08-12:
It's way more complicated than being code side. Spiffs implementation is not strong enough here. I'll make some improvement as I can.