RfidResearchGroup/proxmark3 #319: repo has no tags

ZeroChaos-, 2019-08-14:
from time to time, it's important to leave a tag in the repo so users have a better idea of "the developers think this version is stable".  I realize that this software is under heavy development, but it doesn't help the users to leave them guessing as to if the last commit was a major one that needs some testing or a basic stability fix.  Consider adding a tag every few weeks when you think the code is "stable enough"

doegox, 2019-08-14:
it's in our intentions, still there are major pending issues refraining us to tag anything as "close to stable" right now. And a first step is to deploy some unittests at least on everything we can do without specific tag manipulations.

iceman1001, 2019-08-14:
and the NG format..
and return codes...
and adapt to all @ZeroChaos issues in order to get a  **make install**  level...

then we would do a TAG/release v4.0
and each large enough mods,  we will increment TAG/release V4.0.n

iceman1001, 2019-09-05:
We haven't forgotten it.   We are just not ready for a release yet :)

ZeroChaos-, 2019-09-10:
all of the bugs I had which were critical are fixed.  can I do anything to help you prepare for a release? any specific testing needs done?  I'm a great tester since I have no clue what I'm doing

doegox, 2019-09-10:
The current plan is to get rid of the most important bugs / changes.
To give you an idea, so far we have
* 2/12 public issues solved
* 4/22 private issues solved
So still some work ;)

Public issues you can help with if you wish:

- [ ] https://github.com/RfidResearchGroup/proxmark3/issues/372 update bundled/patched zlib
- [ ] https://github.com/RfidResearchGroup/proxmark3/issues/285 Spiffs continuation/documentation/impact
- [ ] https://github.com/RfidResearchGroup/proxmark3/issues/273 "lf hid wiegand" for 37-bit card off by one.
- [ ] https://github.com/RfidResearchGroup/proxmark3/issues/271 "lf hid clone" for 37-bit cards not working after writing 26-bit cards.
- [ ] https://github.com/RfidResearchGroup/proxmark3/issues/240 Last bit omitted in rawdemod
- [ ] https://github.com/RfidResearchGroup/proxmark3/issues/124 LF HITAG - fails
- [ ] https://github.com/RfidResearchGroup/proxmark3/issues/119 [idea] link in amiibo tool software
- [ ] https://github.com/RfidResearchGroup/proxmark3/issues/126 hf 15: possible truncations
- [ ] https://github.com/RfidResearchGroup/proxmark3/issues/129 ? [idea] implement json/eml/bin everywhere 
- [ ] https://github.com/RfidResearchGroup/proxmark3/issues/209 ? Legic_clone.lua // bad argument

iceman1001, 2019-09-10:
If you are not a coder then I would suggest you just test the commands against readers / tags you have at home.  Especially if you equipment which relates to those issues mentioned

ZeroChaos-, 2019-09-26:
@doegox can you please periodically reassess your checklist here?  I can't personally help with much of this, but others may be able to

doegox, 2019-09-26:
@ZeroChaos- I did already and removed the solved ones from the list ;)

iceman1001, 2020-02-28:
@doegox  ok,  we seem never to get to a release state.   Lets start adding tags.
We can call this first one   **pre.v4**

iceman1001, 2020-04-24:
Some feedback, one more PR is coming,  then we are ready to make release / tag.

iceman1001, 2020-05-21:
And after some intense efforts by @doegox et al we have a release / tag for this repo.
Great accomplishement!   

Testing of  _make install_  hasn't been done much.  We did focus on making the cross compilation on heaps of OS and hardware work nice.  Also added cmake for the client part,  and getting rid of all kinds of warnings on the different GCC versions and CLANG.

Time to make a write-up on everything new inside.   CHANGELOG.md tells just a part of it.

Ref
https://github.com/RfidResearchGroup/proxmark3/releases/tag/v4.9224

closed 2020-05-21 by iceman1001