RfidResearchGroup/proxmark3 #320: use ~/.proxmark3 or similar

ZeroChaos-, 2019-08-14:
this tools drops a few file in whatever cwd is.  this means that when the tool is installed systemwide it drops files wherever I happen to be when I run proxmark3.  This dirty's up the filesystem and means proxmark3 doesn't have access to the files it expects when I happen to run from a different directory.  Please pick a dot directory like .proxmark3 or .pm3 or even a non-dot directory and use it.

iceman1001, 2019-08-14:
yup,  Totally agree

iceman1001, 2019-08-28:
and now we do have a ~/.proxmark3
which is searched for  lualibs / luascripts / dictionaries/  folders.

doegox, 2019-08-28:
it's a start but we're not yet done with  ~/.proxmark3
see https://github.com/RfidResearchGroup/proxmark3/issues/325#issuecomment-523633394 for details

doegox, 2019-08-28:
still imho we can close this one as we're tracking the issue in the more global context of make install

doegox, 2019-08-28:
I think you didn't follow the recent progresses on that one ZeroChaos.
For dicts and luascripts we're automatically searching at specific locations:
* in current dir relative paths, so it works from the non-installed repo
* in ~/.proxmark3 relative paths, for *user provided* dicts and scripts, i.e. stuffs we don't provide ourselves
* in /usr/local/share/proxmark3 (or whatever system path it was used) for stuff deployed via make install

doegox, 2019-08-28:
and yes, history and log are still todo, as shown in #325

doegox, 2019-08-28:
yes, so a local file or a file provided in ~/.proxmark3 can shadow an installed one, as typically user-provided stuffs can shadow system-installed ones.

doegox, 2019-08-28:
and full path is displayed when item is used, so you can see from where it was taken

doegox, 2019-08-28:
history & logfile moved to ~/.proxmark3 now, time for field testing...

iceman1001, 2019-08-29:
to see the positive in this ,  we actually get to release how many files there are laying around.

iceman1001, 2019-08-30:
time to close?

doegox, 2019-08-30:
yep I'm done on my side.
we're ready to have a make install! let's continue with #325

closed 2019-08-30 by doegox