RfidResearchGroup/proxmark3 #321: add official support for installing systemwide

ZeroChaos-, 2019-08-14:
In pentoo I have proxmark3 installed as a package.  I drop the bins in /usr/bin and the rest in /usr/share/proxmark3.  To make this functional, I do the following during install:
```
sed -i -e 's#lualibs/#../../usr/share/proxmark3/lualibs/#' client/scripting.h
sed -i -e 's#scripts/#../../usr/share/proxmark3/scripts/#' client/scripting.h
```
While this exact fix is likely not great for upstream, could you please have it (runtime) check those directories (probably) after checking the relative directories?

doegox, 2019-08-28:
yes exactly, and default is here
```
include/common.h
23:#ifndef PM3_SHARE_PATH
24:# define PM3_SHARE_PATH "/usr/local/share/proxmark3/"
```

doegox, 2019-08-28:
btw `script list` shows you where we look, here from my non-installed repo:
```
[usb] pm3 --> script list
├── /home/phil/devel/00smartcard/readers/proxmark/proxmark3-RRG.git/client/luascripts/          
│   ├── 14araw.lua                     
......
│   └── ul_uid.lua                     
├── /home/phil/.proxmark3/luascripts/          
│   └── mfu_magic.lua                  
└── /usr/local/share/proxmark3/luascripts/ => NOT FOUND          
```

iceman1001, 2019-08-28:
Maybe we shouldn't print not found folders...

doegox, 2019-08-28:
it depends, here when run from repo, it shows where it would get deployed and I think it's useful for a dev working from repo.
but when installed, we shouldn't show sth like /usr/local/bin/luascripts/ which is the "repo path relative to the binary"

~/.proxmark3, even if it doesn't exist, shows to people that they can create and use it if they want.

iceman1001, 2019-08-28:
seems a bit farfetched for a tree command.  Right now the not found gives an impression that something is wrong

doegox, 2019-08-28:
it has been removed

iceman1001, 2019-08-28:
yeah,  it can print an empty folder (or non existing)  thats fine.  but message was confusing.