RfidResearchGroup/proxmark3 #322: Proxmark3 not recognized since 763c943

luminouw, 2019-08-14:
**Describe the bug**
Since commit 763c943 (two previous ones didn't compile) the PM3 (RDV 4.01) isn't recognized anymore on USB and BT.
On linux, lsusb hangs until I unplug the device.
Checking out back to 1da98c7 brings back the device and allows connection with the client.

**To Reproduce**
Steps to reproduce the behavior:
1. Git clone (current master is at db0ac16)
2. make clean
3. make
4. put pm3 in bootloader mode and ./flash-all.sh

**Additional context**
Can't connect the device to MacOS either.

```
$ dmesg
[...]
[28750.704790] usb 3-12: new full-speed USB device number 31 using xhci_hcd
[28765.817317] usb 3-12: device descriptor read/64, error -110
```

luminouw, 2019-08-18:
PM3 still not detected by client with latest build (3ae3adf0a817ac2c3009eace0ed2d7f05c235753).
Manually putting it in flash mode (holding button and powering up) allows flash without problem.

MacOSX dmesg log shows this when plugging / unplugging the device:
```
069768.132286 AppleUSB20HubPort@fa140000: AppleUSBHostPort::disconnect: persistent enumeration failures
069771.941565 AppleUSB20HubPort@fa140000: AppleUSBHostPort::disconnect: persistent enumeration failures
```

iceman1001, 2019-08-18:
usb_cdc.c  change?  the only thing was the timer changes...  from MS to US..  which shouldn't do much...

luminouw, 2019-08-18:
For the sake of exhaustivity, gave a compilation try under WSL (Ubuntu 16.04), without any modification in Makefile.platform and I get the same result in Windows, it compiles ok, I can flash, but after that the PM3 can't be found and is listed as "Unknown USB device".

iceman1001, 2019-08-18:
I just ran latest source on WSL  (ubuntu)  and compiling,  flash bootrom, flash fullimage, run client,  works just like it should. It get identifed and enumerated.  Just lovely.   
My Makefile.platform  has the last two lines enabled. ..

What is you output from flashing?

luminouw, 2019-08-18:
Ok so as I suspected, I still have SPIFFS issues.
https://github.com/RfidResearchGroup/proxmark3/blob/3ae3adf0a817ac2c3009eace0ed2d7f05c235753/armsrc/appmain.c#L1910-L1912

Commenting out this line bypasses the spiffs check and allows the PM3 to boot normally and now being recognized normally.

Should this issue be closed and another one related to the SPIFFS problem I encounter opened ?

iceman1001, 2019-08-18:
lets try,   
```
make clean;
make -j8;
./flash-bootrom.sh
./flash-fullimage.sh
./proxmark3.sh
```

strasharo, 2019-08-18:
Same just happened to me with a RDV4 clone. Never had issues like this before. I'm flashing with:
client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf armsrc/obj/fullimage.elf

Using git bisect I found that the last good commit for me without the issue is this one:
https://github.com/RfidResearchGroup/proxmark3/commit/1da98c7aa6f72c1424069256534bc87a8512b1d2
The next one https://github.com/RfidResearchGroup/proxmark3/commit/961f98c125e6cce26fd593987b209c21669df245 couldn't compile because of this exception:

```
spiffs.c: In function 'copy_in_spiffs':
spiffs.c:282:20: error: implicit declaration of function 'BigBuf_malloc'; did you mean '__builtin_malloc'? [-Werror=implicit-function-declaration]
     uint8_t *mem = BigBuf_malloc(size);
                    ^~~~~~~~~~~~~
                    __builtin_malloc
spiffs.c:282:20: error: initialization makes pointer from integer without a cast [-Werror=int-conversion]
spiffs.c: In function 'rdv40_spiffs_safe_print_fsinfo':
spiffs.c:541:5: error: implicit declaration of function 'DbpString'; did you mean 'Dbprintf'? [-Werror=implicit-function-declaration]
     DbpString(_BLUE_("Flash Memory FileSystem Info (SPIFFS)"));
     ^~~~~~~~~
     Dbprintf
spiffs.c:541:15: error: implicit declaration of function '_BLUE_'; did you mean '__LINE__'? [-Werror=implicit-function-declaration]
     DbpString(_BLUE_("Flash Memory FileSystem Info (SPIFFS)"));
               ^~~~~~
               __LINE__
cc1: all warnings being treated as errors
make[1]: *** [../common_arm/Makefile.common:87: obj/spiffs.o] Error 1
make: *** [Makefile:52: armsrc/all] Error 2

```

And the next one has the issue present:
https://github.com/RfidResearchGroup/proxmark3/commit/0ace6bffb821c13b07cabd95ec74f69308cace17

So the issue appeared in either 961f98c125e6cce26fd593987b209c21669df245, which couldn't compile or 0ace6bffb821c13b07cabd95ec74f69308cace17.

Tried the steps suggested by @iceman1001 and that didn't help:

```
./flash-bootrom.sh
[=] Waiting for Proxmark to appear...
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0           
.Found           
[+] Entering bootloader...           
[+] (Press and release the button only to abort )          
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0           
. Found           
[=] Available memory on this board: 512K bytes
          
[=] Permitted flash range: 0x00100000-0x00180000          
[+] Loading ELF file bootrom/obj/bootrom.elf           
[+] Loading usable ELF segments:          
[+]    0 : V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94          
[+]    1 : V 0x00200000 P 0x00100200 (0x00000df0->0x00000df0) [R X] @0x298          
          
          
[+] Flashing... 
          
[+] Writing segments for file: bootrom/obj/bootrom.elf          
[+]  0x00100000..0x001001ff [0x200 / 1 blocks]          
. OK           
[+]  0x00100200..0x00100fef [0xdf0 / 7 blocks]          
....... OK           

          
[+] All done.           

Have a nice day!      
```    

```
./flash-fullimage.sh
[=] Waiting for Proxmark to appear...
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0           
.Found           
[+] Entering bootloader...           
[+] (Press and release the button only to abort )          
[+] Waiting for Proxmark3 to appear on /dev/ttyACM0           
. Found           
[=] Available memory on this board: 512K bytes
          
[=] Permitted flash range: 0x00102000-0x00180000          
[+] Loading ELF file armsrc/obj/fullimage.elf           
[+] Loading usable ELF segments:          
[+]    0 : V 0x00102000 P 0x00102000 (0x0003ec00->0x0003ec00) [R X] @0x94          
[+]    1 : V 0x00200000 P 0x00140c00 (0x000014bc->0x000014bc) [RW ] @0x3ec94          
          
          
[+] Flashing... 
          
[+] Writing segments for file: armsrc/obj/fullimage.elf          
[+]  0x00102000..0x00140bff [0x3ec00 / 502 blocks]          
...................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................... OK           
[+]  0x00140c00..0x001420bb [0x14bc / 11 blocks]          
........... OK           

          
[+] All done.           

Have a nice day!   
```

```

./proxmark3.sh
[=] Waiting for Proxmark to appear...

```

After it I'm only getting these:

```
[1115618.816379] usb 3-1: USB disconnect, device number 73
[1115620.539966] usb 3-1: new full-speed USB device number 74 using uhci_hcd
[1115636.403992] usb 3-1: device descriptor read/64, error -110
[1115652.274965] usb 3-1: device descriptor read/64, error -110
[1115652.497980] usb 3-1: new full-speed USB device number 75 using uhci_hcd
[1115668.147994] usb 3-1: device descriptor read/64, error -110
[1115684.020986] usb 3-1: device descriptor read/64, error -110
[1115684.127996] usb usb3-port1: attempt power cycle
[1115684.554998] usb 3-1: new full-speed USB device number 76 using uhci_hcd
[1115695.888981] usb 3-1: device not accepting address 76, error -110
[1115696.002952] usb 3-1: new full-speed USB device number 77 using uhci_hcd
[1115707.151964] usb 3-1: device not accepting address 77, error -110
[1115707.151997] usb usb3-port1: unable to enumerate USB device

```

uhei, 2019-08-18:
With latest HEAD fullimage.elf PM3RDV4 is also not detected on MacOS. I can flash the fullimage and bootloader while pressing the button when connecting it by USB.
`git bisect` shows the first commit where this issue starts is  961f98c. 961f98c itself didn't compile.

As @strasharo  wrote this seems to be a SPIFFS issue.

soffchen, 2019-08-19:
> With latest HEAD fullimage.elf PM3RDV4 is also not detected on MacOS. I can flash the fullimage and bootloader while pressing the button when connecting it by USB.
> `git bisect` shows the first [...]

Same issue with the latest HEAD-71c5ae1 firmware, not detected, but can be flashed while pressing the button.

megabug, 2019-08-19:
The problem appears to be due to misaligned work buffers provided to SPIFFS, causing a failure when SPIFFS attempts to read a uint16_t from one.

https://github.com/megabug/proxmark3-rrg/commit/8b22624f36fdba57814c73afdd68296f57058405 fixes this for me.

cjbrigato, 2019-08-19:
Nice catch. Never managed to trigger it with my already bloated spiffs state.

Anyway this garbage work at boot should be conditional on the actual usage of spiffs or a proper initialisation. Still it should have timeouted somehow, but it seems this fallback from some low lever functions are still unreliable. Should be better with the incoming rewrite of spi flash driver.

cjbrigato, 2019-08-19:
@megabug the eventual misalign here is because how how the buffer are made.
We better align ourselves in without the misery trick attributes. It happened when 
The available FDz came to 3 up from 2, when I was fixing the whole garbage collection. We better either step back to 2 now it's fixed or to 4 or to review more deeply the way spiffs author himself proposes buffer calculation. 
Also, there is some mechanism to fight against such misalignment in spiffs implementation itself. We could leverage these. 
Etc etc

luminouw, 2019-08-19:
@cjbrigato @megabug Can confirm this (https://github.com/megabug/proxmark3-rrg/commit/8b22624f36fdba57814c73afdd68296f57058405) fixes it for me too when applied to latest HEAD (71c5ae1e1e66bb459ffc87c910f6f0ee69e75274).
SPIFFS works for me as well now (never been able to use it before) :)

iceman1001, 2019-08-19:
@cjbrigato   as an intermediate fix we can go with megabugs solution?   Or are you making a PR now?

cjbrigato, 2019-08-19:
Intermediate fix is perfect since it is not messy and avoid another spiffs pm patch over original code. I've already spent too much time on making it more reliable and the megabug fixe is technically the best even if it may hide a design flow. If anyhow this fix alongside with the other table alignment I made before is not enough then we will just migrate to Littlefs from mbed, but spiffs aging too quickly with near 0 attention from upstream. 
Still,as long as it works I'd prefer stick with spiffs. 
We can close this and I should find a window to rewrite flashmem driver. I

doegox, 2019-08-19:
Excellent, let's close it as @iceman1001 merged it and @luminouw reported success.

doegox, 2019-08-19:
and thanks a lot to @megabug !!

strasharo, 2019-08-19:
Still fails for me with latest head. :(

doegox, 2019-08-19:
> Same just happened to me with a RDV4 clone

@strasharo what do you mean a RDV4 clone ? A real RDV4 or an Pm3 Easy clone ??

strasharo, 2019-08-19:
It's this one:
https://www.aliexpress.https://www.aliexpress.com/item/32808141763.html/item/32808141763.html?spm=a2g0s.9042311.0.0.655e4c4d6XR1gg

Has the RDV4 hardware in the form factor of an easy:

```
[usb] pm3 --> hw version

 [ Proxmark3 RFID instrument ] 
          

 [ CLIENT ]          
  client: RRG/Iceman          
  compiled with GCC 9.1.1 20190503 (Red Hat 9.1.1-1)          

 [ PROXMARK RDV4 ]          
  external flash:                  absent           
  smartcard reader:                absent           

 [ PROXMARK RDV4 Extras ]          
  FPC USART for BT add-on support: absent           
          
 [ ARM ]
  bootrom: RRG/Iceman/HEAD/1da98c7a 2019-08-19 21:12:34
       os: RRG/Iceman/HEAD/1da98c7a 2019-08-19 21:12:53
  compiled with GCC 7.4.0

 [ FPGA ]
  LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
  HF image built for 2s30vq100 on 2018-09-03 at 21:40:23          

 [ Hardware ]           
  --= uC: AT91SAM7S512 Rev B          
  --= Embedded Processor: ARM7TDMI          
  --= Nonvolatile Program Memory Size: 512K bytes, Used: 265620 bytes (51%) Free: 258668 bytes (49%)          
  --= Second Nonvolatile Program Memory Size: None          
  --= Internal SRAM Size: 64K bytes          
  --= Architecture Identifier: AT91SAM7Sxx Series          
  --= Nonvolatile Program Memory Type: Embedded Flash Memory          

```

iceman1001, 2019-08-19:
Sorry bro,  that is really not RDV4 hardware.  Its an old proxmark3 easy clone with the larger SAM7s512 chip.   You will have trouble if you do this,  ...and you seem to have trouble...

strasharo, 2019-08-19:
Well, I guess I'll have to stay at https://github.com/RfidResearchGroup/proxmark3/commit/1da98c7aa6f72c1424069256534bc87a8512b1d2 in that case. :(

doegox, 2019-08-19:
Or RTFM... https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/4_Advanced-compilation-parameters.md#firmware

closed 2019-08-19 by doegox

soffchen, 2019-08-20:
> Well, I guess I'll have to stay at [1da98c7](https://github.com/RfidResearchGroup/proxmark3/commit/1da98c7aa6f72c1424069256534bc87a8512b1d2) in that case. :(

Checkout the latest code, and `make all PLATFORM=PM3EASY512` will fix your problem.

iceman1001, 2019-08-20:
I think @doegox  was spot on.

strasharo, 2019-08-20:
Thanks a lot @soffchen , PM3EASY512 works fine for me.

fengberd, 2019-10-02:
> > Well, I guess I'll have to stay at [1da98c7](https://github.com/RfidResearchGroup/proxmark3/commit/1da98c7aa6f72c1424069256534bc87a8512b1d2) in that case. :(
> 
> Checkout the latest code, and `make all PLATFORM=PM3EASY512` will fix your problem.

For those who is using RDV2 like me, keep getting "not recognized" device on windows after flashing iceman, and you find PM3EASY512 not exists, simply compile with PM3OTHER and it should work fine :)

andrew867, 2019-11-17:
Could 'PM3EASY512' be added to the table on this page?

https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Use_of_Proxmark/4_Advanced-compilation-parameters.md#PLATFORM

andrew867, 2019-11-17:
In case someone else finds this bug and is using Homebrew on macOS.

I had to git clone the master, compile and then manually flash the pm3 clone (same Aliexpress one as above, hold button then plug in USB) using: `proxmark3 /dev/tty.usbmodemiceman1 --flash --unlock-bootloader --image bootrom/obj/bootrom.elf --image armsrc/obj/fullimage.elf` .

For some reason on my machine even when installing via Homebrew using `export HOMEBREW_PROXMARK3_PLATFORM=PM3OTHER` and then `brew install --HEAD proxmark3` the firmware that is flashed and exists in /usr/local/share/proxmark3 was for the RDV4.

iceman1001, 2019-11-17:
if there is a OSX Homebrew expert around then that would be really good.

andrew867, 2019-11-18:
@iceman1001 I used `brew remove proxmark3` after installing with the default options (which is RDV4), not sure if that deletes the firmware files from the first `brew install`. 

If someone would like to try and reproduce it these were my steps:

- Used a pre-compiled version for Windows x64 (of this iceman repo).
- Flashed the device with the RDV4 firmware (I assume as the client mentioned that the board had no extra hardware) which worked on Windows.
- `brew install --HEAD proxmark3`
- Tried to use the proxmark3 clone on MacOS 10.14.6, wasn't recognized on the USB interface properly
- Flashed firmware using `pm3-flash-all`, still wasn't recognized when booted.
- Tried flashing firmware again using `pm3-flash-*` apps, both holding the button and not.
- `brew remove proxmark3`
- `export HOMEBREW_PROXMARK3_PLATFORM=PM3OTHER` then `brew install --HEAD proxmark3`
- Tried to reflash a few times using the same commands as above.
- Still wasn't recognized and thought I totally screwed up the thing.
- Found this bug report and manually `git clone` and compile using PLATFORM=PM3OTHER
- Held down button, plugged in ran `proxmark3 /dev/tty.usbmodemiceman1 --flash --unlock-bootloader --image bootrom/obj/bootrom.elf --image armsrc/obj/fullimage.elf` from within the master directory.
- Device is now recognized properly and the client doesn't mention anything about RDV4 features.

Hope this helps!

vogelfreiheit, 2021-12-04:
PM3EASY512 is gone in current master, what would be the equivalent for a clone? (repurposing old board I have)

doegox, 2021-12-05:
Don't reuse old issues @vogelfreiheit 
Your answer was written above: compile using PLATFORM=PM3OTHER

vogelfreiheit, 2021-12-05:
Missed that one, thanks! Perhaps this should be mentioned in the Wiki.

doegox, 2021-12-05:
My bad, it's PM3GENERIC nowadays (but PM3OTHER still supported).
It's on the front page https://github.com/RfidResearchGroup/proxmark3 :

## Generic Proxmark3 platforms

In order to build this repo for generic Proxmark3 platforms we urge you to read [Advanced compilation parameters](/doc/md/Use_of_Proxmark/4_Advanced-compilation-parameters.md)