RfidResearchGroup/proxmark3 #323: stop bundling liblua

ZeroChaos-, 2019-08-14:
I'm not certain, but I'm pretty sure that client/liblua is just a bundled version of lua.  This should be already available on every major distro, can this simply be dropped?

ZeroChaos-, 2019-08-14:
I tested removing the bundled liblua and replacing the static lib with "pkg-config --libs lua" for the linking of proxmark3 and got some errors...  Not sure if I just didn't do it right or if the bundled version of lua is older/newer/modified
```
[-] CC pm3_binlib.c
pm3_binlib.c: In function ‘luaopen_binlib’:
pm3_binlib.c:322:5: error: implicit declaration of function ‘luaL_newlib’; did you mean ‘luaL_openlib’? [-Werror=implicit-function-declaration]
     luaL_newlib(L, binlib);
     ^~~~~~~~~~~
     luaL_openlib
pm3_binlib.c: In function ‘set_bin_library’:
pm3_binlib.c:330:5: error: implicit declaration of function ‘luaL_requiref’; did you mean ‘luaL_unref’? [-Werror=implicit-function-declaration]
     luaL_requiref(L, "bin", luaopen_binlib, 1);
     ^~~~~~~~~~~~~
     luaL_unref
cc1: all warnings being treated as errors
```

iceman1001, 2019-08-14:
The liblua we use is kind of old...
For us,  we need to keep stuff working on:
mingw, *nix, windows wsl, OSX, android. 

I guess we should add a depency for the tool chain to install lua instead.

ZeroChaos-, 2019-08-14:
keeping the bundled version is fine, but using the system version when available should be preferred.

iceman1001, 2019-08-14:
Its different today,   around 2014 when liblua was bundled,   the mingw was annoying.
today,  we have a nice proxspace distro ( @Gator96100  maintains it)  
all *nix, osx,  wsl can apt/emerge it...

iceman1001, 2019-08-14:
How do you swap between system and bundled?

ZeroChaos-, 2019-08-14:
if needed, you would basically try to detect if the needed libs are available in system and switch to bundled if not.  this is usually done with a configure script but it's not impossible to do in a Makefile (just uglier)

ZeroChaos-, 2019-08-14:
I would probably set LUALIBS with `pkg-config --libs lua` and if that's blank switch to the bundled version.

Upon some rough investigation I found this in the bundled version
```
lua.h:#define LUA_VERSION_MAJOR   "5"
lua.h:#define LUA_VERSION_MINOR   "2"
lua.h:#define LUA_VERSION_NUM     502
lua.h:#define LUA_VERSION_RELEASE "2"
```
Which may be newer than the version shipped as "stable" with gentoo (5.1.5).  Having a specified minimum version (and maybe checking it) would make this pretty smooth.

iceman1001, 2019-08-14:
502 vs 515...  I would say gentoo has a newer version :)

ZeroChaos-, 2019-08-14:
when I run the bundled compiled version it seems to think it's 5.2, which makes sense with the above Major/Minor

ZeroChaos-, 2019-08-14:
I mean, I have newer available in gentoo, but 5.1.5 is current "stable" :-)

Gator96100, 2019-08-14:
If you need to test liblua under ProxSpace you can simply install it with `pacman -S mingw-w64-x86_64-lua mingw-w64-i686-lua`
The latest version in msys2:
```
#define LUA_VERSION_MAJOR	"5"
#define LUA_VERSION_MINOR	"3"
#define LUA_VERSION_NUM		503
#define LUA_VERSION_RELEASE	"4"
```

eybisi, 2020-03-14:
I'm trying to install luasocket module (on my system) and use it in my lua script but I'm getting this error:
```
pm3 --> script run proxy.lua
[+] Executing Lua script: ~/proxmark3/client/luascripts/proxy.lua, args ''
          
error loading module 'socket.core' from file '/usr/local/lib/lua/5.2/socket/core.so':
	/usr/local/lib/lua/5.2/socket/core.so: undefined symbol: lua_gettop
```
Is this related to using bundled verson ? How can i swap proxmark to use my system instead of bundled?

doegox, 2020-05-21:
I pushed an attempt in branch https://github.com/RfidResearchGroup/proxmark3/tree/liblua
Makefile only.
Still CMake to do.

doegox, 2020-05-21:
FTR it fails on OSX, I'll look into it...

doegox, 2020-05-22:
Fixed on OSX and merged in master. Still no cmake support

iceman1001, 2020-05-24:
Dark Makefile woodo again.   I swear those makefiles is growing

doegox, 2020-05-25:
FTR (makefile only, not cmake), we now detect and use system libs in the client for:
* Lua
* Jansson
* Whereami
* Zlib

iceman1001, 2020-05-25:
Excellent bro!
time to close?

doegox, 2020-05-25:
yes we can close, I've still on my todo to try to bring the same detection in cmake

closed 2020-05-25 by doegox