RfidResearchGroup/proxmark3 #324: -ltermcap is deprecated

ZeroChaos-, 2019-08-14:
I can't find documentation on this right now, but I believe -ltermcap is deprecated.
Fedora lists it as deprecated since 2007-12-12
https://fedoraproject.org/wiki/Deprecated_packages

To build on gentoo we simply remove -ltermcap.

Additionally, ncurses *can* be broken up into multiple libraries (which gentoo does by default) so -lncurses needs to be replaced with `pkg-config --libs ncurses` which results in -lncurses -ltinfo (but use pkg-config, don't hard code it)

iceman1001, 2019-08-14:
... where did I hear about termcap.. o, yes,  termux (Android) said it was included.
We would need to test if termcap is needed or a replacement is to be found

iceman1001, 2019-08-14:
@phcoder   mentioned this over at old repo  https://github.com/iceman1001/proxmark3/pull/291

iceman1001, 2019-08-14:
`
-fPIC is required to compile on aarch64

NVM for -fPIC, turns out setting it in environment variable is good enough. A different question: why do we include -ltermcap ? Termux has no libtermcap and if I remove -ltermcap and add -fPIC to environment variables it compiles and work nice (through tcp-serial proxy)
`

ZeroChaos-, 2019-08-15:
-fPIC is "position independent code", and while that is a good thing it is unrelated to this issue :-)  that said, it appears there is an issue handling CFLAGS from the env so I'll open a new issue :-)

iceman1001, 2019-08-20:
Now that @merlokk  and @doegox  has remade makefile and the travis/appveyor configs ,  we should easily test removing  "-ltermcap"  from client/Makefile  and see if it still compiles.

ZeroChaos-, 2019-08-29:
Considering the testing infrastructure and the general safety of this fix, please accept my PR and I will help with any fallout.

iceman1001, 2019-08-29:
the direct call ..  pkg-config.
shouldn't it be 
```

PGKCFG = pkg-config --libs ncurses

LDLIBS +=  $(PKGCFG) 
```

ZeroChaos-, 2019-08-29:
backticks matter my friend, might be time to increase your font size :-)

doegox, 2019-08-29:
you can't in a makefile @iceman1001 
but we already use `$(shell pkg-config` elsewhere, which I find better

iceman1001, 2019-08-29:
```
LDLIBS += $(shell pkg-config --libs ncurse 2>/dev/null)
```

ZeroChaos-, 2019-08-29:
I'm fine with whichever way you prefer.  Would you like to simply commit your change or shall I amend the PR?

iceman1001, 2019-08-29:
I closed your PR,  pushed the fix.   And give me one sec, and I push changelog aswell.

Lets see if something broke.  

Time to close this issue?

ZeroChaos-, 2019-08-29:
This works for me, and I think it should work for everyone.  @ me if it's broken for others and I'll be happy to help fix.

ZeroChaos-, 2019-08-29:
Also, WOO, that was the last of the issues I was *required* to patch to make it work. Thanks for all the hard work on this "not fun" stuff

closed 2019-08-29 by ZeroChaos-

doegox, 2019-08-29:
while we're at it, I'm challenging the presence of the ncurses library.
Do we really need it at all and if yes for which platform?
Grabbed [n]curses.h includes -> nothing
Tested without ncurses:
* linux
* osx
* android termux
* windows proxspacev32
cc @ZeroChaos- @iceman1001

doegox, 2019-08-29:
Can we just remove entirely that LDLIBS and see if there are any complaints?

ZeroChaos-, 2019-08-29:
I can build without it, no idea what that breaks though

doegox, 2019-08-29:
ok I pushed it ncurses removal, let's see...