RfidResearchGroup/proxmark3 #325: add "make install"

ZeroChaos-, 2019-08-14:
I don't know if this is right, but this is how we do install in pentoo:
```
src_install(){
    #dobin client/{flasher,proxmark3,fpga_compress}
    dobin client/{flasher,proxmark3}
    #install scripts too
    insinto /usr/share/proxmark3/lualibs
    doins client/lualibs/*
    insinto /usr/share/proxmark3/scripts
    doins client/scripts/*
    if use firmware; then
        insinto /usr/share/proxmark3
        doins armsrc/obj/*.elf
        doins bootrom/obj/bootrom.elf
        doins recovery/*.bin
        doins tools/mfkey/mfkey{32,64}
    fi  
    udev_dorules driver/77-pm3-usb-device-blacklist.rules
}
```

basically, flasher and proxmark3 and dropped in /usr/bin, lualibs and scripts are dropped in /usr/share/proxmark3/, and if built, firmware is dropped in /usr/share/proxmark3.  udev rules are put into /lib/udev/rules.d/.

Things are different for me, the defaults should be /usr/local/bin, /usr/local/share/proxmark3, and /etc/udev/rules.d/ for users installing manually.  I can help write the make file if someone can comment about how "correctly" I've done things :-)

Also, it would be awesome to rename the extremely generic "flasher"

doegox, 2019-08-21:
For comparison, see also current brew formula https://github.com/RfidResearchGroup/homebrew-proxmark3/blob/master/proxmark3.rb#L38-L79

Let's try to recap what it should be:
```
# Linux/ProxSpace/WSL
PREFIX=/usr/local
# OSX with brew
PREFIX=/usr/local/Cellar/proxmark3/HEAD-<Commit-ID>

$PREFIX/bin/proxmark3
$PREFIX/bin/proxmark3.sh (<= a bit confusing... rename it pm3 as for short/quick launcher ?)
$PREFIX/bin/proxmark3-flasher
# fpga_compress doesn't need to be installed, it's useless outside compilation
$PREFIX/bin/proxmark3-flash-all.sh
$PREFIX/bin/proxmark3-flash-bootrom.sh
$PREFIX/bin/proxmark3-flash-fullimage.sh
$PREFIX/bin/mfkey32
$PREFIX/bin/mfkey32v2
$PREFIX/bin/mfkey64
$PREFIX/bin/nonce2key

$PREFIX/share/proxmark3/lualibs
$PREFIX/share/proxmark3/luascripts (currently scripts)

$PREFIX/share/proxmark3/smartcard/capk.txt
$PREFIX/share/proxmark3/smartcard/aidlist.json
$PREFIX/share/proxmark3/smartcard/oids.json
$PREFIX/share/proxmark3/smartcard/fido2_defparams.json (currently fido2.json)
$PREFIX/share/proxmark3/smartcard/emv_defparams.json

$PREFIX/share/proxmark3/hardnested/bf_bench_data.bin
$PREFIX/share/proxmark3/hardnested/tables/*

# these are nice for newcomers to learn I think
$PREFIX/share/proxmark3/traces

# These files will need better names, for now it's
$PREFIX/share/proxmark3/dictionaries/{bmp,icbmp,mrzd}_sort_keys.dic
$PREFIX/share/proxmark3/dictionaries/default_keys.dic
$PREFIX/share/proxmark3/dictionaries/default_{iclass,ultralight}_keys.dic
$PREFIX/share/proxmark3/dictionaries/default_pwd.dic

$PREFIX/share/proxmark3/firmware/bootrom.elf
$PREFIX/share/proxmark3/firmware/fullimage.elf
$PREFIX/share/proxmark3/firmware/SIM011
$PREFIX/share/proxmark3/firmware/SIM011.sha512.txt

# Linux only:
/etc/udev/rules.d/77-pm3-usb-device-blacklist.rules
```

This requires of course adaptations of the code to find its data.
It also goes with modifications for the user files:
```
~/.proxmark3/history (current .history)
~/.proxmark3/log.txt (current proxmark3.log)
~/.proxmark3/dictionaries/ (user dics)
~/.proxmark3/luascripts/ (user lua scripts)
~/.proxmark3/cmdscripts/ (user cli scripts)
~/.proxmark3/smartcard/fido2.json (user fido prefs)
```

I believe user traces and dumps should go to their current working directory, not in `~/.proxmark3`

Maybe we should also think of some history/log rotation, else people will forget these files and they'll grow and grow...

Please comment and tell me if sth is wrong or missing.

There are a bunch of useful tools tp handle EML files, what should we do?
```
eml2lower.sh
eml2UPPER.sh
pm3_mfdread.py
pm3_mfd2eml.py
pm3_eml2mfd.py
pm3_eml_mfd_test.py
```

iceman1001, 2019-08-22:
as discussed elsewhere,  lets us start changing some stuff inside the repo,  in order to make things even clearer.   like renaming client/scripts -> client/luascripts etc..  I am guess by the time I write this @doegox  has already changed it.

doegox, 2019-08-22:
Yeah I started.
Don't worry it will be more flexible than my first move on luascripts, with overridable PREFIX defaulting to /use/local and to ??? with brew

doegox, 2019-08-28:
what about these ones @iceman1001 ? 
```
client/loclass/ikeys.c
663:    if (fileExists("iclass_key.bin")) {
665:    } else if (fileExists("loclass/iclass_key.bin")) {
667:    } else if (fileExists("client/loclass/iclass_key.bin")) {

client/loclass/elite_crack.c
618:        if (fileExists("iclass_dump.bin")) {
620:        } else if (fileExists("loclass/iclass_dump.bin")) {
622:        } else if (fileExists("client/loclass/iclass_dump.bin")) {
```
are they supposed to be in the repo and installed ? I see a client/loclass/iclass_dump.bin in the repo.

iceman1001, 2019-08-29:
welll...  
yes, iclass_dump.bin is a testfile for **hf iclass loclass**
and there are more files,  "iclass_decryption.bin" ,  which is user provided.  
iclass_key.bin == masterkey leaked.
iclass_Decryptionkey.bin == hid transport 3des key

Those files should be searched for in all our places..

edward-p, 2019-08-29:
> PREFIX=/usr/local

I think `PREFIX:=/usr/local` is better. So it can be override like `make PREFIX=/usr`.

iceman1001, 2019-08-29:
@doegox   I think all those fido.json, smartcard.json, aid.json,  iclassfiles,  should be direct under  /.promark3/  no need to complicate it with subfolders with one lone file inside.

edward-p, 2019-08-29:
> /etc/udev/rules.d/77-pm3-usb-device-blacklist.rules

For user who build the project themselves `/etc/udev/rules.d/77-pm3-usb-device-blacklist.rules` is a good choice. But for package maintainers, `/usr/lib/udev/rules.d77-pm3-usb-device-blacklist.rules`  will be better. So I think something like `UDEV_PREFIX:=/etc/udev/rules.d` is better.

doegox, 2019-08-29:
> > PREFIX=/usr/local
> 
> I think `PREFIX:=/usr/local` is better. So it can be override like `make PREFIX=/usr`.

Because a make install compatible with brew requires to separate bin & share, we can't work with a unique PREFIX, so we're using overridable PM3_BIN_PATH and PM3_SHARE_PATH, defaulting to /usr/local/bin/ and /usr/local/share/proxmark3/

edward-p, 2019-08-29:
> Because a make install compatible with brew requires to separate bin & share, we can't work with a unique PREFIX, so we're using overridable PM3_BIN_PATH and PM3_SHARE_PATH, defaulting to /usr/local/bin/ and /usr/local/share/proxmark3/

That's acceptable.

doegox, 2019-08-29:
> I think something like `UDEV_PREFIX:=/etc/udev/rules.d` is better.

ok

doegox, 2019-08-29:
> all those fido.json, smartcard.json, aid.json, iclassfiles, should be direct under /.promark3/

ok, so installed under /usr/local/share/proxmark3/

iceman1001, 2019-08-29:
yeah.  let have them all in one place.

iceman1001, 2019-08-29:
And lets have ParrotOS maintainer tagged here.  (debian packaging and tag me @palinurosec)

We now have,   Pentoo/Gentoo/Archlinux/ParrotOS   ..

not to mention Homebrew but thats not the same.

iceman1001, 2019-08-29:
Hm,  Kali and Nethunter left.

doegox, 2019-08-29:
> > PREFIX=/usr/local
> 
> I think `PREFIX:=/usr/local` is better. So it can be override like `make PREFIX=/usr`.

For my own understanding, could you elaborate on `:=` vs `=` ?
Currently we have `CC = gcc` and we can override it with `make CC=clang`, no need for `:=` I think.

ZeroChaos-, 2019-08-29:
I don't really understand := vs ?=, but documentation is here:
https://www.gnu.org/software/make/manual/html_node/Setting.html#Setting

It is generally nice to use ?= in place of = for CC et al, but as you see it's not required

doegox, 2019-08-29:
well `?=` is a different story... we use it for CFLAGS but we can't for CC because CC is predefined by make as `cc`:

```
$(info $(CC))
CC?=gcc
$(info $(CC))
CC=gcc
$(info $(CC))
```
=>
```
make
cc
cc
gcc
```
```
CC=foo make
foo
foo
gcc
```
```
make CC=foo
foo
foo
foo
```

edward-p, 2019-08-29:
> > > PREFIX=/usr/local
> > 
> > 
> > I think `PREFIX:=/usr/local` is better. So it can be override like `make PREFIX=/usr`.
> 
> For my own understanding, could you elaborate on `:=` vs `=` ?
> Currently we have `CC = gcc` and we can override it with `make CC=clang`, no need for `:=` I think.

Ah, I got a bad memory. Actually I mean `?=`. So it can be override by environment variables.

ZeroChaos-, 2019-08-29:
This is an updated copy of how I install (live in my git repo) to help build make install (or tell me I'm doing it wrong)
https://github.com/pentoo/pentoo-overlay/blob/master/net-wireless/proxmark3/proxmark3-9999.ebuild

doegox, 2019-08-29:
> This is an updated copy of how I install (live in my git repo) to help build make install (or tell me I'm doing it wrong)

* astyle is a `suggest` dependency, not a strong dependency (only for committers actually)
* hf-mattyrun -> HF_MATTYRUN
* hf_colin -> HF_COLIN
* hf_bog -> HF_BOG
* maybe no need for jtag/recovery stuff to be deployed

ZeroChaos-, 2019-08-29:
what are these files for?
```
$PM3_SHARE_PATH/firmware/SIM011
$PM3_SHARE_PATH/firmware/SIM011.sha512.txt
```
thanks for the updates :-)

merlokk, 2019-08-29:
For update iso7816 module. I'm not sure if it needs to install

iceman1001, 2019-08-29:
It is the firmware for SIM Module.

SIM011 and the sha512,  is very much needed.   
From factory,  it doesn't contain latest firmware for the simmodule,   SIM011.bin is the latest fw and the sha512 is used for verfication of the firmware file during upgrade.

iceman1001, 2019-08-29:
@ZeroChaos-   your nice live update iin your repo,  does it use those patches you have aswell?
Do you still need it:  https://github.com/pentoo/pentoo-overlay/blob/master/net-wireless/proxmark3/files/cflags.patch

ZeroChaos-, 2019-08-29:
@iceman1001 no, that file wasn't used anymore, I cleared it :-)

What about SIM0110?

iceman1001, 2019-08-29:
> It is the firmware for SIM Module.
> 
> SIM011 and the sha512, is very much needed.
> From factory, it doesn't contain latest firmware for the simmodule, SIM011.bin is the latest fw and the sha512 is used for verfication of the firmware file during upgrade.

ZeroChaos-, 2019-08-29:
probably want to rename tools/simmodule/readme.txt to something less likely to collide so it can be installed in /usr/share/proxmark3/firmware with the files so folks know what they are.

and sorry, had a typo in that last one, what about SIM010.BIN?

iceman1001, 2019-08-29:
SIM010.bin is the previous fw.
SIM011 is the latest.

doegox, 2019-08-29:
ignore the readme.txt for now, its content should be migrated in the docs/

edward-p, 2019-08-30:
Do `PM3RDV4` and `PM3OTHER` share any targets?

- [ ] bootrom
- [ ] os
- [ ] flash-bootrom
- [ ] flash-os
- [ ] flash-all
- [ ] recovery
- [ ] client
- [ ] mfkey
- [ ] nonce2key
- [ ] fpga_compress

I want know if I can package one package for both  `PM3RDV4` and `PM3OTHER`.

doegox, 2019-08-30:
Common:

- [x] bootrom
- [ ] os
- [x] flash-bootrom > not for packaging
- [x] flash-os > not for packaging
- [x] flash-all> not for packaging
- [ ] recovery
- [x] client
- [x] mfkey
- [x] nonce2key
- [x] fpga_compress > not for packaging

doegox, 2019-08-30:
So basically you only need to bring both fullimage.elf and possibly both proxmark3_recovery.bin if you really want to ship systematically JTAG images...

doegox, 2019-08-30:
But IMHO if someone needs to operate JTAG because he bricked his device, he should be skilled enough to work with the git repo

iceman1001, 2019-08-30:
We don't use "os"  any longer.  We only use the combined new  "fullimage" since a couple of years back.
os + fpga images = fullimage.

I would like to that you all have  proxmark3_recovery.bin 
Its an easy file to keep,  if you jtag you maybe still don't want to git clone if you did a make install....

doegox, 2019-08-30:
@iceman1001 "os" was referring to the make Target, `make os` to get fullimage

iceman1001, 2019-08-30:
its still called `make os` ???  hmm..   that should be renamed to `make fullimage` ?

doegox, 2019-08-30:
yep, this is more coherent, done. `make fullimage` (and added `make fullimage/clean` support)

doegox, 2019-08-30:
Hmm I realize the current brew was just wrong on its share path and should be `$PM3_BREW_PREFIX/share/proxmark3`, so we can just live with the usual PREFIX in the makefile and it will work fine for brew too, I'll change that soon.

iceman1001, 2019-08-30:
I change homebrew to share/proxmark3 ,  hardcoded :)

edward-p, 2019-08-30:
For those shell scripts (which need `PM3_SHARE_PATH`):

- flash-all.sh
- flash-bootrom.sh
- flash-fullimage.sh

Should hardcode with `@PM3_SHARE_PATH@`, then be replaced when installing?

edward-p, 2019-08-30:
This is currently how I package the Arch Linux package: [PKGBUILD](https://github.com/archlinuxcn/repo/blob/master/archlinuxcn/proxmark3-iceman-git/PKGBUILD)

doegox, 2019-08-30:
hmm true, the scripts aren't yet ready for being installed and find their stuff, I'll work on them...

doegox, 2019-08-30:
So here is the updated current plan for make install:

```
PREFIX ?= /usr/local

# Brew (OSX, possibly Linux)
Installation PREFIX = $(brew --cellar)/proxmark3/HEAD-$(git describe --always)
Runtime PREFIX = $(brew --prefix)

$PREFIX/bin/proxmark3
$PREFIX/bin/proxmark3-flasher (currently flasher)
$PREFIX/bin/proxmark3-mfkey32 (currently mfkey32)
$PREFIX/bin/proxmark3-mfkey32v2 (currently mfkey32v2)
$PREFIX/bin/proxmark3-mfkey64 (currently mfkey64)
$PREFIX/bin/proxmark3-nonce2key  (currently nonce2key)
# fpga_compress doesn't need to be installed, it's useless outside compilation
#scripts:
$PREFIX/bin/pm3 (currently proxmark3.sh)
$PREFIX/bin/pm3-flash-all (currently flash-all.sh)
$PREFIX/bin/pm3-flash-bootrom (currently flash-bootrom.sh)
$PREFIX/bin/pm3-flash-fullimage (currently flash-fullimage.sh)
$PREFIX/bin/pm3-eml2lower (currently pm3_eml2lower.sh)
$PREFIX/bin/pm3-eml2upper (currently pm3_eml2upper.sh)
$PREFIX/bin/pm3-mfdread (currently pm3_mfdread.py)
$PREFIX/bin/pm3-mfd2eml.py (currently pm3_mfd2eml.py)
$PREFIX/bin/pm3-eml2mfd.py (currently pm3_eml2mfd.py)

$PREFIX/share/proxmark3/lualibs **search DONE**
$PREFIX/share/proxmark3/luascripts **search DONE**

$PREFIX/share/proxmark3/resources/capk.txt **search DONE**
$PREFIX/share/proxmark3/resources/aidlist.json **search DONE**
$PREFIX/share/proxmark3/resources/oids.json **search DONE**
$PREFIX/share/proxmark3/resources/fido2_defparams.json (currently fido2.json)
$PREFIX/share/proxmark3/resources/emv_defparams.json **search DONE**
$PREFIX/share/proxmark3/resources/iclass_dump.bin **search DONE**
$PREFIX/share/proxmark3/resources/hardnested_bf_bench_data.bin **search DONE**
$PREFIX/share/proxmark3/resources/hardnested_tables/* **search DONE**

# these are nice for newcomers to learn I think
$PREFIX/share/proxmark3/traces/*  **search DONE**
$PREFIX/share/doc/proxmark3/* (doc/)

$PREFIX/share/proxmark3/dictionaries/* **search DONE**

$PREFIX/share/proxmark3/firmware/bootrom.elf
$PREFIX/share/proxmark3/firmware/fullimage.elf
$PREFIX/share/proxmark3/firmware/SIM011
$PREFIX/share/proxmark3/firmware/SIM011.sha512.txt

# Linux only:
UDEV_PREFIX:=/etc/udev/rules.d
$UDEV_PREFIX/77-pm3-usb-device-blacklist.rules
```

This requires of course adaptations of the code to find its data.
It also goes with modifications for the user files:
```
~/.proxmark3/history.txt  **DONE**
~/.proxmark3/log_%Y%m%d.txt  **DONE**
~/.proxmark3/dictionaries/ (user dics) **search DONE**
~/.proxmark3/luascripts/ (user lua scripts) **search DONE**
~/.proxmark3/lualibs/ (user lua libs) **search DONE**
~/.proxmark3/cmdscripts/ (user cli scripts)
~/.proxmark3/resources/iclass_key.bin **search DONE**
~/.proxmark3/resources/iclass_decryptionkey.bin **search DONE**
~/.proxmark3/resources/fido2.json (user fido prefs read/write...)
```

[ ] Still need to adapt the scripts `proxmark3.sh` and `flash-*.sh` for being able to work installed

Please comment and tell me if sth is wrong or missing.

iceman1001, 2019-08-30:
I think you can keep original name for these ones.  There are not proxmark3 bound in that way. 
```
$PREFIX/bin/proxmark3-mfkey32 (currently mfkey32)
$PREFIX/bin/proxmark3-mfkey32v2 (currently mfkey32v2)
$PREFIX/bin/proxmark3-mfkey64 (currently mfkey64)
$PREFIX/bin/proxmark3-nonce2key  (currently nonce2key)
```

And what about the other python scripts in /tools ?

doegox, 2019-08-30:
yes but they would come with proxmark3 package on some OSes, what if original author wants to package it separately ? we're squatting his namespace

doegox, 2019-08-30:
So here is the updated current plan for make install:

```
PREFIX ?= /usr/local

# Brew (OSX, possibly Linux)
Installation PREFIX = $(brew --cellar)/proxmark3/HEAD-$(git describe --always)
Runtime PREFIX = $(brew --prefix)

make install:
$PREFIX/bin/pm3 (currently proxmark3.sh)
$PREFIX/bin/pm3-flash-all (currently flash-all.sh)
$PREFIX/bin/pm3-flash-bootrom (currently flash-bootrom.sh)
$PREFIX/bin/pm3-flash-fullimage (currently flash-fullimage.sh)
$PREFIX/share/proxmark3/tools/pm3_eml2lower.sh
$PREFIX/share/proxmark3/tools/pm3_eml2upper.sh
$PREFIX/share/proxmark3/tools/pm3_mfdread.py
$PREFIX/share/proxmark3/tools/pm3_mfd2eml.py
$PREFIX/share/proxmark3/tools/pm3_eml2mfd.py
$PREFIX/share/proxmark3/tools/findbits.py
$PREFIX/share/proxmark3/tools/rfidtest.pl
$PREFIX/share/proxmark3/tools/xorcheck.py
$PREFIX/share/proxmark3/tools/jtag_openocd/ (or $PREFIX/share/proxmark3/jtag_openocd/ ?)
$PREFIX/share/proxmark3/traces/*  **search DONE**
$PREFIX/share/doc/proxmark3/* (doc/)
$PREFIX/share/proxmark3/firmware/SIM011
$PREFIX/share/proxmark3/firmware/SIM011.sha512.txt
# Linux only:
UDEV_PREFIX:=/etc/udev/rules.d
$UDEV_PREFIX/77-pm3-usb-device-blacklist.rules

# make install includes also all the following ones:

make client/install:
$PREFIX/bin/proxmark3
$PREFIX/bin/proxmark3-flasher (currently flasher)
$PREFIX/share/proxmark3/lualibs **search DONE**
$PREFIX/share/proxmark3/luascripts **search DONE**
$PREFIX/share/proxmark3/resources/capk.txt **search DONE**
$PREFIX/share/proxmark3/resources/aidlist.json **search DONE**
$PREFIX/share/proxmark3/resources/oids.json **search DONE**
$PREFIX/share/proxmark3/resources/fido2_defparams.json (currently fido2.json)
$PREFIX/share/proxmark3/resources/emv_defparams.json **search DONE**
$PREFIX/share/proxmark3/resources/iclass_dump.bin **search DONE**
$PREFIX/share/proxmark3/resources/hardnested_bf_bench_data.bin **search DONE**
$PREFIX/share/proxmark3/resources/hardnested_tables/* **search DONE**
$PREFIX/share/proxmark3/dictionaries/* **search DONE**

make mfkey/install:
$PREFIX/share/proxmark3/tools/mfkey32
$PREFIX/share/proxmark3/tools/mfkey32v2
$PREFIX/share/proxmark3/tools/mfkey64

make nonce2key/install:
$PREFIX/share/proxmark3/tools/nonce2key

make bootrom/install:
$PREFIX/share/proxmark3/firmware/bootrom.elf

make fullimage/install:
$PREFIX/share/proxmark3/firmware/fullimage.elf

make recovery/install:
$PREFIX/share/proxmark3/firmware/proxmark3_recovery.bin
```

This requires of course adaptations of the code to find its data.
It also goes with modifications for the user files:
```
~/.proxmark3/history.txt  **DONE**
~/.proxmark3/log_%Y%m%d.txt  **DONE**
~/.proxmark3/dictionaries/ (user dics) **search DONE**
~/.proxmark3/luascripts/ (user lua scripts) **search DONE**
~/.proxmark3/lualibs/ (user lua libs) **search DONE**
~/.proxmark3/cmdscripts/ (user cli scripts)
~/.proxmark3/resources/iclass_key.bin **search DONE**
~/.proxmark3/resources/iclass_decryptionkey.bin **search DONE**
~/.proxmark3/resources/fido2.json (user fido prefs read/write...)
```

[ ] Still need to adapt the scripts `proxmark3.sh` and `flash-*.sh` for being able to work installed

Please comment and tell me if sth is wrong or missing.

iceman1001, 2019-08-30:
I like this
``` $PREFIX/share/proxmark3/jtag_openocd/ ```

Othervise I would say it is looking like a solid plan.  Not like for one week ago :)

doegox, 2019-08-30:
ack

ZeroChaos-, 2019-08-30:
I agree with @doegox that being able to easily build for both proxmark firmware targets at once would be nice.  However, it does quickly reach an obvious limit since there are a bunch of "standalone" options when building the firmware and it makes less sense to ship 12 firmwares.

Not that I'm against that either really, since the firmware is like 300k

doegox, 2019-08-30:
We've some discussions for having more than one standalone mode in the FW, and some client command to store which one to activate, this would solve this aspect

doegox, 2019-08-30:
`make install` is live \o/
See https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Development/Maintainers.md for - I hope - useful tips.

iceman1001, 2019-08-31:
@doegox  Impressive!   Talk about next level.

PalinuroSec, 2019-08-31:
hello, i am the guy from parrot.
to package this software and reuse the existing makefile, i need to know if all the installation paths have the $DESTDIR prefix

doegox, 2019-08-31:
Yes, else it's a bug :)

PalinuroSec, 2019-08-31:
debian packaging almost ready, i'll open a pull request soon, and upload the package on our build platform asap

PalinuroSec, 2019-08-31:
first experimental pull request (may need some fixes later)
https://github.com/RfidResearchGroup/proxmark3/pull/382

link to our build platform
https://build.parrotsec.org/package/show/parrot-packages/proxmark

if the package works properly, i would like to create a dedicated project on our build platform to build binary artifacts for other distributions like ubuntu, debian, kali or any other debian based plaform you need. we can also build for other systems (suse, fedora, arch etch) but i have never packaged for such distros.

iceman1001, 2019-09-04:
Is all package maintainers happy with the outcome of **make install** ?
@ZeroChaos-  , @edward-p,  @PalinuroSec

edward-p, 2019-09-04:
So far so good!

ZeroChaos-, 2019-09-04:
is there a way to break up "make install" into a few sub tasks?  right now in Pentoo a user can build everything, just the client/proxmark3 target, or the "client/proxmark3 mfkey nonce2key" targets.

Can I get some slightly broken up install targets? :-)

[15 more replies]

closed 2019-09-05 by doegox