RfidResearchGroup/proxmark3 #326: rfidresearchgroup/proxmark3/proxmark3 is a head-only formula when trying to install blue shark

L07tB0y, 2019-08-14:
on OSX Mojave trying to install the bluetooth module getting the following. 

**brew install --with-blueshark proxmark3**
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> New Formulae
zrepl
==> Updated Formulae
annie              cockroach          eccodes            gatsby-cli         godep              h3                 postgresql@9.4     schismtracker
caddy              crystal            exploitdb          glooctl            goreleaser         jruby              pre-commit         webpack
chakra             deno               fselect            go                 h2o                libomp             puzzles            wp-cli

Error: rfidresearchgroup/proxmark3/proxmark3 is a head-only formula
Install with `brew install --HEAD rfidresearchgroup/proxmark3/proxmark3`**

**brew install --HEAD rfidresearchgroup/proxmark3/proxmark3 --with-blueshark**
Warning: rfidresearchgroup/proxmark3/proxmark3 HEAD-47b60d8 is already installed and up-to-date
To reinstall HEAD, run `brew reinstall proxmark3`

**brew install --with-blueshark --HEAD rfidresearchgroup/proxmark3/proxmark3**
Warning: rfidresearchgroup/proxmark3/proxmark3 HEAD-47b60d8 is already installed and up-to-date
To reinstall HEAD, run `brew reinstall proxmark3`

iceman1001, 2019-08-14:
There is a thread over at the forums.
http://www.proxmark.org/forum/viewtopic.php?pid=36163#p36163

bettse, 2019-08-17:
>To reinstall HEAD, run brew reinstall proxmark3

When you see this message, do a `brew remove proxmark3` before attempting another install.  head-only formulas don't seem to support changing the install parameters (adding --with-blueshark) without first removing.

bettse, 2019-08-17:
Also, there is an open issue that isn't macos specific (https://github.com/RfidResearchGroup/proxmark3/issues/322) that might be impacting, so try this command (after doing a `brew remove proxmark2`)
`HOMEBREW_TRAVIS_COMMIT=1da98c7aa6f72c1424069256534bc87a8512b1d2 brew install --with-blueshark --HEAD rfidresearchgroup/proxmark3/proxmark3` (the sha comes from that other issue)

bettse, 2019-08-17:
And as a reminder, the firmware version needs to always match the desktop version, so be sure to flash after the install:
`sudo proxmark3-flasher /dev/tty.usbmodemiceman1 -b /usr/local/Cellar/proxmark3/HEAD-1da98c7/share/firmware/bootrom.elf /usr/local/Cellar/proxmark3/HEAD-1da98c7/share/firmware/fullimage.elf`
If you flashed before and your device isn't showing up in /dev/, then try the hold-down-the-button-while-pugging-it-in-and-flashing-thing:
1. Unplug proxmark3
1. Start the flasher command (it'll say "Waiting for Proxmark3 to appear on...")
1. Hold proxmark3 button (keep holding until instructed otherwise)
1. plug in proxmark3
1. flasher command should find the device and flash it
1. release proxmark3 button
1. unplug proxmark3
1. plug in proxmark3 (basically to give it a power cycle)

iceman1001, 2019-08-20:
There was a bug on OS X,   https://github.com/RfidResearchGroup/proxmark3/commit/6f9ec6dd2426ead3eda737918ff00771700a939b  Which was fixed by @megabug.

The homebrew repo instructions has been updateded,  I think also the relevant documents also.

I think we are ready to close this one.

L07tB0y, 2019-08-28:
> [@L07tB0y](https://github.com/L07tB0y) can we close?

Yes. This issue is resolved

-------- Original Message --------

doegox, 2019-08-28:
@L07tB0y can we close?

closed 2019-08-28 by doegox