RfidResearchGroup/proxmark3 #327: CFLAGS/LDFLAGS not respected

ZeroChaos-, 2019-08-15:
while it looks like CFLAGS and LDFLAGS are respected when I read the Makefile, the build output shows a different story
```
[-] CC uart/uart_win32.c
gcc -MT obj/uart_win32.o -MMD -MP -MF obj/uart_win32.Td -Os -march=native -mtune=native -pipe -frecord-gcc-switches -std=c99 -D_ISOC99_SOURCE -DPRESETS -I. -I../include -I../common -I/opt/local/include -I../common/zlib -Iuart -I./liblua -I../common/mbedtls -I./jansson -I./tinycbor -Wall -Werror -g -O3 -DHAVE_GUI -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -c -o obj/uart_win32.o uart/uart_win32.c
[-] CC dump.c
gcc -DHAVE_STDINT_H -I. -O2 -Wall -Wno-unused-variable -Wno-unused-function -c -o dump.o dump.c  
```
My flags are "-Os -march=native -mtune=native -pipe -frecord-gcc-switches" and you can see they are respected with the first pasted line but not the second.  This appears wrong to me, but maybe it's intentional for some reason?

doegox, 2019-08-15:
No, you're right, we're aware we've to fix the treatment of all standard variables.
E.g. today it's impossible to choose Clang and more aggressive warning compilation flags without editing manually settings in several Makefiles.
See also the https://github.com/RfidResearchGroup/proxmark3/wiki/Roadmap
and if you feel up to the task, don't hesitate to PR :)

doegox, 2019-08-15:
what is the list of defs that should be taken from the env and possibly extended by the Makefile?
* CFLAGS, CXXFLAGS, LDFLAGS, ?

what is the list of vars that should be defined only if not yet defined?
* CC, CXX, LD and helper tools TAR, RM, MV, TOUCH, FALSE, etc?

Makefiles to go through:
* client/Makefile
* client/jansson/Makefile
* client/liblua/Makefile
* client/tinycbor/Makefile
* common/mbedtls/Makefile
* tools/deprecated-hid-flasher/flasher/Makefile
* tools/fpga_compress/Makefile
* tools/mfkey/Makefile
* tools/nonce2key/Makefile

And for ARM:
* common_arm/Makefile.common

doegox, 2019-08-15:
I see two common ways:

* export CFLAGS=-Dfoo; make # Makefile: CFLAGS?=-Dbar  
* make CFLAGS=-Dfoo # Makefile: CFLAGS=-Dbar

Which one should be adopted?

ZeroChaos-, 2019-08-15:
so typically it works like this.  In the Makefile you specify some *non-required* "sane defaults" which allow override like this:

CFLAGS ?= "-O2 -Wall --stupid-fast-math"

and then you add in the things which are required on the next line like:

CFLAGS += "-Drequired-to-build"

This allows the user to push their own CFLAGS and still get the right results:

export CFLAGS="-Os" make
gcc -Os -Drequired-to-build

or

make CFLAGS="-Os"
gcc -Os -Drequired-to-build

iceman1001, 2019-08-15:
That sounds reasonable.    I second @ZeroChaos- suggestion  :)

doegox, 2019-08-16:
Yes but the proposal doesn't work.
```make
CFLAGS ?= -O2 -Wall --stupid-fast-math
CFLAGS += -Drequired-to-build

all:
	@echo gcc $(CFLAGS)
```

```sh
$ export CFLAGS="-Os"; make
gcc -Os -Drequired-to-build
$ make CFLAGS="-Os"
gcc -Os
```

One possibility is:

```make
CFLAGS ?= -O2 -Wall --stupid-fast-math
MYCFLAGS = -Drequired-to-build

all:
	@echo gcc $(MYCFLAGS) $(CFLAGS)

```

and we get the expected:
```sh
$ export CFLAGS="-Os"; make
gcc -Drequired-to-build -Os
$ make CFLAGS="-Os"
gcc -Drequired-to-build -Os
```
But that requires we have explicit rules for all *.o in all Makefiles

ZeroChaos-, 2019-08-16:
Hmm, I've never actually noticed that (because package managers and I don't set flags that way typically), nice catch!

ZeroChaos-, 2019-08-16:
I suppose if you don't want explicit rules for all .o files you could just "not support" setting CFLAGS that way.  After checking some of the packages I maintain that seems to be a fairly common issue.

doegox, 2019-08-16:
ok so just `export`, `?=` and `+=` tools

doegox, 2019-08-16:
This was more complex than I though because "+=" becomes immediately evaluated if there is an external CFLAGS from env, instead of being deferred as usual.
@ZeroChaos- please test https://github.com/RfidResearchGroup/proxmark3/pull/331
CFLAGS is taken from env.
CC can also be ruled, but then with make vars.
So now you can do e.g.
```
CFLAGS="-DFOOBAR -Wall -Werror -O3" make client CC=clang CXX=clang++ LD=clang++ V=1
```
Note if you test other CFLAG, that compilation fails if no `-O` is provided: some fcts supposedly inlined are becoming undefined. Yet another issue...

iceman1001, 2019-08-20:
@ZeroChaos-   @doegox  has remade the Makefiles,  should honor the CFLAGS from env much better now.  Are we one step closer? or?

ZeroChaos-, 2019-08-29:
both CFLAGS and LDFLAGS appear to be respected, I'm good here, thanks!

closed 2019-08-29 by ZeroChaos-