RfidResearchGroup/proxmark3 #328: unbundled mbedtls

ZeroChaos-, 2019-08-15:
I think support for this is pretty common on systems these days and it would be safe to unbundle.

iceman1001, 2019-08-15:
its just we don't use all of it,  and we are using it for both device and client side so we made minor adaptions to it.

ZeroChaos-, 2019-08-15:
Any chance of cleaning up the modifications and sending them upstream?

doegox, 2019-08-15:
So basically
* mbedtls for client could be delivered by system dependency (still probably tricky to cover all targeted host platforms)
* mbed for pm3 firmware needs to be official sources + configuration header file tuned to our needs. Nothing to report upstream, that's just how mbedtls works, you can tune its parameters for your target.

Is there any gain to follow that direction or isn't  it just simpler to keep the same mbed src for both firmware and client ?

ZeroChaos-, 2019-08-15:
it would certainly be better to use the system version for the client if
possible.

iceman1001, 2019-08-15:
So how is that done?  Like in Makefile or?

ZeroChaos-, 2019-08-15:
as with the liblua suggestion, usually it's done in a configure script, but it can be done in a Makefile by testing the presence of a file or something like that then switching to the system version.  I can try to work up some patches if everyone wants to stick with pure Makefiles

doegox, 2019-08-15:
Sure, go ahead!
Thanks!

iceman1001, 2019-08-15:
Yes,  thanks!   Pure Makefiles sounds good.

iceman1001, 2019-08-16:
We actually also have libjansson,  libtinycbor,   zlib, ...

ZeroChaos-, 2019-08-16:
zlib will be easy to unbundle, literally everything has that :-)

doegox, 2019-08-16:
not that easy...
common/zlib/README:
```
// This version of zlib is modified for use within the Proxmark3 project..
// Files from the original distribution which are not required for this
// purpose are not included. All modifications can easily be found
// by searching for #ifdef ZLIB_PM3_TUNED and #ifndef ZLIB_PM3_TUNED.
```

iceman1001, 2019-09-16:
yeah,  I don't see how we going to unbundle mbedtls or zlib just like that. 
Depends if @slurdge  finishes his replacement of zlib -> lz4..    Then we could remove zlib all in all.

 How did it go with unbundle lua @ZeroChaos-  ?

ZeroChaos-, 2019-09-16:
Apparently lua upstream doesn't even support dynamic linking and gentoo is considering removing our patches to allow it.  Honestly that seems bad for everyone to me, but it's not my call.

In the case of zlib and mbedtls it's not a crisis so long as your modified versions are kept in date with upstream and only used for the firmware (not host tools).

doegox, 2019-09-16:
Both cases are different

* (modified) zlib is used in the host tools as well as it's part of the communication between host and firmware and modifications have to match... So I don't see how to change that, unless the modifications aren't much necessary and could be skipped, to be checked, I've no idea.
* as said earlier, probably mbedtls on host could be replaced by vanilla mbed dependency, we need to make the exercise to see.

slurdge, 2019-09-16:
@doegox 

> as it's part of the communication between host and firmware and modifications have to match

Which parts? I don't remember such a match (apart from FPGA upload, but it's more compilation & run).

lz4 has also to be changed, but it's such a simple change (some inline functions shouldn't but inlined for 256k target) that a simple sed can do it. So the update mechanism would be a simple pull. I could also see if #ifdef magic could do it patchless.

I seem to recall that I updated mbedtls to vanilla when doing the replacement for 3des&aes, but I can't recall how it went.

doegox, 2019-09-16:
Yes fpga upload: the host compresses the image and the firmware decompresses it.
cf ZLIB_PM3_TUNED in the code.

iceman1001, 2019-09-16:
no..the device pulls the compress fpga image from its flash,  decompresses it and send to fpga.

doegox, 2019-09-16:
and who compresses  the image? ;) the host...

iceman1001, 2019-09-16:
no.. the compiler runs the fpga-compress.exe   Once.   then we flash it on to the device.
After that host has nothing to do with it..

doegox, 2019-09-16:
We're talking about compilation dependencies to be solved on the *host* OS, not about *client*<>pm3 relationship.
* the host (here via fpga_compress) is using zlib to compress the fpga image
* the ARM firmware is using zlib to decompress the fpga image
So zlib code exists in a specialized state (cf ZLIB_PM3_TUNED) both on the host code and the ARM code.
And one question to explore is to see if we really need these tunings or if we can live with the host OS zlib. 
Another possibility in exploration is to replace zlib by another compressor but the same question about using host OS libs remains.

iceman1001, 2019-09-16:
When you phrase it like that...  who am I to disagree?

doegox, 2019-09-16:
I knew there was no disagreement, just misunderstandings ;)

slurdge, 2019-09-16:
lz4 should be nice in this regard, since you can tune compression & decompression maintaining format (I guess it would also be true for plain zlib).

iceman1001, 2019-09-17:
zlib allows already for it.  What got my juices turning was the speedup you presented. 30% faster.

ZeroChaos-, 2019-09-17:
https://github.com/lz4/lz4/blob/dev/README.md

On its worst setting zlib still compresses 20% better than lz4. Not that I discourage lz4, just know that fast and small are mutually exclusive. Xz is better compression than both and nearly as fast for decompression (thanks to much slower compression, which in this case likely means 2 or 3 more seconds)

doegox, 2019-09-17:
Just some factors to take into consideration:

* fpga image:
  * large single blob, compression only on host, decompression only on ARM
  * compression time doesn't matter much, it's at compilation time
  * decompression time matters, it slows down interactions every time we switch from HF to LF commands and back as the new image must be transferred from fw to fpga.
* client<>arm communications:
  * at this point no compression is used but now that with BT dongle, communication spped is significantly slower than USB, some commands communications might benefit from compression (e.g. when uploading/downloading LF samples buffer)
  * compression/decompression both on host & ARM. Speed is the most important factor, as well as the memory requirements for compression on the ARM side.
* ideally, we should use the same compression algorithm for both use cases, else we get a larger firmware. Still, the balance is to be discussed.

ZeroChaos-, 2019-09-17:
lz4 is meant to be a fast and light stream compressor, built for use case 2.  Xz would be better for use case 1, except use case two being lz4 would mean any gains with firmware compression would be lost by needing both types.

I believe the solution is lz4hc for use case 1 and lz4 for case 2. Lz4hc is a higher compression variant of lz4 which is slower to compress but requires nothing special to decompress. So basically you get the smallest possible firmware and have the lz4 lib to handle the stream compression.

iceman1001, 2019-09-17:
When it comes to use case 2,  we talk about 40kb to be compressed.   Or if we dump whole external flash,  that would mean 256kb.   Since we only have 64kb ram to play with,   we need to keep the memory footprint small.

slurdge, 2019-09-17:
For the lz4 branch, I already used lz4hc. Gains were not that much because of the limit of bigbuff which is around 38kb for decompression buffer (so your compression window cannot reference back more than 38kb).
One thing to consider for lz4 vs zlib vs xz is that since the data compressed is quite small, the size of the decompression routine is not negligible. That's why I had so little differences between lz4dec + lz4data vs zlibdec + zlibdata. zblibenv vs lz4enc will accentuate this and probably it will fall into lz4 advantage (not backed by hard numbers, just my hunch).
I didn't try xz because it seemed much more complicated to embed.

@doegox When you mean speeding up, you specifically speaks about bandwidth, right ? Ping/Latency is a separate problem ?

doegox, 2019-09-17:
@slurdge I was talking about speed of the compression/decompression such that total time budget is a win when compared with comms without compression. Latency is another parameter to take into account indeed

ZeroChaos-, 2019-09-17:
stream compressors are meant for those, and lz4 is a stream compressor.

Use max compression lz4hc for anything you compress once.
Use pretty normal lz4 settings for any thing which needs to be compressed and decompressed live to save a little bandwidth
Use no compression for things which are extremely latency sensitive.  That said, the point of stream compressors like lz4 is speed, so throughput tends to be very high and it may not hurt latency intolerably.

doegox, 2020-05-23:
Hmm this thread is a bit messy with mbedtls / lua / z<>lz4 discussions.
Lua: done
z: can't be done, specialized
Back to mbedTLS:
I tried but stumbled upon two issues:
* pkg-config support is not yet there, see https://github.com/ARMmbed/mbedtls/pull/1753 => I tried hardcoding include / lib locations for now
* Missing support for CMAC in mbedtls default configuration, see https://stackoverflow.com/questions/41973075/aes-cmac-using-mbedtls-undefined-reference-error
So it fails with
```
/usr/bin/ld: obj/crypto/libpcrypto.o: in function `aes_cmac8':
libpcrypto.c:(.text+0x1e6): undefined reference to `mbedtls_aes_cmac_prf_128'
/usr/bin/ld: obj/crypto/libpcrypto.o: in function `ecdsa_signature_r_s_verify':
libpcrypto.c:(.text+0x9b4): undefined reference to `ecdsa_signature_to_asn1'
/usr/bin/ld: obj/crypto/libpcrypto.o: in function `aes_cmac':
libpcrypto.c:(.text+0x1b9): undefined reference to `mbedtls_aes_cmac_prf_128'
/usr/bin/ld: obj/emv/test/cryptotest.o: in function `ExecuteCryptoTests':
cryptotest.c:(.text+0xd9): undefined reference to `mbedtls_cmac_self_test'
```

iceman1001, 2020-05-24:
How about we close this one then,   with we tried and it isn't worth the hassle?

ZeroChaos-, 2020-05-24:
Works for me

doegox, 2020-05-24:
ok, btw I'll also have a look at unbundling jansson, libcbor and libwhereami, see what's possible or not

closed 2020-05-24 by doegox