RfidResearchGroup/proxmark3 #329: latest reveng stuff breaks on proxspace.

iceman1001, 2019-08-15:
**Describe the bug**
reveng commits breaks on proxspace

**To Reproduce**
read https://github.com/Gator96100/ProxSpace/issues/16 

**Expected behavior**
a clean compilation

**Desktop (please complete the following information):**
latest proxspace env.

**Additional context**

doegox, 2019-08-15:
So now that's the mingw64 which is broken while mingw32 is fine! Sigh... I'll have a look this evening.
His error message seems to indicate that even on mingw64, revenge wants 32b setup

doegox, 2019-08-15:
ok on mingw64 an unsigned long is only 32b, while it's 64b on linux64.
`sizeof(unsigned long)` is the dynamic way, but I've to find static way with compilation vars...
Or, if not possible, reveng helper to modify these vars dynamically...

doegox, 2019-08-15:
should be fixed by last commit

closed 2019-08-15 by doegox

iceman1001, 2019-08-15:
And now I will look at reveng 2.0.1...