RfidResearchGroup/proxmark3 #332: Travis CI on MacOSX does not checkout the proper branch

doegox, 2019-08-16:
See e.g. https://travis-ci.org/RfidResearchGroup/proxmark3/jobs/572902998 which should have failed building branch `makefile_restruct` but instread succeeded in building branch `master`

I guess the brew mechanism fetches unconditionally the master.

Now I understand why a previous PR which was apparently ok, broke stuffs once merged...

iceman1001, 2019-08-17:
@merlokk   our CI expert...

merlokk, 2019-08-17:
i think problem is here `brew install --HEAD proxmark3`
ill look how to fix

iceman1001, 2019-08-17:
We would need to compile two versions on Travis CI for OSX.
With and without BT support.
```
brew install --HEAD proxmark3
brew install --with-blueshark proxmark3
```

iceman1001, 2019-08-17:
I also think it needs to be  like this,  but not sure
```
brew install --HEAD --with-blueshark proxmark3
```

doegox, 2019-08-17:
It would be far easier if Brew was like ProxSpace: only responsible to provide the environment.
Then we take over as usual with git/make instructions.
So no need for magic tricks to get BT support, to fetch a branch, to let Travis testing the branch and the PR (so the merge of the branch in master), etc.
And when we're starting to work on the make install, we can have make install running on MacOSX too, without relying on Brew and being tied to the brew formula expectations in terms of directories etc.

merlokk, 2019-08-17:
there was an idea that brew can be used by users to have proxmark installed.
i'd prefer that we have sh file to get and make, but maybe it will be better to fix current system.
ill try and lets see.

RfidResearchGroup, 2019-08-17:
Thanks @Merlokk  .   I added you and @doegox  as collabs to that repo so you can merge PRs  :)
Also changed some textual things and removed old directives of hid-flasher.

merlokk, 2019-08-17:
now it looks it works with branches, but not works with PR's
@doegox if you have changes to repo - check please branches testing (i dont want to make strange commits)

merlokk, 2019-08-18:
ok. it works with branches.

merlokk, 2019-08-18:
1. make work my script for osx
2.  PR compiling. maybe fully get rid of brew install and move logic to plain make all (needs to think how to debug brew install or just let it work wo this)
3. make compile matrix to check some modes (now for btaddon)

iceman1001, 2019-08-19:
time to close?

doegox, 2019-08-19:
ok

closed 2019-08-19 by doegox