RfidResearchGroup/proxmark3 #347: ISO15693 access right

Mera-balou, 2019-08-20:
When an ISO15693 tag is dumped with PM3 RDV4 (`hf 15 dump`), table with memory content doesn't show acces rights :
```
block#   | data         |lck| ascii
---------+--------------+---+----------
  0/0x00 | 09 08 73 C5  | 0 | ..s.
  1/0x01 | 42 1E 40 00  | 0 | B.@.
  2/0x02 | 1E 80 53 42  | 0 | ..SB
```
"**NFC TagInfo**" (Andoid apk by NXP) give more details about access right (same tag here):

[image]

Is it possible to add these informations (r / w / x ...) inside table (and dump file) when `hf 15 dump` command is executed?

iceman1001, 2019-08-20:
Yup,  its possible and it would be a nice addition.

iceman1001, 2019-08-20:
you have  "lock"  column,  but I don' think its actually working as of now.

martinbrook, 2024-11-11:
Not working on v4.16191 but lock status is showing OK on mainline c74edafff