RfidResearchGroup/proxmark3 #349: separate building flasher from proxmark3 client

ZeroChaos-, 2019-08-20:
running "make client" builds both flasher and proxmark3.  While they both happen to be in the "client" folder, building flasher *and no firmware* seems like a waste.  please separate out flasher into it's own target and depend on it for the firmware and firmware flashing targets.

doegox, 2019-08-20:
flasher could be moved (and renamed) in /tools as for the HID flasher, agreed.
For the dependencies I don't know. The host for cross-compiling is not automatically the same as the host for client & flashing operations. E.g. on Android you can make client & flasher but you've to use a FW built somewhere else.

doegox, 2019-08-20:
BTW right now you can already `make client/proxmark3`  and `make client/flasher`

iceman1001, 2019-08-20:
yup,  if its the separation you are after.

doegox, 2019-08-20:
trying to do the exercise to get the flasher in /tools...
Actually flasher code is heavily entangled with client code, that would be a lot of work to separate them and promote common functions to /common.
I guess it's not worth, you can already compile proxmark3 client alone with proper make invocation.

ZeroChaos-, 2019-08-20:
oh nice, I didn't realize that was possible.  that's enough for me.  if you don't want to detangle everything you may close this bug.  I'll leave it open to track your choice but I'm fine with what is available.

iceman1001, 2019-08-20:
someone didnt look in the makefile...  ;)

closed 2019-08-20 by iceman1001

ZeroChaos-, 2019-08-20:
> someone didnt look in the makefile... ;)

oh I looked, I just wasn't wearing my glasses