RfidResearchGroup/proxmark3 #350: OSX : port discrepancies instructions <> script, which one is correct?

doegox, 2019-08-20:
https://github.com/RfidResearchGroup/proxmark3/blob/master/doc/md/Installation_Instructions/Mac-OS-X-Homebrew-Installation-Instructions.md mentions
* `/dev/tty.usbmodemiceman1`
* `/dev/tty.usbmodem881`

while https://github.com/RfidResearchGroup/proxmark3/blob/master/proxmark3.sh#L24 mentions
* `/dev/pm3-*`
* `/dev/cu.usbmodem*`

OSX users, can you help me understanding which statements are correct and what needs to be fixed?

And is it correct that you always need sudo on OSX for proxmark3 or flasher ?

doegox, 2019-08-20:
Poking @cjbrigato @drandreas

iceman1001, 2019-08-20:
this one is invalid since I renamed it to iceman :)
```
/dev/tty.usbmodem881
```

doegox, 2019-08-20:
yeah but people can come with pm3s with other firmwares and want to migrate to RRG/iceman fork ;)

doegox, 2019-08-20:
https://stackoverflow.com/questions/8632586/macos-whats-the-difference-between-dev-tty-and-dev-cu/8632603 => so we got both `/dev/tty.usbmodemiceman1` and `/dev/cu.usbmodemiceman1` but tty is safer to use.
Tested: indeed if trying to open a second time on `tty` I get *claimed by another process* error while trying to open on `cu` I get a *invalid serial port*.
I'll fix the script.

luminouw, 2019-08-20:
Always had ```/dev/tty.usbmodemiceman``` here, I think the flash-all.sh script isn't really working as intended on mac as well, I always end up using the flasher directly (I'll take a look and report)

closed 2019-08-21 by doegox