RfidResearchGroup/proxmark3 #351: make -j4 all fails

ZeroChaos-, 2019-08-22:
https://github.com/pentoo/pentoo-overlay/issues/497

I'm still troubleshooting this, but -j, -j1, -j12 all work, yet -j4 fails on a freaking mkdir call

ZeroChaos-, 2019-08-22:
The easiest thing to do it probably to just use mkdir -p in place of mkdir in Makefile.host.  not sure what generates Makefile.host yet though

blshkv, 2019-08-22:
```mkdir -p``` is not recommended:

https://www.gnu.org/software/make/manual/make.html#Utilities-in-Makefiles
```
Generally, stick to the widely-supported (usually POSIX-specified) options and features of these programs. For example, don’t use ‘mkdir -p’, convenient as it may be, because a few systems don’t support it at all and with others, it is not safe for parallel execution. 
```

ZeroChaos-, 2019-08-22:
I mean, mkdir -p solves the issue and I honestly only care about linux.

Upstream already uses mkdir -p in tools/deprecated-hid-flasher/flasher/Makefile and tools/install-gnuarm4.sh, so I'm going to hope they like this solution.

Please change Makefile.host to use "mkdir -p" in place of mkdir :-)

blshkv, 2019-08-22:
https://adirondackfirefly.wordpress.com/2012/04/12/makefiles-howto-threadsafe-mkdir/

```mkdir -p $@
Race conditions for make -j creating directories and targets.```

The solution:
The user function ‘mkdir_deps’

blshkv, 2019-08-22:
also, https://stackoverflow.com/questions/99132/how-to-prevent-directory-already-exists-error-in-a-makefile-when-using-mkdir

iceman1001, 2019-08-22:
ha, I got that problem too,  and added mkdir -p yesterday.   Do you still have this issue @ZeroChaos- ?

blshkv, 2019-08-22:
https://github.com/RfidResearchGroup/proxmark3/commit/3cf64f9f23712830fbeaef2028ca46b9224f1030

That should fix it. We haven't tried the latest commit.

Feel free to close this issue. Well, unless you want to fix it properly, as per my links.

iceman1001, 2019-08-22:
yeah,  its a dirty fix so it works better now.  We make a better fix according to the suggestions in your referenced links later.  So much to do.  Feel free to contribute!

closed 2019-08-22 by iceman1001