RfidResearchGroup/proxmark3 #354: GdkPixbuf-CRITICAL and warnings

gtpy, 2019-08-22:
**Describe the bug**
get pixbuf error in client.
(a little bit offtopic, but maybe anyone has an idea :) )

**To Reproduce**
Steps to reproduce the behavior:
1. latest source
2. make clean && make all
3. ./proxmark.sh
4. See error

**Expected behavior**
No pixbuf warning / error

**Screenshots**
```
** (proxmark3:14457): WARNING **: 22:04:30.661: Invalid borders specified for theme pixmap:
        /usr/share/themes/Yaru/gtk-2.0/assets/scale-horz-trough-active.png,
borders don't fit within the image
```

**Desktop (please complete the following information):**
 - OS: ubuntu 19.04

**Additional context**
It is a ubuntu problem on my parrot system is all fine.

Anyone an idea how to fix this pixbuf warnings

doegox, 2019-08-22:
Apparently it's a problem linked to your Yaru theme.
I tried to reproduce the issue by installing Yaru theme on my Cinnamon but that doesnt' trigger the warning.
Other projects reporting similar errors didn't show easy fix besides fixing the theme itself or removing the feature that triggers the bug (e.g. a menu separator)

gtpy, 2019-08-22:
Thanks @doegox 
I’ll take a try to fix the theme

closed 2019-08-22 by gtpy