RfidResearchGroup/proxmark3 #355: Hitag AES

TACIXAT, 2019-08-23:
**Is your feature request related to a problem? Please describe.**
I have a Hitag AES chip (PCF7939MA). I attempt to do a read and get:

```
proxmark3> lf hitag info
#: DEBUG: Error - failed getting UID 
```

```
proxmark3> lf hitag reader 26
#: DEBUG: Error - hitag failed 
```

On the first try it takes a second and seems like it is doing an actual read. After that it returns instantly with the same error.

I am not sure if this is a bug or it is because the Hitag AES protocol is not implemented. If it is just the protocol being wrong, I'd be happy to implement it if I could find a spec...

**Describe the solution you'd like**
Get the Hitag AES protocol implemented (happy to do it, not sure where to start).

TACIXAT, 2019-08-23:
I can not get a read on the LF card provided with the device either. Maybe related to the LF antenna issues?

doegox, 2019-08-23:
Provided card is a blank t55xx. Try to do some lf xx write and then read it back.
Hitag demodulation is pretty weak right now, try different antenna position and distance

mwalker33, 2019-08-23:
What does 'hw tune' show?
What model proxmark?
On my rdv4 I found I got better results by placing the pm3 on a small peice of foil.  I.e. foil - PM3 - lf card. Note, ensure the cover is on, you wont want to short out something.

The rdv4.01 low q antenna works well with no mods.  So if you have the rdv4.01 ensure the lf antenna switch is in the low q position and try again.

If anyone has a concern with the foil trick, please let me know, I dont want to recommend bad things.

sliceratwork, 2019-08-23:
@mwalker33  sounds like you made a little reflector for the antenna with that foil, that should help with the signal as it increases the antenna gain.

The foil is just on one side of the antenna right, not covering it all around?  If it's all around, that's no bueno :)

mwalker33, 2019-08-23:
Yeah, one side.... like in the attached picture, but all lined up.
[image]

sliceratwork, 2019-08-23:
@mwalker33, yeah that behaves like a reflector. makes sense that you'd get a better signal.

I was gonna try something similar with some aluminum adhesive tape stuck on one side of the plastic case. Also want to stick some adhesive ferrite on the PCB/components (not the antennas) to get rid of some of the electrical noise. Every bit counts :)

mwalker33, 2019-08-23:
HF did not like it being there, so removable is needed.

Anyway, I dont want to hijack this issue, just thought it was worth a try.

iceman1001, 2020-04-25:
@TACIXAT   time to close?

aczid, 2020-06-25:
HITAG AES uses an entirely different command set from HITAG2, so `lf hitag reader 26` will never work on this transponder. You could try all 5-bit commands by using `lf cmdread` in HITAG BPLM modulation to get some kind of response from it.