RfidResearchGroup/proxmark3 #356: unable to only get the UID from ISO14443B'

thunder0666, 2019-08-23:
with the latest pm3 :
[ ARM ]
  bootrom: RRG/Iceman/master/440c05cd 2019-08-23 14:45:16
       os: RRG/Iceman/master/440c05cd 2019-08-23 14:45:18
  compiled with GCC 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

 [ FPGA ]
  LF image built for 2s30vq100 on 2019-07-31 at 15:57:16
  HF image built for 2s30vq100 on 2018-09-03 at 21:40:23

unable to get the iso14443B' UID.
with the following command : 
hf search ---> found nothing
hf 14b info ----> found nothing
hf 14b raw -c -p 05 00 08
get this:
[usb] pm3 --> hf 14b list
[+] Recorded Activity (TraceLen = 14 bytes)
[=]
Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
ISO14443B

      Start |        End | Src | Data (! denotes parity error)                                           | CRC | Annotation
------------+------------+-----+-------------------------------------------------------------------------+-----+--------------------
          0 |       6544 | Rdr |05  00  08  39  73                                                       |  ok | WUPB

---------------------------------------------------------------------------------
so i try with a SCL3711 usb with libnfc

nfc-list
nfc-list uses libnfc libnfc-1.7.1-219-g454a8c4
NFC device: SCM Micro / SCL3711-NFC&RW opened
1 ISO14443B' passive target(s) found:
ISO/IEC 14443-4B' (106 kbps) target:
                DIV: 02  4f  9f  68
                ATS: 3b  6f  00  00  80  5a  0a  01  02  20  03  11  02  4f  9f  68  82  90  00
--------------------------------------------------------------------------------------------
DIV is the UID

working in source of libnfc i found this : 
to get the information in two parts :
// Application Family Identifier (AFI) must equals 0x00 in order to wakeup all ISO14443-B PICCs (see ISO/IEC 14443-3)
first send  NMT_ISO14443B --> 00

// APGEN
and send NMT_ISO14443BI --> 01 0B 3F 80
the last values is CRC on 2 bytes

unfortunately i try 
pm3 --> hf 14b raw -s -p 00
result: 
          0 |       6884 | Rdr |05  00  08  39  73                                                       |  ok | WUPB

pm3 --> hf 14b raw -s -p 010B3F80
result:
          0 |       6884 | Rdr |05  00  08  39  73                                                       |  ok | WUPB

stranger things append ?

thanks for reading

but with the libnfc and nfc tools

iceman1001, 2019-08-24:
https://github.com/RfidResearchGroup/proxmark3/issues/215

doegox, 2019-08-28:
closing, as duplicate of #215

closed 2019-08-28 by doegox