RfidResearchGroup/proxmark3 #357: OSX Compilation fail on cmdhfmfhard.com on 6ac9d2f

cjbrigato, 2019-08-26:
**Describe the bug**
```
Undefined symbols for architecture x86_64:
  "_inflate", referenced from:
      _init_bitflip_bitarrays in cmdhfmfhard.o
  "_inflateEnd", referenced from:
      _init_bitflip_bitarrays in cmdhfmfhard.o
  "_inflateInit2_", referenced from:
      _init_bitflip_bitarrays in cmdhfmfhard.o
ld: symbol(s) not found for architecture x86_64
``` 

**To Reproduce**
compile 6ac9d2f on osx
Don't know exactly on what commit it started

**Expected behavior**
compilation working

**Desktop (please complete the following information):**
 - OSx

**Additional context**

Add any other context about the problem here.

cjbrigato, 2019-08-26:
more to come

cjbrigato, 2019-08-26:
```
~/lastpr/summer/proxmark3 master*
❯ grep _init_bitflip_bitarrays -r .
Binary file ./client/obj/cmdhfmfhard.o matches
```
Well I guess it's hard to link against if we don't have the actual code.
Did I mess up with the repo or am I not woke up enough ?

cjbrigato, 2019-08-26:
ok my bad, was on my side. 
This is complicated monday.

Closing

cjbrigato, 2019-08-26:
Well not so early,
last fullimage just won't boot. hum

cjbrigato, 2019-08-26:
aaand my bad again, I was with my full ARM (no thumb) boot rom, so it was expecting ARM start.c etc. closing

closed 2019-08-26 by cjbrigato

doegox, 2019-08-26:
[image]